OctoFarm/OctoFarm #715: Change the log time stamp from GMT to other GMT

ldiegos, 2021-07-26:
**Please point out the url of the documentation issue**
e.g. https://octofarm.net/installation#Docker

**Please describe the correct steps that should be updated on the documentation**
The log files from octofarm, seems to be always in GTM, and i'm in GMT+1 - Paris, Madrid. Could you explain or detail in a "guide" in the octofarm site how to change this?

**Additional context**
[image]

davidzwa, 2021-07-26:
To answer your question directly, we do not have this option without changing the code. That can change as it is quite an easy fix (at least for the logging format). In that case I kindly ask for your patience so we can apply that to the next release.

#### Changing the code (this is reverted when you update OctoFarm!)
In `server_src/lib/logger.js` we've set-up a formatter for dates. Find the code here:
https://github.com/OctoFarm/OctoFarm/blob/e99cabd3f27f3ab6e21c27f0da5e8e319c4a2751/server_src/lib/logger.js#L9
[image]
We've manually enforced UTC there. If I change it the timezone value to "Europe/Amsterdam" the format changes to GMT+1/+2.

Europe/Madrid or Europe/Paris would result in the same +1/+2 offset.
https://en.wikipedia.org/wiki/List_of_tz_database_time_zones

Hope that this temporarily offers a solution.

ldiegos, 2021-07-26:
Thank you so much @davidzwa !!! 
Sure, i'm not in a hurry, i will wait as long as you need. 

I will stop updating the project when it start (i have a script to download the last version and start it) and i will change that line in the *.js

Again, thank you so much.

NotExpectedYet, 2021-07-26:
Worth a .env variable for this?

ldiegos, 2021-07-26:
Hi,
    What ever you want, but please, invest some minutes to create a guide or some notes in the octofarm web site.
Thank you so much!

NotExpectedYet, 2021-07-27:
> Hi,
> What ever you want, but please, invest some minutes to create a guide or some notes in the octofarm web site. Avoiding to lose it when you update the project with the "git pull"
> 
> Thank you so much!
> Luis.

Sorry that was a question for david aha. 

Yeah that's what this will sort :)

davidzwa, 2021-07-27:
@NotExpectedYet I'll answer just in case, adjusted the title to what my plan was.

NotExpectedYet, 2021-07-27:
> @NotExpectedYet I'll answer just in case, adjusted the title to what my plan was.

I saw :)

closed 2021-10-03 by github-actions[bot]