OctoFarm/OctoFarm #718: Database issue in docker system (Windows host)

ilpaso, 2021-08-02:
### OctoFarm Version

1.1.13

### OctoPrint Version(s)

1.6.0

### pm2 or docker?

docker

### Operating System

Windows

### Location of the problem

MongoDb

### Description of the problem

```bash
after running the "docker-compose up -d" command the web interface returns the page:
"OctoFarm failed to connect with MongoDB. Please adjust, test and save your connection setting below."
but the MONGO variable is correct because when I test the connection the page returns: 
"Last OctoFarm response:
Successfully tested, you can save this setting!"

The same with other adrresses in MONGO variable like:
- ip host
- 127.0.0.1

This is my docker-compose.yml:
services:
  mongodb:
    image: mongo:latest
    container_name: mongodb
    command: ["--bind_ip_all"]
    environment:
      MONGO_INITDB_ROOT_USERNAME: user
      MONGO_INITDB_ROOT_PASSWORD: pass
      MONGO_INITDB_DATABASE: octofarm
    ports:
     # HOST:CONTAINER
    - 27017:27017
    volumes:
    # Local volume (change to mongodb-data for a named volume folder)
    - ./mongodb-data:/data/db
    restart: unless-stopped

  octofarm:
    container_name: octofarm
    # choose octofarm/octofarm:latest or octofarm/octofarm:alpine-latest    
    image: octofarm/octofarm:latest
    restart: always
    mem_limit: 400m # Feel free to adjust! 400 MB is quite high and a safety limit.
    ports:
    - 4000:4000 # port of SYSTEM : port of CONTAINER
    environment:
    - MONGO=mongodb://user:user@mongodb:27017/octofarm?authSource=admin
    volumes:
    # Volumes as local relative folders (validate with 'docker-compose config')
    - ./OctoFarm/logs:/app/logs
    - ./OctoFarm/scripts:/app/scripts
    - ./OctoFarm/images:/app/images
```

### Steps taken to reproduce

```bash
1. edit the docker-compose.yml file
2. run "docker-compose up -d"
3. open the web interface to "http://localhost:4000/"
```

### What did you expect to happen

```bash
avoid the "Database Issue!" page
```

### To avoid the problem I had to

```bash
...
```

davidzwa, 2021-08-02:
@ilpaso thanks for reporting your issue. I think you should check your MONGO url for the `octofarm` service. It says `user:user` but looking at your password it should be `user:pass`
The full URL would then become
```
    - MONGO=mongodb://user:pass@mongodb:27017/octofarm?authSource=admin
```

ilpaso, 2021-08-02:
@davidzwa thank you for your fast reply.
Yes, the MONGO variable was wrong here only because I changed my user and password to a generic "user" and "pass".
My "docker-compose.yml" file has the right user and pass.
The "test connection" works with my MONGO variable but the "Database Issue!" page is always there.

davidzwa, 2021-08-02:
Ok I suspected that it was just a typo here. No problem. For your information: this error page is often shown when any startup error is thrown, regardless if its a database issue or not. Something we'd like to rectify soon!

About your actual problem, I'll need you to share logs. If this is uncomfortable, you can also reach me on discord or you can check the logs folder for the `pm2` output or error logs. Whatever works for you.

ilpaso, 2021-08-02:
Attached here you can find the pm2 logs files.
[pm2.log](https://github.com/OctoFarm/OctoFarm/files/6916723/pm2.log)
[pm2.error.log](https://github.com/OctoFarm/OctoFarm/files/6916725/pm2.error.log)

davidzwa, 2021-08-02:
Ok this is a database problem, confirmed.
> `MongooseServerSelectionError: connect ECONNREFUSED 172.22.0.2:27017`

I spot the `172.22.0.2` ip address in your pm2.log, which is a correct IP address for the docker network. This means that `mongodb` is correctly resolved to `172.22.0.2`. So by elimination I would point to the mongo docker service.

Please try to remove or tell me why you added this:
`    command: ["--bind_ip_all"]`

davidzwa, 2021-08-02:
> Ok this is a database problem, confirmed.
>
>
> I spot the `172.22.0.2` ip address in your pm2.log, which is a correct IP address for the docker network. This means that `mongodb` is correctly resolved [...]

I was just informed that the `bind_ip_all` binds to 0.0.0.0 meaning that `mongodb` or `172.22.0.2` would not work anymore. As docker automatically exposes the mongodb container to your LAN these days, I would just remove the command as stated above.

Curious, where did you learn to add that addition?

ilpaso, 2021-08-02:
I found `command: ["--bind_ip_all"]` in the mongodb documentation.
Without `command: ["--bind_ip_all"]` the problem is not solved.

I found where is the issue: 
Mongodb takes longer to start. So the octofarm container tests the connection before mongodb is fully started.
Restarting only octofarm container after a while solved the problem.

In order to automate the process in docker-compose maybe this link can help: [https://selahattinunlu.medium.com/how-to-wait-for-a-container-to-be-ready-757bc1a86468](url)

davidzwa, 2021-08-02:
Ah so this is the problem of https://github.com/OctoFarm/OctoFarm/issues/674! Very good find @ilpaso and a great tip. We were a bit lost on that one.

We will make quick work of that :) Much appreciated.