OctoFarm/OctoFarm #730: Printer groups not working as intended

pat1121, 2021-08-06:
### OctoFarm Version

1.1.13

### OctoPrint Version(s)

1.6.1

### pm2 or docker?

docker

### Operating System

Unraid

### OctoPrint Plugins

none used

### Location of the problem

Web Interface

### Description of the problem

```bash
...
When creating groups of printers, items named the same are replicated in the group dropdown.

Also, when selecting one of these groups, all printers disappear from the view.
```

### Steps taken to reproduce

```bash
1.Put 2 printers in the same group (ie Group 1)
2. 'Group 1' will appear in the filter menu twice
3. Selecting group 1 as a filter will make all printers disappear from the view
...
```

### What did you expect to happen

```bash
Only a single group to appear for each item (ie Group 1)
The printers inside that group to be on the screen after the 'group 1' filter is selected
```

### To avoid the problem I had to

```bash
have not figured out a way around the problem yet.
```

pat1121, 2021-08-06:
[image]

NotExpectedYet, 2021-08-06:
Cheers for loggin that will get it sorted. Next release might be a while that includes the fix but can let you know when it hits development.

NotExpectedYet, 2021-10-16:
[image]
 
woop

commit 93beffd9 (2021-10-17): Bug Fixes Galore (#760)

* update versions for 1.2 realease

* clean up

* further clean up

* further further clean upper

* octofarm client package fix

* fix incorrect client import

* fix no connection logs spinner

* update packages

* fix incorrect information been passed to error modal

* fix scan network for printers

* fix importing of printers and websocket issue

* fix cors check

* also remove ping message

* Printer Manager bulk commands update

* Printer Manager plugins + fix uniqueness

* Fix double listener issue, should also fix re-creating the connection with different API ke [... cut]
closed 2022-05-19 by NotExpectedYet