OctoFarm/OctoFarm #733: After enabling InfluxDB, OctoFarm won't connect to MongoDB

bepstein111, 2021-08-10:
### OctoFarm Version

latest

### OctoPrint Version(s)

latest

### pm2 or docker?

docker

### Operating System

Ubuntu Server

### Location of the problem

MongoDb

### Description of the problem

```bash
I have OctoFarm setup via a compose stack in Portainer. After configuring everything to my liking, I tried to enable the InfluxDB capibilities, and after the required restart after that, I'm just getting the "database error" page when I browse to my octofarm instance. Possibly related, is there supposed to be a "save" button after you successfully do the test on that page? It says to test before saving, but I see no option to save.
second side note, docker compose seems to be having issues with an environment variable having both a ":" and a "=" in the value, since neither can be properly used as a separator between key and value. doesn't seem to be a problem since I'm able to hit the test button and successfully connect...
```

### Steps taken to reproduce

```bash
1. Start docker-compose stack in portainer, browse to instance, configure 2 local printers
2. sync filament-manager
3. input influxdb details, enable influxdb, restart when it asks you to
4. bam. good luck.
```

### What did you expect to happen

```bash
Log in to octofarm, see everything, and also have my data appear in influxDB.
```

### To avoid the problem I had to

```bash
I wish I knew!
```

davidzwa, 2021-08-10:
@bepstein111 thanks for your bug report. FYI the database page shows in case of any startup error whether its related to database or not. This is something we are working on to improve as your database is probably fine (not the InfluxDb, I mean MongoDB).

In order to proceed we'd need you to share logs or look into the pm2 logs to spot the actual error, if you please. This error could be something like "Influx could not connect" or something else.

bepstein111, 2021-08-10:
ahhhhh this makes way more sense. okay. just out of curiosity, is there somewhere I can go to manually change the influx settings to un-bork it, or do I just need to do it right next time?
Here are my log files. I thumbed through them and couldn't find anything that stuck out other than the miles of `Safe to ignore CastError: Cast to ObjectId failed for value "18" at path "_id" for model "Profile"`

[pm2.error.log](https://github.com/OctoFarm/OctoFarm/files/6964979/pm2.error.log)

[pm2.log](https://github.com/OctoFarm/OctoFarm/files/6964978/pm2.log)

davidzwa, 2021-08-11:
Looking at the bottom of your pm2.log there appear Mongoose errors. Mongoose controls Mongo so there must be a MongoDb error when the docker stack is started up.

This means its a duplicate of #718 and #674. I'll await your confirmation.

Here's the error which lead me to this conclusion:
```
2021-08-10T15:35:49: MongooseServerSelectionError: getaddrinfo ENOTFOUND mongodb
2021-08-10T15:35:49:     at Function.Model.$wrapCallback (/app/node_modules/mongoose/lib/model.js:4898:32)
2021-08-10T15:35:49:     at /app/node_modules/mongoose/lib/query.js:4420:21
2021-08-10T15:35:49:     at /app/node_modules/mongoose/lib/helpers/promiseOrCallback.js:31:5
2021-08-10T15:35:49:     at new Promise (<anonymous>)
2021-08-10T15:35:49:     at promiseOrCallback (/app/node_modules/mongoose/lib/helpers/promiseOrCallback.js:30:10)
2021-08-10T15:35:49:     at model.Query.exec (/app/node_modules/mongoose/lib/query.js:4419:10)
2021-08-10T15:35:49:     at model.Query.Query.then (/app/node_modules/mongoose/lib/query.js:4502:15)
2021-08-10T15:35:49:     at runMicrotasks (<anonymous>)
2021-08-10T15:35:49:     at processTicksAndRejections (internal/process/task_queues.js:93:5) {
2021-08-10T15:35:49:   reason: TopologyDescription {
2021-08-10T15:35:49:     type: 'Single',
2021-08-10T15:35:49:     setName: null,
2021-08-10T15:35:49:     maxSetVersion: null,
2021-08-10T15:35:49:     maxElectionId: null,
2021-08-10T15:35:49:     servers: Map(1) { 'mongodb:27017' => [ServerDescription] },
2021-08-10T15:35:49:     stale: false,
2021-08-10T15:35:49:     compatible: true,
2021-08-10T15:35:49:     compatibilityError: null,
2021-08-10T15:35:49:     logicalSessionTimeoutMinutes: null,
2021-08-10T15:35:49:     heartbeatFrequencyMS: 10000,
2021-08-10T15:35:49:     localThresholdMS: 15,
2021-08-10T15:35:49:     commonWireVersion: 13
2021-08-10T15:35:49:   }
2021-08-10T15:35:49: }
```

bepstein111, 2021-08-11:
So I was able to un-bork my instance by downloading Robo 3T (since I couldn't figure out mongosh and didn't care to spend hours learning), connecting to the mongo database, navigating to octofarm>serversettings>ObjectId>influxExport, editing the document and changing the "active" value for influx from "true" to "false". It seems that with influxdb export disabled, there is no problem.
that being the case, I don't know that this is the same issue as mentioned above. If it is, why doesn't my `    depends_on:
      - "mongodb"` line in my compose file fix it, and further, why is it affected at all by the connection to influx?

davidzwa, 2021-08-12:
Your questions are correct.

@NotExpectedYet could you be so kind as to test InfluxDB on master or dev for a repro?

NotExpectedYet, 2021-08-12:
> You're correct to ask these questions.
> 
> > connecting to the mongo database, navigating to octofarm>serversettings>ObjectId>influxExport, editing the document and changing the "active" value for influx from "true" to "false". It seems that with influxdb export disabled, there is no problem.
> 
> For editing the mongo database I'd refer to Mongo Compass, although this is a last resort. https://docs.mongodb.com/compass/current/install/
> 
> > that being the case, I don't know that this is the same issue as mentioned above. If it is, why doesn't my depends_on: - "mongodb" line in my compose file fix it, and further, why is it affected at all by the connection to influx?
> 
> No idea to be fair. I run into the same issue where `depends_on` does not work properly. That's why we added the database issue page and why we've enabled the Restart OctoFarm button on the unreleased code of OctoFarm.
> 
> > Even further, if that were the solution, shouldn't restarting the octofarm container (after the whole stack has come up and tested successfully, even with influxdb export enabled) fix it?
> 
> Yes restarting **only** the OctoFarm container sould work if the MongoDB race condition was the problem (see #674). Do note: you've shared only the pm2/pm2 error logs, so for me its a guess whether the Mongoose Error spotted in that log is in the same moment as your influx related error. This is why I would highly suggest sharing the log dump and not just the pm2 log. This can be done on the system page.
> 
> @NotExpectedYet could you be so kind as to test InfluxDB on master or dev for a repro?

@davidzwa That cast error is a typing issue. Seems something has changed representation maybe in this version/before as I got rid of all those. Could be problematic going forward if people already have databases enabled but will double check.

NotExpectedYet, 2021-12-12:
I believe this is fixed for V1.2. 

It was a bug in filament manager plugin causing the database issue to be triggered...

commit 499a9850 (2021-12-12): fixed #733
closed 2022-05-19 by NotExpectedYet