OctoFarm/OctoFarm #746: Admin account not created on new install

gundown64, 2021-08-31:
### OctoFarm Version

Latest 1.1.13-hotfix

### OctoPrint Version(s)

1.5.2

### pm2 or docker?

pm2

### Operating System

Ubuntu 18.04.5

### Location of the problem

Web Interface

### Description of the problem

```bash
So I seem to have the exact same issue as this user did here (including the same browser console logs):
https://github.com/OctoFarm/OctoFarm/issues/298
However, it does not look like a great solution was found other than a complete reinstall of Ubuntu, which I am trying to avoid, and to "start with a fresh database" but I'm not sure how to do that.

I have no admin account upon creating the first user after a new install of OctoFarm. I was never given the little orange box that informed me that the account would be the admin account since it was the first one. I installed OctoFarm on a windows machine and did receive that prompt upon initial user creation.

I've tried removing everything (MongoDB, NodeJS and OctoFarm) and reinstalling it (a clean uninstall/reinstall of sorts), but the account that I created is never removed and I never get the option to create a new account with admin privileges. I probably did not uninstall the right things or do it properly.

Any tips on how to resolve this?
```

### Steps taken to reproduce

```bash
1. Installed following the guide: https://docs.octofarm.net/installation/install-linux-ubuntu.html
2. Upon account registration, no orange information box appeared telling me that this was the first account and, therefore, the admin account.
```

### What did you expect to happen

```bash
Expected to have admin account upon initial installation of OctoFarm
```

NotExpectedYet, 2021-09-01:
Hello

You don't have to re-install Ubuntu. But you will need a fresh database to try again. The easiest way is appending a "2" or something in your .env file for the database. 

When we were reproducing that issue it seems to be either caused by entering your user name with a capital and login in without or double clicking the sign up button causing two users to be created. 

You can check with mongo compass and check the users table. If you need help with that give us a shout. 

Just to clarify too the user warning is shown on registration of a first user not after a successful registration.

closed 2021-09-01 by gundown64

NotExpectedYet, 2021-09-02:
Welcome glad it was solved