OctoFarm/OctoFarm #747: Printer Status Getting Stuck

blooest, 2021-09-06:
### OctoFarm Version

1.1.13-hotfix

### OctoPrint Version(s)

1.6.1

### pm2 or docker?

docker

### Operating System

Unraid 6.9.2 OctoFarm host / OctoPi 0.17 printer host

### OctoPrint Plugins

bed visualizer, cost estimation, dashboard, display layer progress, exclude region, m73 progress, octolapse, print job history, printtimegenius, slicer thumbnails, spool manager, telegram, themeify

### Location of the problem

Not sure

### Description of the problem

```bash
After an indeterminate amount of time, the printer status in OctoFarm is getting stuck and no longer updating. This has resulted in OF stating that my printer has been in the "Completed" state for days on end while I've been printing, or in the "Active" state for days on end while I've not been printing, etc. Because of this, the History side of OctoFarm has also failed to capture numerous prints. I have 88 recorded in PrintJobHistory on the Pi in the same timeframe that OF has only captured 59.
```

### Steps taken to reproduce

```bash
1. Set up printer.
2. Wait?
3. Resync and it'll work again temporarily.
```

### What did you expect to happen

```bash
Printer status would continue to be updated and OctoFarm would stay in sync with what was actually happening.
```

### To avoid the problem I had to

```bash
Wish I knew!

I can't see any obvious signs of faults in the log. If I had to take a wild uneducated guess, some kind of update cron isn't running while the web UI isn't open, for some reason or another.
```

blooest, 2021-09-06:
[octofarm_dump.zip](https://github.com/OctoFarm/OctoFarm/files/7113114/octofarm_dump.zip)

NotExpectedYet, 2021-09-06:
Hello, 

It's hard to advise without knowing a little more about your setup. Is your network WiFi or Ethernet, do you shutdown your Pi's or keep them running all the time? 

How are those domains resolved? if you ping prusapi.localdomain from the OF host does it consistently keep responding? 

It's looking like network issues as: 
```
24/08/2021, 23:17:53 | ERROR | OctoFarm-State | Error attempting to connect to API: api/settings | http://prusapi.localdomain | timeout: 10000data:"\"request to http://prusapi.localdomain/api/settings failed, reason: connect EHOSTUNREACH 192.168.1.173:80\"" |
```
There's a lot of this, but if you've shutdown your Pi then the same error would be produced. 
```
25/08/2021, 02:52:50 | ERROR | OctoFarm-State | Error attempting to connect to API: api/settings | http://prusapi.localdomain | timeout: 1000data:"\"timeout\"" | 
```
There's a lot of these as well, but if the websocket can't keep a connection established that would be why your missing prints in your history. I can see it did capture some. If you could provide us with a bit more information we can try and advise / help further.

blooest, 2021-09-06:
Fiber to the server, ethernet to the Pi. No wifi involved. Both have been running continuously since I rebooted both of them ~12 days ago.

```
pi@prusapi:~ $ uptime
 23:49:34 up 12 days,  6:27,  1 user,  load average: 0.85, 0.97, 0.94
root@Blooest:~# uptime
 22:49:50 up 12 days,  8:05,  1 user,  load average: 0.58, 0.55, 0.52
```

That DNS reso is handled by my router -- a Unifi Dream Machine Pro.
The first error at least indicates it resolved to the proper IP before timing out when connecting to the Pi for..  some reason.
I do think those two might correspond to something I was doing at the time, might've rebooted the Octoprint service individually, I do see I was logged into the Unraid server via SSH at the time from its logs, but it shouldn't be repeating 

```
; <<>> DiG 9.11.5-P4 <<>> prusapi.localdomain
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 19818
;; flags: qr aa rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 4096
;; QUESTION SECTION:
;prusapi.localdomain.           IN      A

;; ANSWER SECTION:
prusapi.localdomain.    0       IN      A       192.168.1.173

;; Query time: 4 msec
;; SERVER: 192.168.1.1#53(192.168.1.1)
;; WHEN: Sun Sep 05 23:51:09 MDT 2021
;; MSG SIZE  rcvd: 64
```

Now that I'm looking at that State log, I don't see any such timeouts between the last time I resynced it on the 2nd and when I resynced it just now on the 6th, while it did lose tracking in that time.

NotExpectedYet, 2021-09-06:
Thanks for all that though that was very thorough! 

Those 2 packets loses are not really something that would cause a drop out on the websocket I don't think. Your network seems robust enough regardless, and even if there was a drop out the ping/pong check should restore the connection, but that's where it seems to be falling over. 

It's certainly the Ping/Pong that seems to be capturing a dead connection as there's a lot of "Error with websockets... resetting up!". When that's detected it currently tries to do a full re-scan. 

The re-scan happens and then causes the timeouts as it fails to get a response from /api/settings on OP. 

25/08/2021 - these are all way before this date though. Anything afterwards there doesn't seem to be any errors at all. 

Has it missed prints after this date? if it's before then I'd probably say whatever issue it was has resolved. If there's still stuff afterwards missing then I'm at a loss currently and there's nothing in the logs that would indicate what was going on. Yay! xD

blooest, 2021-09-06:
Unfortunately, yes. As I mentioned, I resynced on the 2nd, got a couple of prints out of it, and then it dropped again.
[image]
[image]
(PJH has a number more prints above this before we get to the spiralslider one that matches after my resync, but you get the gist.)
Just now noticed OF is displaying in PDT and OctoPrint is in my actual MDT, so there's an hour difference in the timestamps here, but that _should_ be of no consequence. The point here is that OctoFarm lost track sometime before that "VR" print finished.

The overnight ping test got to 37512 before I opted to cancel it, so here's those stats for completeness. Still nothing out of spec, but OF still hasn't desynced again yet either.
```
--- prusapi.localdomain ping statistics ---
37512 packets transmitted, 37507 received, 0.0133291% packet loss, time 37599919ms
rtt min/avg/max/mdev = 0.247/1.559/20.848/3.251 ms
```
Guess I should start looking at the OctoPrint log, but that rolled over just an hour prior to that "spiralslider" print. No related errors in the log there, it sees the client connection being open when I go for the resync;
```
2021-09-05 20:58:53,594 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.59.2
2021-09-05 20:58:58,198 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2021-09-05 20:59:01,931 - octoprint.server.util.flask - INFO - Passively logging in user blooskies from ::ffff:192.168.59.2
2021-09-05 20:59:01,934 - octoprint.access.users - INFO - Cleaning up user session 079E92CAD15D4B90843F7A592C1ADDF3 for user blooskies
2021-09-05 20:59:01,935 - octoprint.access.users - INFO - Logged out user: blooskies
2021-09-05 20:59:01,935 - octoprint.access.users - INFO - Cleaning up user session F9A21FA5BB504ADD964E5D942E663B3E for user blooskies
2021-09-05 20:59:01,936 - octoprint.access.users - INFO - Logged out user: blooskies
2021-09-05 20:59:01,936 - octoprint.access.users - INFO - Cleaning up user session 4BC239EF304E4BCD94E333BD802B0946 for user blooskies
2021-09-05 20:59:01,941 - octoprint.access.users - INFO - Logged out user: blooskies
2021-09-05 20:59:01,993 - octoprint.access.users - INFO - Cleaning up user session DD0CF9C53FA2430DABF9D5FA932176BB for user blooskies
2021-09-05 20:59:01,995 - octoprint.access.users - INFO - Logged out user: blooskies
2021-09-05 20:59:01,997 - octoprint.access.users - INFO - Logged in user: blooskies
2021-09-05 20:59:08,121 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
2021-09-05 20:59:08,635 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.59.2
2021-09-05 20:59:08,838 - octoprint.server.util.sockjs - INFO - User blooskies logged in on the socket from client ::ffff:192.168.59.2
```
(For the sake of information, I have the 192.168.59.0/24 range configured as a VLAN for NAS infrastructure on my network, while the Pi is just sitting on the normal 192.168.1.0/24 main untagged network. The server running OctoFarm is of course 192.168.59.2. There are no firewall rules limiting packet exchange between these networks, it's purely organizational.)

I'll have to monitor it and try to catch things when they've just gone out of step. I'll update you before I resync it next time, but is there anything I should do, any kind of further debugging options that I might want to enable so I have more information to give you?

blooest, 2021-09-08:
Well, it happened again, and I wasn't quick enough to catch it in the bad state before OctoPrint's log rolled over again. I do have this from turning on debug level logging in sockjs. I don't know enough about websockets or sockjs or anything to be qualified on what this means, but I'm imagining this points to a plugin being problematic.
```
2021-09-08 00:43:40,843 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: event
2021-09-08 00:43:40,850 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:40,867 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:41,706 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:41,711 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:41,733 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:41,740 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:42,647 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:42,817 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:44,483 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:44,521 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:45,832 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
2021-09-08 00:43:46,274 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
```
I just have no idea which one.

```
2021-09-07 17:09:39,011 - octoprint.plugin.core - INFO - 30 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (1.1.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Cost Estimation (3.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_costestimation
|  Dashboard (1.18.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.26.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Exclude Region (0.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_excluderegion
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.8.11) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  M73 Progress Plugin (0.2.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_m73progress
|  Octolapse (0.4.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse
|  Pi Support Plugin (2021.8.2) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  PrintJobHistory (1.14.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory
|  PrintTimeGenius Plugin (2.2.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
|  Slicer Thumbnails (1.0.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prusaslicerthumbnails
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  SpoolManager Plugin (1.4.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager
|  Telegram Notifications (1.6.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram
|  Themeify (1.2.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_themeify
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
```
From the information I presume you collect from past issues/whatnot, are any of these known to be incompatible? Or known to be compatible so I can rule them out, for that matter?

NotExpectedYet, 2021-09-08:
None look like any I would personally pin this on. I am yet to actually pin a connection issue on a specific plugin. I can tell you which I have never used if it helps narrow it down for you. 

Dashboard (sure I've other users with this one), Display Layer Progress (sure I've other users with this one), Exclude region, M73 Progress Plugin, OctoLapse,  PrintJobHistory (I'd try this first personally, wonder if it's getting in the events way, absolute guess), Spool manager Plugin. I use the rest personally so wouldn't have said it was from them. I've never seen that "Throwing message away" personally, that's a new one aha. Be nice if it specified which one did it.

blooest, 2021-09-08:
Well, that only shows up if enough messages get held back for other reasons. If you've been causing socket failures on the Octoprint side, you've probably been quick about sorting it out. So it makes sense you wouldn't see that in particular. Would be nice if the handler would show you the full message it's holding back/throwing away though, yeah.

So I'm a dolt and didn't think about the probability that OctoPrint kept rotated logs for a time, which it sure does. Thus, I have the beginning of the backlog that caused it this time in sight.
```
2021-09-07 14:12:47,312 - octoprint.server.util.flask - INFO - Passively logging in user blooskies from ::ffff:192.168.1.39
2021-09-07 14:12:47,314 - octoprint.access.users - INFO - Cleaning up user session 8008DA1D97EA4DAAB3321B65B208D3B1 for user blooskies
2021-09-07 14:12:47,315 - octoprint.access.users - INFO - Logged out user: blooskies
2021-09-07 14:12:47,315 - octoprint.access.users - INFO - Cleaning up user session 54CCB019EC084EFABC07F5B4C8A81FA3 for user blooskies
2021-09-07 14:12:47,316 - octoprint.server.util.sockjs - INFO - User blooskies logged out, logging out on socket
2021-09-07 14:12:47,322 - octoprint.access.users - INFO - Logged out user: blooskies
2021-09-07 14:12:47,323 - octoprint.access.users - INFO - Cleaning up user session 277668309B2647BC968582CAFD55D5C1 for user blooskies
2021-09-07 14:12:47,323 - octoprint.access.users - INFO - Logged out user: blooskies
2021-09-07 14:12:47,324 - octoprint.access.users - INFO - Logged in user: blooskies
2021-09-07 14:12:48,227 - octoprint.server.util.sockjs - DEBUG - Socket message held back until permissions cleared, added to backlog: plugin
2021-09-07 14:12:48,625 - octoprint.server.util.flask - INFO - Passively logging in user blooskies from ::ffff:192.168.1.39
2021-09-07 14:12:48,626 - octoprint.access.users - INFO - Logged in user: blooskies
2021-09-07 14:12:49,900 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.39
2021-09-07 14:12:50,010 - octoprint.server.util.sockjs - DEBUG - Socket message held back until permissions cleared, added to backlog: event
```
The backlog is getting filled by plugin message spam, sure, but I can't confirm it's specifically a plugin that's causing the holdback in the first place. Back to square one.
That's a login from my PC and not the server, but after this point it's just a lot of messages "held back until permissions cleared", whatever that means, until the backlog is filled. Then it throws them away. With the plugin messages, that only took 5 minutes.
```
2021-09-07 14:17:05,047 - octoprint.server.util.sockjs - DEBUG - Socket message held back until permissions cleared, added to backlog: plugin
2021-09-07 14:17:08,056 - octoprint.server.util.sockjs - DEBUG - Socket message held back, but backlog full. Throwing message away: plugin
```
I think I finally found the actual culprit though, and it sure seems to have nothing to do with you, or perhaps even a plugin. Got this error when I reconnected my web interface just now.
```
2021-09-08 13:51:01,736 - octoprint.server.util.sockjs - DEBUG - Sending 100 messages on the socket that were held back
2021-09-08 13:51:01,947 - octoprint.server.util.sockjs - ERROR - Could not send message to client ::ffff:192.168.1.39
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/sockjs.py", line 590, in _do_emit
    self.send({type: payload})
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/sockjs/tornado/conn.py", line 51, in send
    self.session.send_message(message, binary=binary)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/sockjs.py", line 69, in send_message
    json_dump(octoprint.vendor.sockjs.tornado.util.bytes_to_str(msg)),
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/json/__init__.py", line 25, in dump
    allow_nan=False,
  File "/usr/lib/python3.7/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/usr/lib/python3.7/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.7/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/json/__init__.py", line 49, in encode
    raise TypeError
TypeError
```
I did have to manually update to Python3 because of the OctoPi 0.17 install, although I used the "official" script to do so... so I guess I need to export and reimage on 0.18 to see if that sorts it.
It's probably safe to close this out as not OctoFarm related, but it may be prudent for me to keep you updated, since the only symptom of this to me as an end user was on OctoFarm. Knowing the full outcome would be good info for you down the line as Octoprint starts nagging for/enforcing Python 3 in 1.7.0+. If this is a common issue with Py2->Py3 migrated installs, you'll probably see it again.

NotExpectedYet, 2021-09-10:
Could you send through your later state logs please? Wonder if anythings revealed itself this time.

blooest, 2021-09-10:
Absolutely. Doesn't look too helpful to me though. I see a number of errors but they all seem related to the several OctoPrint restarts I had to do, or otherwise not related to the websocket.
[octofarm_dump-2021-09-10.zip](https://github.com/OctoFarm/OctoFarm/files/7142972/octofarm_dump-2021-09-10.zip)

08/09/2021, 22:16:37 is when I reconnected after the reimage to 0.18. I watched the octoprint logs religiously and removed the SpoolManager plugin around 22:37 (causing an Octoprint restart and thus, errors in your connection) because I was already getting hangups in Tornado that coincided with database tests/connections it was running.
I installed the old FilamentManager at the same time because I didn't want to lose the precise-ish measures I had, since it seems like that's known to be stable.

I do see this error, not sure what it's referring to exactly but I'm presuming it's from enabling the OF-FilamentManager sync and nothing noteworthy. Didn't interrupt the websocket sync immediately, at any rate.
```
09/09/2021, 01:09:00 | INFO | OctoFarm-State | PrintDone0: ws://prusapi.localdomain/sockjs/websocket
09/09/2021, 01:09:00 | INFO | OctoFarm-State | PrintDone0: ws://prusapi.localdomain/sockjs/websocket
09/09/2021, 01:25:46 | ERROR | OctoFarm-State | Couldn't find profiledata:{"stringValue":"\"14\"","kind":"ObjectId","value":14,"path":"_id","reason":{}} | 
09/09/2021, 01:52:47 | INFO | OctoFarm-State | PrintDone0: ws://prusapi.localdomain/sockjs/websocket
09/09/2021, 01:52:47 | INFO | OctoFarm-State | PrintDone0: ws://prusapi.localdomain/sockjs/websocket
```
This desync happened while it was cooling down after that print at 1:52 finished, not printing, and I caught it pretty quickly when I looked at OctoFarm -- the hotend temperature was frozen at 110. Didn't lose any actual PrintDone events/print jobs for the history as a result.

Also interesting that it was double-capturing the PrintDone events there. It didn't do that before the reimage, and I rebooted OctoPrint at ~02:34, after the print where I noticed the desync finished, to test removing Octolapse (as it was the "unknown" plugin I cared the least about/with the least impact to my production environment) and it hasn't done that since then either.
It also hasn't desynced again since then. Weird to think Octolapse might be (part of) the cause, but we'll see how this goes.

blooest, 2021-09-19:
Well, it's been a minute, haven't been printing much because I'm troubleshooting an unrelated issue with the printer, but it desynced again, so that's PrintJobHistory, SpoolManager, and OctoLapse all cleared of fault. The rest of the plugins you listed off, I've personally used before without having this issue on a previous, non-docker install of OctoFarm. So I'm well out of ideas. The Printer page in OctoFarm shows the pings/pongs going out/coming in without errors, so I'm still inclined to think it's something on OctoPrint's side, but I just have no idea what.

NotExpectedYet, 2021-09-25:
> Well, it's been a minute, haven't been printing much because I'm troubleshooting an unrelated issue with the printer, but it desynced again, so that's PrintJobHistory, SpoolManager, and OctoLapse [...]

No same, out of interest have you tried with a ip address rather than local domain? 

The OctoFarm state log is just saying it couldn't find the printer profile for the filament you had selected by the looks, that wouldn't cause a disconnect though I don't think... or shouldn't. You could try disabling the filament manager plugin in OF to see if it's that causing it. 

the double capture I bet is a connection not getting shutdown after a re-sync or loss of connection. That is certainly a bug and has been worked on in RC-2 currently.

blooest, 2021-10-04:
I hadn't. I also hadn't been printing anything in the past week because I got frustrated at being unable to fix the unrelated issue I mentioned in the last update.

I just switched it over to the IP, but I don't set up DHCP reservations for my IPs so, however unlikely because the Pi runs 24/7, it _could_ change. I'm hoping it's not the DNS, I can't imagine how it could be, but it is a variable that needs removing...

NotExpectedYet, 2021-10-17:
> I hadn't. I also hadn't been printing anything in the past week because I got frustrated at being unable to fix the unrelated issue I mentioned in the last update.
>
> I just switched it over to the [...]

Hello! Did we have anymore drops after changing from DNS to IP?

closed 2022-05-19 by NotExpectedYet