OctoFarm/OctoFarm #751: OctoFarm gets progressively sluggish with use, and must be sudo reboot every few hours.

ndarrow, 2021-09-20:
### OctoFarm Version

1.1.13-hotfix

### OctoPrint Version(s)

1.6.1

### pm2 or docker?

pm2

### Operating System

Raspberry Pi

### Location of the problem

Not sure

### Description of the problem

```bash
...
Hardware: Raspberry Pi 4- running on SD card with heat sinks and fan cooling.  

This has been a problem since I first started using OctoFarm (about 7 months ago, not sure which version that was). 
The longer OctoFarm is online, the slower the UI gets. All screens, updating temps, updating printer status in List view, lag when pressing buttons ("Play/print" button in list view), loading the "Control your printer" screen...everything in the UI lags to the point that OctoFarm eventually freezes and won't successfully reload. 

The lag seems to happen faster with more use. ie. restarting my 20 printers twice in one day sees slower lag propagation than restarting 20 printers 10 times in a day. 

SSH > sudo reboot restores speed immediately after reboot. 
Restarting OctoFarm via UI> System > Server > Restart button, does not fix the problem. 

No matter what, I have to restart at least once every 24 hours (low OctoFarm usage), but as frequent as once every 2 hours (high OctoFarm usage).

I have the log dump, not sure how to upload to the GitHub Issue Report page...
```

### Steps taken to reproduce

```bash
1. I just use it as normal and the lag creeps in.
2. More OctoFarm use = more lag. 
3. The amount of time OctoFarm is active also causes lag. 
...
```

### What did you expect to happen

```bash
Constant speed
```

### To avoid the problem I had to

```bash
sudo reboot via ssh
```

ndarrow, 2021-09-20:
[octofarm_dump.zip](https://github.com/OctoFarm/OctoFarm/files/7199208/octofarm_dump.zip)

[image]

NotExpectedYet, 2021-09-25:
Hmm there's nothing instantly jumps out at me in your logs. 

Could you tell me a little bit more about your setup? 
Like is there any Wifi?
what's running on the PC your access OF from?
Does the slow down persist on another client at the same time? 
Are there camera's in the setup? 
Anything else running on the OF Pi? 

What do you mean by restarting your printer? Is that a "Re-Sync" in OF, or actually restarting OP? 

Just want to try and rule some stuff out, I don't personally use it on a Pi but I've not experienced any slow down in ages. With restarting just OF not seeming to fix it I'd be leaning towards it been something with the Pi your running OF on but that's just a complete guess and if your screenshot of the CPU is from when the slow down is happening it doesn't look to be anything with OF chewing up resources. Have you tried with a different SD card to rule that out?

ndarrow, 2021-09-29:
**Like is there any Wifi?**
-I have 20 Pi 3B+'s running only Octoprint, connected to 20 printers. All PI's are on the same WiFi network (21 Pi's in total, plus whatever client device I'm using). 

**what's running on the PC your access OF from?**
-Normally, it is my home laptop on Chrome. The slow down is present on any device I connect with (tablet- Silk, phone- Safari, a different laptop- Firefox). The speed at the time is consistent across all connected devices. If OF has just been reset, lightning fast on whatever device I'm using. Once it slows down, its slow on all devices. I normally only have one device connected though. So that rules out browser or access device issues.

**Does the slow down persist on another client at the same time?**
-Yes, if 2 devices are connected at the same time, OF speed is the same on both. 

**Are there camera's in the setup?**
-I currently have 20 cameras connected, but the problem existed prior to the cameras. 

**Anything else running on the OF Pi?**
-I set the OF Pi up as detailed in the instructions, so no, I did not add any other operations or tasks. 

**What do you mean by restarting your printer? Is that a "Re-Sync" in OF, or actually restarting OP?**
-I mean pressing Print. I get all the printers printing, they finish, I harvest parts, then start printing again. No power cycles or resyncing, just restarting the print jobs.

**Have you tried with a different SD card to rule that out?**
-I have. I tried several different SD cards earlier on in my adventures with OF, and most recently I used a new SD card when I updated to 1.1.13-hotfix. 

I'll try to get a log dump at the latest possible moment before it slows down so much it crashes.  
I'm going to try booting from USB 3.0 with an SSD instead of an SD card next, when I get a chance to shut the print farm down. 
https://www.youtube.com/watch?v=r27WcPRtpWM

I have 14 more printers/pi's that need to get added to OF, but I don't really want to add more until I get this fixed. 

I'll send another log dump if I can sang one, and let you know how the boot from SSD works out. Let me know if there is anything else you think I can try. 

Thank you

NotExpectedYet, 2021-09-30:
Epic thanks the thorough reply! 

I think some htop outputs might help but it's certainly sounding like OF from your description. Just a little confused as to what specifically it could be. Im wondering if the current code base is overloading the pi, or there's some bigger goof in the code at work, it could also be overloading/saturating your WiFi but you mentioning it been like it before the cameras should rule that out. 

I'd be interested to see if the latest RC made any difference. You would need to try it on a seperate database so you may roll back if required as there are some database migrations that make rolling back difficult. If you need some instructions let me know, that version is already substantially lighter than .13-hotfix it's just not really ready for prime time. 

I will get back to you with a proper answer closer to the weekend whilst I have a think how to figure it out.

ndarrow, 2021-10-01:
[octofarm_dump Nick D 9-30-21 bug #751.zip](https://github.com/OctoFarm/OctoFarm/files/7263366/octofarm_dump.Nick.D.9-30-21.bug.751.zip)

I grabbed this a few minutes before it stopped responding.

NotExpectedYet, 2021-10-01:
Looking at the state log there's a lot of EHOSTUNREACH I gather these are switched off? 

There's also a lot of file information scans taking too long to generate. 

What Wifi setup do you have there? decent router or home router?

I'm wondering if the network is just getting overloaded with all the connections / requests, would make sense to why a reboot on the OF device fixes things as the requests would reset, it could also be OF just not handling that amount of files very well but you'd more than likely see a large spike in CPU usage with that scenario. 

I had a similar scenario with a user where he was wired with everything but had a VPN running on the OF device which in effect doubled all the connections up. 

Is the OF device wired or wireless? 

If you can, I'd test with the OF device wired. As far as I know the Pi4 should handle the amount of printers but I had one bloke lay cat6e everywhere because his farm was doing similar with his own Wifi. He was at 100 printers last I checked in with him. 

Sorry for all the questions like, just a bit unsure really and whilst I've got time before I start working on the next release just want to make sure we've ruled everything we can out.

ndarrow, 2021-10-01:
Soooo, I goofed a little on one part of my description. The Pi4 running OF is wired to the router. Sorry, I forgot I had done that. 
The router is a Nighthawk R6700 AC1750.
I just counted the total number of devices connected to it on _both_ bands (not just the 2.4), and there's 34... This router is meant for up to 25. 
[image]

I think there are enough grey areas I need to correct before you spend any more time on this. 

1. Get a better router that can handle more devices, and isolate OF and OP devices on their own router.
2. Switch to latest OF.
3. Run OF from SSD via USB 3.0.
4. Remove all cameras. 
5. Capture logs along the way. 
6. Report back here. 

This might take me some time to accomplish. Trying to find a good time to shut down operations has been difficult lately. 

Hey, thank you so much for your support. I really appreciate it. I have a clear direction to go now. 
I probably should have done some more homework on my own before jumping to conclusions. 
And thanks for the sticker!

NotExpectedYet, 2021-10-01:
You are totally welcome and it's no trouble, I could have been more clear with those initial questions. 

Hmm, that device limit might explain some issues I'm having at work thanks for pointing that out 😅. 

I think the cameras shouldn't need to be removed, I think the issue is the connection amount not the bandwidth usage. Mjpeg streams ain't massive unless your doing a me with 1080p streams everywhere. 5 of those took 30Mb/s! 

No rush at all, and heck there are certainly things OF could be doing better in these situations so it's good feedback to get. 

You seem to have a good plan going forward so I'll keep this open until you've resolved it because I can add network requirements/suggestions to the docs page :) good luck with it all and hopefully it resolves your issue!

ldiegos, 2021-10-03:
Hi,
Same issue on a vmware virtual machine with windows 10 prof installed in a vmware sphere server.
After an hour of octofarm working, the UI starts to be lazy and unresponsive, i must reboot octofarm several times if i want to see the progress of the printers.

VMware sphere server:
intel NUC i5 3.6GhZ
RAM: 32Gb
Ethernet wired.

OF VMware machine
Cores: 4
RAM: 4Gb
Ethernet wired.
Version: 1.1.13-hotfix
Connection to posgreSQL on other VMWare machine to share the filament manager information.

Printing farm:
4 raspi 3B+ wired to the network using the posgreSQL without problems.

CPU info:
[image]

Dump files, 5min after restart:
[octofarm_dump20211003-0640.zip](https://github.com/OctoFarm/OctoFarm/files/7272727/octofarm_dump20211003-0640.zip)
OF state:
[image]

Dump files, 1h after restart:

OF State:

NotExpectedYet, 2021-10-03:
> 
> 
> Hi, Same issue on a vmware virtual machine with windows 10 prof installed in a vmware sphere server. After an hour of octofarm working, the UI starts to be lazy and unresponsive, i must reboot octofarm several times if i want to see the progress of the printers.
> 
> VMware sphere server: intel NUC i5 3.6GhZ RAM: 32Gb Ethernet wired.
> 
> OF VMware machine Cores: 4 RAM: 4Gb Ethernet wired. Version: 1.1.13-hotfix Connection to posgreSQL on other VMWare machine to share the filament manager information.
> 
> Printing farm: 4 raspi 3B+ wired to the network using the posgreSQL without problems.
> 
> CPU info: [image]
> 
> Dump files, 5min after restart: [octofarm_dump20211003-0640.zip](https://github.com/OctoFarm/OctoFarm/files/7272727/octofarm_dump20211003-0640.zip) OF state: [image]
> 
> Dump files, 1h after restart: [octofarm_dump20211003-0736.zip](https://github.com/OctoFarm/OctoFarm/files/7272766/octofarm_dump20211003-0736.zip) OF State [image]
> 
> :

What printer count are you at out of interest? 

Thanks for the additional information will take a look too. Might not be network as your setup rules that out pretty much.

ldiegos, 2021-10-03:
> Only 4 printers too.
> Thanks for the additional information will take a look too.
> Might not be network as your setup rules that out anything WiFi related.

Do you want more logs? is not a problem to continue dumping log files, during a few hours if you want.

Maybe a little video to feel the responsiveness?

NotExpectedYet, 2021-10-03:
> Do you want more logs? is not a problem to continue dumping log files, during a few hours if you want.
>
> Maybe a little video to feel the responsiveness?

The logs won't really show what is causing the slow down sadly, I checked through yours and the previous users. 

Do you have the ability to test out the latest RC to see if it's any better? 

A video wouldn't go a miss though :)

ldiegos, 2021-10-03:
> The logs won't really show what is causing the slow down sadly, I checked through yours and the previous users.
>
> Do you have the ability to test out the latest RC to see if it's any better?
>
> A video [...]

Ok, perfect, let me download the RC and prepare two short videos, one with OF just started and the other being lazy a few hours later.

Thank you so much @NotExpectedYet

NotExpectedYet, 2021-10-03:
> Ok, perfect, let me download the RC and prepare two short videos, one with OF just started and the other being lazy a few hours later.
>
> Thank you so much @NotExpectedYet

No trouble! Just bare in mind there's some db migrations so would be better with a fresh database for the RC test. 

Thanks for the additional info, hopefully will find the cause.

ldiegos, 2021-10-03:
No prob, i will backup the mongoDB and start from scratch. Is not an issue for me, OF helps a lot, but my life do not depend on the saved data lol.

sstoiana, 2021-11-01:
Seeing this on my farmpi as well. Only three octopi instances connected on a Raspberry Pi 3B+. 

```
Current Version:
1.1.13-hotfix
```

```
pi@farmpi:~$ node --version
v14.17.2
```

I just rebooted the Pi and will keep an eye on cpu and see what might be triggering it. 

```
 System Uptime:
2h, 31m, 27s

 OctoFarm Uptime:
2h, 31m, 10s
```

sstoiana, 2021-11-04:
Something is definitely off. Been monitoring node's V8 VM for the past 24 hours using NewRelic's agent, and it looks like there is more GC happening over time. GC pause duration and frequency are increasing over time. The event loop is getting slower too.

Most of the memory ballooning is non-heap. Makes me think there might be some sort of resource leak (connections not closed properly perhaps?). This is happening with only 3 Octoprint instances connected on a Rasppbery Pi 3B+.

[image]

NotExpectedYet, 2021-11-05:
> Something is definitely off. Been monitoring node's V8 VM for the past 24 hours using NewRelic's agent, and it looks like there is more GC happening over time. GC pause duration and frequency are [...]

You could be right with the connections not closing, Done a lot of clean up for 1.2 in that regard. Thanks for the additional info! I would "think" that should occur with printers going online and offline, but I'll make sure to double check re-connections as well and make sure they're properly closing. 

I am hoping to get a stable 1.2 RC-2 at some point soon, just had a lot on with work. So fingers crossed it squashes all these issues. 

Bring on 2.0 when I can get rid of my early learning coding goofs!

sstoiana, 2021-11-10:
Been testing branch bugfix/of-systems for a day, and it's been surprisingly stable in terms of GC, memory and event metrics. If anything, non-heap memory dropped significantly  around 5:40am. Octofarm did stop communicating with the Octoprint instances some time during the night and needed a re-sync of the printers. I suspect that's a different bug lurking in this branch.

As far as I can tell, the slowness is fixed on this bugfix/of-systems branch.

[image]

NotExpectedYet, 2021-11-10:
> Been testing branch bugfix/of-systems for a day, and it's been surprisingly stable in terms of GC, memory and event metrics. If anything, non-heap memory dropped significantly around 5:40am. [...]

Brilliant, thanks for the response! So good to know it's licked.

I'll double check the re-sync issue sounds like #747 possibly. Would you mind uploading/sending me some state logs to that ticket? Or over email is fine as it contains API keys/IP's. There's a few reconnect situations I decided to stop trying to re-poll from and I'm just wondering if I goofed and was a bit too stringent with that.

ndarrow, 2021-11-28:
I just saw this (wasn't getting notifications for some reason), thanks for the time and effort. Looking forward to V1.2!

NotExpectedYet, 2021-11-29:
No need to close I'll leave it open until 1.2 is released and it's squashed :)

ndarrow, 2022-01-26:
FYI, my latest OF crash has not recovered. The UI is not accessible but I can SSH to the PI. I know it could be a corrupt SD card or probably a million other things, but if I can read the SD card- would it be useful to grab logs from somewhere before I try to reimage the card?

NotExpectedYet, 2022-01-26:
@ndarrow certainly!

ndarrow, 2022-01-27:
How do I get logs of the card? I only knew how to hit the log dump button in the UI... I went through the viewable files via windows explorer (USB), but I didn't find anything "log" related. I assume they are on the unreadable partition- and can only be accessed via SSH... No idea how to do that though.

NotExpectedYet, 2022-01-27:
How did you install OctoFarm? Was it FarmPi or manual? 

The partition would need to be read on a linux pc or windows PC with ext3/4 support I think... I honestly forget what they partition them as. 

It sounds like your PC is windows, I personally use this. https://github.com/billziss-gh/sshfs-win

ldiegos, 2022-03-09:
Hi,
I want to write again the last issues with this, the last version, instead of get sluggish, actually crash the machine.
When you loggin in(from another computer to the web page), the octofarm starts to create powershell instances with a 35Mb consumption untill the machine crash.

Octofarm initial start:
[image]

Octofarm when loggin:
[image]

Powershell starting to act like a virus and consumption of processor at 99%
[image]

Powershell continue "cloning itself"
[image]

Memory consumption increasing till death
[image]

NotExpectedYet, 2022-03-10:
That seems like something with pm2, what do you mean last version? Still 1.1.13-bugfix to confirm? 

Or have you updated packages? 

I took a break for my birthday at the moment but will be back Saturday.

ldiegos, 2022-03-10:
> That seems like something with pm2, what do you mean last version? Still 1.1.13-bugfix to confirm?
> 
> Or have you updated packages?
> 
> I took a break for my birthday at the moment but will be back Saturday.

Yes.. 1.1.13-bugfix but with the latest updates to the git repository

I make a git pull every time that i start octofarm.

NotExpectedYet, 2022-03-10:
> > That seems like something with pm2, what do you mean last version? Still 1.1.13-bugfix to confirm?
> > Or have you updated packages?
> > I took a break for my birthday at the moment but will be back Saturday.
> 
> Yes.. 1.1.13-bugfix but with the latest updates to the git repository. I make a git pull every time that i start octofarm.
> 
> And also npm warn my about some critical issues and i update it too

Weird the only changes were a README change. Thanks for the confirmation though.

I'm probably going to work on a V1.1.14 to resolve these issues because V1.2 is taking so much time at the minute. 

It's not recommended to update npm because it has a chance of making breaking changes with packages. I can only gather something has changed with pm2 dependency. Would you mind passing me the package.json file so I can double check the versions? 

I'll update here once I do as it may take a bit of force pulling to get your version back inline.

NotExpectedYet, 2022-03-10:
> > Would you mind passing me the package.json file so I can double check the versions?
> 
> Do you mean the one that is in the root folder? /Octofarm/packages.json Here you are: [package.zip](https://github.com/OctoFarm/OctoFarm/files/8225947/package.zip)
> 
> Is there anything that i can do to continue using Octofarm? any previous version?

If you copy out your images folder (will save anything grabbed in history like thumbnails and camera's, if you didn't use those no need)
Nuke the old OctoFarm fodler
Reclone and re-run the setup you should be fine. 
Just use the same database url you used previously.

ldiegos, 2022-03-20:
Good morning,
I downgraded to version 1.1.12 it seems to get sluggish later than the 1.1.13-bugfix, but still getting sluggish... 

Waiting that 1.1.14 to give it a try!
Luis.

closed 2022-05-19 by NotExpectedYet