OctoFarm/OctoFarm #759: OctoFarm on Windows, doesn't seem to integrate with MongoDB

KalixtShawxo, 2021-10-12:
### OctoFarm Version

1.1.13

### OctoPrint Version(s)

1.70

### pm2 or docker?

pm2

### Operating System

Windows 10

### Location of the problem

Not sure

### Description of the problem

```bash
Installed NodeJS on Windows, seems to go ok. Then installed MongoDB, seems to install ok.
Then when I do the NPM Install it completes with a bunch of vulnerabilities. 
This is where I am lost. I look and there is no CONFIG directory.

I am following the instructions from "Chris' Basement" video
https://www.youtube.com/watch?v=9U-QTOmx49c
```

### Steps taken to reproduce

```bash
1. uninstalled everything
2. reinstalled everything
3.
...
```

### What did you expect to happen

```bash
...
I expected to be able to run OctoFarm
```

### To avoid the problem I had to

```bash
...
```

KalixtShawxo, 2021-10-13:
ok, so here's my "fix" after multiple installs, uninstalls, reinstalls, and they all fail. 
I decided to try something different and installed an older version of MongoDB. 
I had been using 5.03 (latest) and I instead tried it with 4.4.9 and did a clean install of 
Node.JS, then  MongoDB, then OctoFarm, and now there was no "database Issue" and I am setting up the software now. 

Hope this information is helpful.

NotExpectedYet, 2021-10-16:
All I'm seeing are timeouts, this could be network issues or I sometime's see it happen when OP is printing and it takes a bit to respond. It's the first scan into the system so it won't let it proceed. 

You'd have to check the OP logs to see if the connections are going through at that end.

KalixtShawxo, 2021-10-16:
well, thanks for looking into it. I'm at a loss, I don't see how it'd be a network issue, I mean they are all plugged into the same switch. I'll look into the logs, and I'm setting up a second OP Pi, so I will have two for it to try to connect with.

NotExpectedYet, 2021-10-16:
It's certainly something causing the response not to come back to OF, network is just my first guess. 

Feel free to post the logs and I can take a look as well as your correct, wired shouldn't be taking any time at all to respond. My wired network fires back before the first time out of a 1000ms, yours isn't getting a response in 10000ms which is the fall back after the first fail.

The only other time I've seen this is with a weird plugin, but me and the user could never figure out which specific plugin it was. You can rule that out by disabling all the non-bundled ones or attempting with a fresh install.

NotExpectedYet, 2021-10-16:
> 
> 
> I'll check around some more.... it is really weird. one of the first things I did when I kept getting stymied was disabled most of the plugins, looking for something causing a conflict.
> 
> We'll see what happens when I do the second OP Pi.
> 
> Thanks again.

Most welcome. 

Yeah it's weird like, I've had a user with a dodgy SD card produce a similar timeout issue as well thinking about it. Funny thing was it was running OP for a while totally fine, they changed it on the off chance and it started working after that. He'd re-installed a few times on the original card before switching it too. Just an example of the weirdness, not saying it's the issue here.

KalixtShawxo, 2021-10-17:
> One thing to double check.
>
> There's a settings cog on the printer manager table, if you open it, it will show the user OF is trying to connect with.

I'm confused now. I can't see that screen . I can't get into anything in octofarm from the web, it is stuck at the "database issue". 

It still seems that although of is running, it still can't connect to the mongodb. Even though the string is formatted correctly. or so it seems.
Question, shouldn't I be able to log into mongodb compass with the "mongodb://127.0.0.1:27017/octofarm" ?It gives an error as well.

NotExpectedYet, 2021-10-17:
Yeah mongo compass should let you in. That sounds like there's an issue with mongodb starting. I don't personally use it on windows apart from developing but it should just start up the service on boot. 

That database page is a bit of a pain tbh, it can be triggered just because mongodb is currently booting and I'm trying to figure out a way to "wait" for it... 

Does it allow a connection from that page to succede?

KalixtShawxo, 2021-10-17:
> When your on, the settings page I showed is from this cog button on printer manager.

no, i can not do anything in compass, or from the octofarm webpage. i can wait hours after starting and nothing. OF sees the database program exists bug fails to connect. compass fails to connect. how can i verify the db has started? i'm in uncharted territory here. could the whole problem be that simple that the db s failing to start?

NotExpectedYet, 2021-10-17:
It sounds like it with compass not been able to check. 

Hmm, let me see if I can remember windows ahaha... not my most knowledgeable platform. 

MongoDB on my system isn't in CTRL + ALT+ DEL and then startup.

It's service should be running though:
[image]
If you right click there, and press open services. 

Then find the MongoDB Server (MongoDB) service and double check it's running and has start up type as Automatic. 
[image]

NotExpectedYet, 2021-10-17:
I'd also make sure there isn't two, as you'd previously installed 5.0.

KalixtShawxo, 2021-10-17:
well i went into windows services, found mongodb and it was set to automatic but not running. i manually set it to start and then went back to the of webpage...
lo and behold, test and no error.  i feel kinda dumb, but still no idea why it failed to start on its own. saved settings, did restart, refreshed webpage,and after deveral failed pw attempts, i am now in again.
update after i can see if it sees the printers now.

KalixtShawxo, 2021-10-17:
> Sweet one down!
>
> Yeah I'm not entirely sure myself why it didn't start. My usual go to is to just shout windose into the ether.

ok. one pi found and online, the one without a printer attached. its being delivered tomorrow. but i preconfigured an octopi for it. the original one still showed offline. so i blew it all out, created new api and re configured OF and now it sees them both. 
so, two problems down. 
but. both shoe under printer state "API issues detected" in the printer manager. so i went back to the dashboard and if shows one active printer, one disconnected printerand the active printer says it is over 51%  progress. 
so, it is working, but still have the API issues detected. 
its after 5am here, i think this id a good break point. i will follow up later today.thanks for all the help.

KalixtShawxo, 2021-10-17:
> If you click that settings cog on printer manager it will show you what the problem was. If it's anything like your previous logs it will be the "Settings Check"
>
> Your printer issue sounds like something didn't get cleared out in the backend. i.e... the websocket as I'd just fixed a bug with that prepping for v1.2. What I would suggest, is going into the System settings and nuking the printer database, and restarting OctoFarm. That should clear our anything errant in the database until I get this version pushed with the fix. Will give you a good starting ground anyway.
>
> No trouble as well, and rest easy.

ok. I'm tired but hardheaded.I went in and nuked everything.logged back in as a new user, detected printers and got this:

still shows an API issue with the existing (printing) printer
below is the details of said printer, along with an error popup. 

Now I'm going to bed. :)

NotExpectedYet, 2021-10-17:
I don't think your images sent over mobile?

KalixtShawxo, 2021-10-17:
lets try the pix again

NotExpectedYet, 2021-12-17:
> Have my followup messages not been going through?

Hey man sorry about the late reply, yeah but I can't see no images on them. You can see above on the main github repo what was sent.

NotExpectedYet, 2022-01-09:
Closing this in favour of #766 

Please open another ticket if you'd like to continue with the connection issue.

closed 2022-01-09 by NotExpectedYet