OctoFarm/OctoFarm #768: Offline Octoprint causes Camera, List and Panel views to not show any information

TheLion, 2021-11-02:
### OctoFarm Version

1.1.13-hotfix

### OctoPrint Version(s)

1.7.2

### pm2 or docker?

docker

### Operating System

Octoprint on Raspberry Pi OS, OctoFarm on Docker on Synology

### Location of the problem

Web Interface

### Description of the problem

```bash
When one of the OctoPrint instances/servers is offline and not available on the network, the Camera, List and Panel views do not show any information. Dashboard and Current Operations show the correct information.
```

### Steps taken to reproduce

```bash
1. Add an offline or not existing OctoPrint printer
2. Open OctoFarm, check views
3. Delete offline or not existing Octorprint printer
4. Open OctoFarm, check views
```

### What did you expect to happen

```bash
I expected that those views would show information, even when a particular OctoPrint instance/server is not reachable on the network.
```

### To avoid the problem I had to

```bash
Delete the offline Octoprint instance/server from the printers list.
```

TheLion, 2021-11-04:
Update: The issue doesn't occur when Octoprint isn't reachable, just also had the issue with a reachable Octoprint, but with an offline printer, due to it being powered down. When I powered up the printer, connected it in Octoprint and did a re-sync. everything was fine again.

Just disconnected the printer in Octoprint, did a re-sync again, and the issue was there again.

[image]

Seems like when the printer is disconnected, something seems to lock up, causing the Camera, List and Panel views to not show any information.

NotExpectedYet, 2021-11-05:
> Update: The issue doesn't occur when Octoprint isn't reachable, just also had the issue with a reachable Octoprint, but with an offline printer, due to it being powered down. When I powered up the [...]

Yeah I agree, it seems there is a state not been accounted for here on those views. Will investigate thanks for the report!

TheLion, 2021-11-05:
> Yeah I agree, it seems there is a state not been accounted for here on those views. Will investigate thanks for the report!

You're welcome. If there is anything I need to test, please let me know.

NotExpectedYet, 2021-11-05:
@TheLion it's strange it been disconnected state that's causing it. The power of your printer shouldn't matter, the codes completely seperate. 

Narrows it down at least. 

Any clear steps for it would be awesome. I have some test instances installed with a power plugin to test so I can check myself as well and hopefully find the cause. It's good timing as I can squash it for 1.2 :)

TheLion, 2021-11-05:
> Edit: re-read your first post you gave me steps. Think I know what that is thanks! I believe I've already snagged this on 1.2 but I will double check.

That's fast! Great!

NotExpectedYet, 2021-11-06:
> 
> 
> > Edit: re-read your first post you gave me steps. Think I know what that is thanks! I believe I've already snagged this on 1.2 but I will double check.
> 
> That's fast! Great!

https://github.com/OctoFarm/OctoFarm/pull/767 - That's what I've left to do for 1.2. It's mostly checking. 

If you have the ability to test with a second OF instance (there's still some broken on development) I'd love confirmation. You should be able to get onto the view screens, and printer manager should be fine, it's just not ready for prime time yet.

TheLion, 2021-11-06:
> #767 - That's what I've left to do for 1.2. It's mostly checking.
>
> If you have the ability to test with a second OF instance (there's still some broken on development) I'd love confirmation. You [...]

Sure. Can I deploy the development branche with Docker?

NotExpectedYet, 2021-11-06:
> Sure. Can I deploy the development branche with Docker?

You can, the tag is :dev :monolithic-dev :alpine-dev depending which image you use. I'd probably wait till tomorrow though, I'm making quite a lot of changes today, so might as well get the latest and I still need to build the client with the current development changes.

TheLion, 2021-11-06:
> You can, the tag is :dev :monolithic-dev :alpine-dev depending which image you use. I'd probably wait till tomorrow though, I'm making quite a lot of changes today, so might as well get the latest and I still need to build the client with the current development changes.

That's probably a good idea ;-) Will test it tomorrow/day after tomorrow and will update here.

TheLion, 2021-11-12:
Hi,

I deployed the :dev on my docker. Setup looks ok, except for the large spinning circle in the bottom-left. But creating a user and logging in works fine. But when switching to the printers setup, the screen stays like shown in the video below.

https://user-images.githubusercontent.com/3755011/141480946-87f4483d-a506-477a-9dc0-0d7370f3b267.mov

The log shows some errors:

```
15:12:28 0|OctoFarm  | 2021-11-12T15:12:28: Had to redirect resource request: /assets/dist/bootstrap.min.js
15:12:28 0|OctoFarm  | 2021-11-12T15:12:28: 12/11/2021, 14:12:28 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/bootstrap.min.js
15:12:28 0|OctoFarm  | 2021-11-12T15:12:28: Had to redirect resource request: /assets/dist/vendor.min.js
15:12:28 0|OctoFarm  | 2021-11-12T15:12:28: 12/11/2021, 14:12:28 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/vendor.min.js
15:12:28 0|OctoFarm  | 2021-11-12T15:12:28: Had to redirect resource request: /assets/dist/amialive.service.min.js
15:12:28 0|OctoFarm  | 2021-11-12T15:12:28: 12/11/2021, 14:12:28 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/amialive.service.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: Had to redirect resource request: /assets/dist/octofarm.css
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: Had to redirect resource request: /assets/dist/client-error.modal.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: 12/11/2021, 14:12:33 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/client-error.modal.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: Had to redirect resource request: /assets/dist/bootstrap.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: 12/11/2021, 14:12:33 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/bootstrap.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: Had to redirect resource request: /assets/dist/vendor.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: 12/11/2021, 14:12:33 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/vendor.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: Had to redirect resource request: /assets/dist/amialive.service.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: 12/11/2021, 14:12:33 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/amialive.service.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: Had to redirect resource request: /assets/dist/printer-management.runner.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: 12/11/2021, 14:12:33 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/printer-management.runner.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: Had to redirect resource request: /assets/dist/bootstrap.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: 12/11/2021, 14:12:33 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/bootstrap.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: Had to redirect resource request: /assets/dist/vendor.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: 12/11/2021, 14:12:33 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/vendor.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: Had to redirect resource request: /assets/dist/amialive.service.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: 12/11/2021, 14:12:33 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/amialive.service.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: Had to redirect resource request: /assets/dist/printer-management.runner.min.js
15:12:33 0|OctoFarm  | 2021-11-12T15:12:33: 12/11/2021, 14:12:33 | ERROR | OctoFarm-Server | Javascript resource was not found /assets/dist/printer-management.runner.min.js
```

NotExpectedYet, 2021-11-12:
@TheLion ah bugger that's because I haven't built the new client. 

Will push what I've got ready tomorrow morning and do a new Dev image :). Sorry about that!

NotExpectedYet, 2021-11-12:
I'll make it an rc4 just so you can tell it's updated.

TheLion, 2021-11-13:
> 
> 
> I'll make it an rc4 just so you can tell it's updated.

Great. Let me know when it's available.

NotExpectedYet, 2021-11-14:
> 
> 
> > I'll make it an rc4 just so you can tell it's updated.
> 
> Great. Let me know when it's available.

Just about to merge and build now :)

NotExpectedYet, 2021-11-14:
> 
> 
> > I'll make it an rc4 just so you can tell it's updated.
> 
> Great. Let me know when it's available.

Should be good, just checking it myself

NotExpectedYet, 2021-11-14:
All good :)

https://github.com/OctoFarm/OctoFarm/projects/30

Check here for what's left to do / check. Feel free to point out any bugs you may discover.

TheLion, 2021-11-15:
Hi,

I've got the rc4 running, but regarding this issue, it looks the same.

When having a disconnected printer, the Camera, List and Panels view still stay empty.

The disconnected printer show as "unknown" in the printers section.

After removing one offline printer, the panels do load and show only the acgtive printer. Even though the filter is set to "All printers".

I added the printer I removed earlier and it still shows only the online printer, but briefly shows the only offline printer.

https://u.pcloud.link/publink/show?code=XZ0HoUXZvPAANI2rScjqyqQfCqsprkJb9zpk

After I re-sync in OctoFarm, the two online printer show instantly and the one offline printer shows briefly.

https://u.pcloud.link/publink/show?code=XZhHoUXZEmmSKkBuBFydmSEn0dXqJf7F4k7V

Anything specific you want me to test, or any logs you need?

P.S. I did notice that the link to your website (https://notexpectedyet.com/) in the About section wasn't working ;-) Tried to use it to jump to your Github quickly ;-)

NotExpectedYet, 2021-11-16:
Thanks for the videos will check those out. Still not sorted then grr. 

As one extra bit of information could you pass me what settings you have for the client please? In System -> Client -> Views. 

Think that's all I should require. 

Oh! Thanks for pointing that out aha, I completely forgot that link was there and I've been staring at that code recently. I'll update with better linkage.

NotExpectedYet, 2021-11-16:
It's looking like it's those settings tbh, as when I've got those off it shows as expected. Please do let me know if I'm wrong, just watched your videos and that's certainly what it would do. 

I think I'll default those to show for future, think it's defaulted to hide both now.

NotExpectedYet, 2021-11-16:
[image]
 
 That's better :D

TheLion, 2021-11-16:
> As one extra bit of information could you pass me what settings you have for the client please? In System -> Client -> Views.

Those are set like this:

[image]

I did change this yesterday, to see if that would solve anything. When it was in default, the behavior was the same though.

NotExpectedYet, 2021-11-16:
> > As one extra bit of information could you pass me what settings you have for the client please? In System -> Client -> Views.
> 
> Those are set like this:
> 
> [image]
> 
> I did change this yesterday, to see if that would solve anything. When it was in default, the behavior was the same though.

Cheers for the extra info! Will give them a try. Those settings would certainly hide it but I'll have a double check here.

NotExpectedYet, 2021-11-16:
[image]

Hmmm these settings are working fine for me. Confused! 

[image]

NotExpectedYet, 2021-11-16:
[image]

Wonder if it's something with the settings not getting set right on boot. Will investigate :)

NotExpectedYet, 2021-11-17:
Think I've snagged it. 

Will do another development version if you don't mind confirming. Will be a day or two. 

It was those settings, or more the view not reacting to the specific settings. Do you use a user? I think I inadvertently caused a bug on the latest release to not update a users settings when changing which was a new feature.

NotExpectedYet, 2021-11-23:
> > As one extra bit of information could you pass me what settings you have for the client please? In System -> Client -> Views.
> 
> Those are set like this:
> 
> [image]
> 
> I did change this yesterday, to see if that would solve anything. When it was in default, the behavior was the same though.

The latest was pushed last night, forgot to mention. I "think" that has squashed this bug :) hopefully anyway.

TheLion, 2021-11-24:
Hi,

I did deploy the new Docker container from docker.io/octofarm/octofarm:dev, but I am unable to login. Right after restart I do get a login screen, but when I enter login-credentials and press Login button, it freezes up.

The log shows:

```
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47: TypeError: Cannot read property '_id' of undefined
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at ensureCurrentUserAndGroup (/app/server_src/config/users.js:7:53)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at Layer.handle [as handle_request] (/app/node_modules/express/lib/router/layer.js:95:5)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at next (/app/node_modules/express/lib/router/route.js:137:13)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at ensureAuthenticated (/app/server_src/config/auth.js:11:14)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at Layer.handle [as handle_request] (/app/node_modules/express/lib/router/layer.js:95:5)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at next (/app/node_modules/express/lib/router/route.js:137:13)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at Route.dispatch (/app/node_modules/express/lib/router/route.js:112:3)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at Layer.handle [as handle_request] (/app/node_modules/express/lib/router/layer.js:95:5)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at /app/node_modules/express/lib/router/index.js:281:22
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at Function.process_params (/app/node_modules/express/lib/router/index.js:335:12)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at next (/app/node_modules/express/lib/router/index.js:275:10)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at Function.handle (/app/node_modules/express/lib/router/index.js:174:3)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at router (/app/node_modules/express/lib/router/index.js:47:12)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at Layer.handle [as handle_request] (/app/node_modules/express/lib/router/layer.js:95:5)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at trim_prefix (/app/node_modules/express/lib/router/index.js:317:13)
10:26:47 0|OctoFarm  | 2021-11-24T10:26:47:     at /app/node_modules/express/lib/router/index.js:284:7
```
When I reload the page, It doesn't load. Only after restarting the Docker container, the login page loads.

I did not create a new database. Just used the one from the previous RC. Should I create a new, empty database?

NotExpectedYet, 2021-11-24:
> 

That's something I've missed by the looks, thanks for that will check it out. 

It should work fine with a fresh install, would be good to get confirmation regardless. 

I appologise as well, there's so many changes been done for V1.2 it seems there's more checking to be done...

NotExpectedYet, 2021-11-24:
[image]

Yup... missed that off. I'm a fair bit off the next build as well.

NotExpectedYet, 2021-11-25:
I've just tried it with the docker version I used to confim originally and I haven't got the current issue. I've just pushed the change to development anyway. Just waiting for my container to realise there's an update. Fingers crossed that should solve it.

NotExpectedYet, 2021-11-25:
I have tested and seems to work with login. 

testing database inc old user account - [x]
testing database, nuked client settings (should be your situation) - [x] 
testing fresh database, create user - [x]

The views are borked for me though -_-. 

I'm going to have to wait till I get a chance to investigate a bit more as it's the recent change with the client settings.

NotExpectedYet, 2021-11-26:
Pushed the fix. 

Tested again and all seems to be working now. It was totally those wonky settings, and the initial user creation was failing to update the settings cache. Should be good for a test now, just building the docker now shouldn't be too long.

TheLion, 2021-11-26:
> Pushed the fix.
> 
> Tested again and all seems to be working now. It was totally those wonky settings, and the initial user creation was failing to update the settings cache. Should be good for a test now, just building the docker now shouldn't be too long.

Great. Let me know when the docker is up-to-date.

Can I use the current database I created to test, or should I create a new, empty database?

NotExpectedYet, 2021-11-26:
> Great. Let me know when the docker is up-to-date.
>
> Can I use the current database I created to test, or should I create a new, empty database?

It should be built by now. I forgot to update the RC number as well doh. I'm having trouble with my UnRaid setup to test pulling but let me know if it hasn't and I'll do another push. 

Either should work fine I think, previous or new. I've checked new, and previous and was fine so fingers crossed aha.

TheLion, 2021-12-03:
I am running rc5 now. Seems to work ok now.

Will do some more testing in the next days..

I do get this error in the log

```
22:12:02 0|OctoFarm  | 2021-12-03T22:12:02: 03/12/2021, 21:12:02 | ERROR | OctoPrint-API-Service | Error connecting to OctoPrint API: /api/settings | http://octopi-x1.fourpets.net | timeout: 1000 data: "timeout" | 
22:12:02 0|OctoFarm  | 2021-12-03T22:12:02: 03/12/2021, 21:12:02 | ERROR | OctoPrint-API-Service | Error connecting to OctoPrint API: /api/settings | http://ratos.fourpets.net:5000 | timeout: 10000 data: "timeout" | 
22:12:02 0|OctoFarm  | 2021-12-03T22:12:02: 03/12/2021, 21:12:02 | ERROR | OctoFarm-State | timeout data: Couldn't grab initial connection for Printer: http://ratos.fourpets.net:5000 | 
22:12:02 0|OctoFarm  | 2021-12-03T22:12:02: 03/12/2021, 21:12:02 | ERROR | OctoPrint-API-Service | Error connecting to OctoPrint API: /api/settings | http://octopi-mk3s.fourpets.net | timeout: 10000 data: "timeout" | 
22:12:02 0|OctoFarm  | 2021-12-03T22:12:02: 03/12/2021, 21:12:02 | ERROR | OctoFarm-State | timeout data: Couldn't grab initial connection for Printer: http://octopi-mk3s.fourpets.net | 
```

Even though all three Octoprint are running and two out of three are connected to the printer.

NotExpectedYet, 2021-12-04:
> I am running rc5 now. Seems to work ok now.
>
> Will do some more testing in the next days..
>
> I do get this error in the log
>
> ```
> 22:12:02 0|OctoFarm  | 2021-12-03T22:12:02: 03/12/2021, 21:12:02 | [...]

I wouldn't worry about those, that settings API sometimes takes a while to come back and I think my initial timeout is a bit weak, seems to have snagged the data on the second try. 

Brilliant as well, good to know it's snagged as well, thanks for getting back to me. 

Just to warn the multi-upload / multi-print stuff is broken. Should have a fix soonish with another (and hopefully final rc)

TheLion, 2021-12-24:
I tried to deploy a new dev docker, because the old one refused to connect to my Octoprint statues, strangely enough, but it fails to start with the following lines in the log.

Not sure whay that is.

```
18:05:04 0|OctoFarm  | 2021-12-24T18:05:04: Warning: connect.session() MemoryStore is not
18:05:04 0|OctoFarm  | 2021-12-24T18:05:04: designed for a production environment, as it will leak
18:05:04 0|OctoFarm  | 2021-12-24T18:05:04: memory, and will not scale past a single process.
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07: 24/12/2021, 17:05:07 | ERROR | OctoFarm-Server | MongooseServerSelectionError: getaddrinfo ENOTFOUND 192.168.206.2000
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at NativeConnection.Connection.openUri (/app/node_modules/mongoose/lib/connection.js:797:32)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at /app/node_modules/mongoose/lib/index.js:332:10
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at /app/node_modules/mongoose/lib/helpers/promiseOrCallback.js:32:5
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at new Promise (<anonymous>)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at promiseOrCallback (/app/node_modules/mongoose/lib/helpers/promiseOrCallback.js:31:10)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at Mongoose._promiseOrCallback (/app/node_modules/mongoose/lib/index.js:1153:10)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at Mongoose.connect (/app/node_modules/mongoose/lib/index.js:331:20)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at Object.<anonymous> (/app/app.js:45:6)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at Module._compile (internal/modules/cjs/loader.js:1072:14)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1101:10)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07: Error: Cannot find module './routes/serverChecks'
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07: Require stack:
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07: - /app/server_src/app-fallbacks.js
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07: - /app/app.js
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:889:15)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at Module.Hook._require.Module.require (/usr/local/lib/node_modules/pm2/node_modules/require-in-the-middle/index.js:61:29)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at require (internal/modules/cjs/helpers.js:92:18)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at serveDatabaseIssueFallbackRoutes (/app/server_src/app-fallbacks.js:79:28)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at serveDatabaseIssueFallback (/app/server_src/app-fallbacks.js:95:3)
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:     at /app/app.js:68:7 {
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:   code: 'MODULE_NOT_FOUND',
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07:   requireStack: [ '/app/server_src/app-fallbacks.js', '/app/app.js' ]
18:05:07 0|OctoFarm  | 2021-12-24T18:05:07: }
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33: TypeError: /app/views/layout.ejs:60
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     58|                 </a>
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     59|             </li>
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:  >> 60|             <% if(monitoringSettings.panel){ %>
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     61|                 <li class="nav-item">
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     62|                     <a title="Panel Layout Monitoring View" class="nav-link text-light text-center btn-sm"
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     63|                        id="menuPanelView" href="/mon/panel"><i class="fas fa-columns"></i>
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33: 
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33: Cannot read property 'panel' of undefined
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     at eval (eval at compile (/app/node_modules/ejs/lib/ejs.js:662:12), <anonymous>:48:30)
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     at layout (/app/node_modules/ejs/lib/ejs.js:692:17)
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     at tryHandleCache (/app/node_modules/ejs/lib/ejs.js:272:36)
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     at View.exports.renderFile [as engine] (/app/node_modules/ejs/lib/ejs.js:489:10)
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     at View.render (/app/node_modules/express/lib/view.js:135:8)
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     at tryRender (/app/node_modules/express/lib/application.js:640:10)
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     at Function.render (/app/node_modules/express/lib/application.js:592:3)
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     at ServerResponse.render (/app/node_modules/express/lib/response.js:1012:7)
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     at /app/node_modules/express-ejs-layouts/lib/express-layouts.js:113:20
18:10:33 0|OctoFarm  | 2021-12-24T18:10:33:     at tryHandleCache (/app/node_modules/ejs/lib/ejs.js:278:5)
18:10:34 0|OctoFarm  | 2021-12-24T18:10:34: TypeError: /app/views/layout.ejs:60
18:10:34 0|OctoFarm  | 2021-12-24T18:10:34:     58|                 </a>
18:10:34 0|OctoFarm  | 2021-12-24T18:10:34:     59|             </li>
18:10:34 0|OctoFarm  | 2021-12-24T18:10:34:  >> 60|             <% if(monitoringSettings.panel){ %>
18:10:34 0|OctoFarm  | 2021-12-24T18:10:34:     61|                 <li class="nav-item">
[... cut ...]
18:10:34 0|OctoFarm  | 2021-12-24T18:10:34:     at tryRender (/app/node_modules/express/lib/application.js:640:10)
18:10:34 0|OctoFarm  | 2021-12-24T18:10:34:     at Function.render (/app/node_modules/express/lib/application.js:592:3)
18:10:34 0|OctoFarm  | 2021-12-24T18:10:34:     at ServerResponse.render (/app/node_modules/express/lib/response.js:1012:7)
18:10:34 0|OctoFarm  | 2021-12-24T18:10:34:     at /app/node_modules/express-ejs-layouts/lib/express-layouts.js:113:20
18:10:34 0|OctoFarm  | 2021-12-24T18:10:34:     at tryHandleCache (/app/node_modules/ejs/lib/ejs.js:278:5)
```

NotExpectedYet, 2021-12-25:
> > I tried to deploy a new dev docker, because the old one refused to connect to my Octoprint statues, strangely enough,

Merry Christmas! 

I think that's because I broke the reconnect, Just noticed now whilst re-writing the websocket implementation. if you go to System -> Server -> Timeout and increase the initial api timeout to 2-3 seconds it should let you connect.
[image]

That last one is an issue with the settings... hmmm I'll double check that again thanks for the report! 

Also that docker is installing the development environment which it shouldn't be doing I'll make sure that's correct too huh...

TheLion, 2021-12-25:
> > > I tried to deploy a new dev docker, because the old one refused to connect to my Octoprint statues, strangely enough,
> 
> Merry Christmas!

Yes. Same to you. Hope you have wonderful days and aren't coding to squash any bugs ;-)

> I think that's because I broke the reconnect, Just noticed now whilst re-writing the websocket implementation. if you go to System -> Server -> Timeout and increase the initial api timeout to 2-3 seconds it should let you connect. [image]

I removed the failing dev docker, so I can't go back :-(

> That last one is an issue with the settings... hmmm I'll double check that again thanks for the report!

Great. Always glad to help. 

> Also that docker is installing the development environment which it shouldn't be doing I'll make sure that's correct too huh...

That is what I intended. Should I have used a different tag? There are a lot to choose from ;-)

NotExpectedYet, 2021-12-26:
> > > > I tried to deploy a new dev docker, because the old one refused to connect to my Octoprint statues, strangely enough,
> > 
> > 
> > Merry Christmas!
> 
> Yes. Same to you. Hope you have wonderful days and aren't coding to squash any bugs ;-)

Did a tiny bit whilst I was waiting for people to wake up ahaha. 

> I removed the failing dev docker, so I can't go back :-(
>
>
> Great. Always glad to help.
>
>
> That is what I intended. Should I have used a different tag? There are a lot to choose from ;-)

No it should just be the -dev tag. There's a development tag as a well because of another dev but I'm going to clean that up and stop pushing to it as it's pointless. I'm just a bit out of touch with all the dockers but I'll get back up to speed. 

I'll give you a shout for the next dev then! and fingers crossed it's licked. It will have my new websocket and api connection manager on it, I'm just linking that up now. No double listeners, should be no need for resyncing, and should give me a better window into what's going on if someone does! aha.

TheLion, 2021-12-29:
> I'll give you a shout for the next dev then! and fingers crossed it's licked. It will have my new websocket and api connection manager on it, I'm just linking that up now. No double listeners, [...]

Great. Just let me know and I will deploy it again.

commit ef305cc0 (2021-11-17): Fixed #768 - Incorrect reaction to hide settings on views
commit 956baeef (2021-11-22): Bugfix/rc5 check (#771)

* automate the columns for printers on group view

* space out cards better on super list view

* fix sorting not working on list view types

* file manager improvements

* fix single file sync same was as previous

* no need to render html everytime for release notes

* imrpove loggin

* Fixed #768 - Incorrect reaction to hide settings on views

* add settings to system information dump

* react to correct hide settings

* date and time dashboard widget

* tweaks to allowed widget sizes

* fix history chart colours wrong way wound

* slight tweaks to filament manager  [... cut]
closed 2022-05-19 by NotExpectedYet