OctoFarm/OctoFarm #777: 400 Bad request on PSU Control Calls from WebIF

itsyoshio, 2021-12-08:
### OctoFarm Version

1.1.13-hotfix

### OctoPrint Version(s)

1.7.2

### pm2 or docker?

pm2

### Operating System

FarmPi (Ubuntu 20.04.2 LTS)

### OctoPrint Plugins

PSU Control

### Location of the problem

Web Interface

### Description of the problem

```bash
PSU Control Api requests get 400 Bad Requests

OctoFarm is on a Raspberry Pi 3 (installed via FarmPi)\
on that same Raspberry Pi 3, there is a (manual) install of OctoPrint.

OctoPrint has the PSU Plugin, which works in the OctoPrint Webinterface.\
The PSU Plugin also works with curl (Rest API on OctoPrint works).

Requests to control the PSU with the OctoFarm UI result in 400 Bad Requests regardless of how i configure it (i did Empty Cache and Hard Reload after changing the Power Settings).

CORS is Enabled, Controlling Works, Printing Works, Upload Works.
```

### Steps taken to reproduce

```bash
Set up Power Control in OctoFarm

Versions i've tried : 

`[PrinterURL]/api/plugin/psucontrol?apikey=[PrinterAPI]`

`http://localhost:80/octoprint/api/plugin/psucontrol?apikey=[PrinterAPI]`
(Afterwards i've realised the request is a JS Fetch request made from my Client, not the Server)

`http://192.168.188.50/octoprint/api/plugin/psucontrol?apikey=[PrinterAPI]`

each with and without APIkey.
```

### What did you expect to happen

```bash
Turning PSU On, Off, Toggling and getting State in OctoFarm.
```

### To avoid the problem I had to

```bash
No resolution yet
```

NotExpectedYet, 2021-12-09:
I have documentation for those here. 

https://docs.octofarm.net/guides/octoprint-supported-plugins.html#octoprint-s-power-control-plugins

PSU control you can use the examples on the power settings page. You need to paste in those commands though it won't work without those hence the 400 error. Hope that helps.

NotExpectedYet, 2021-12-09:
Just to add I don't think you should have to nuke cache. The commands are built dynamically so should work. The only limitation I remember was you'd need a page refresh to make the button show up.

NotExpectedYet, 2021-12-09:
Out of curiosity how's the b3+ been handling OP and OF? 

If you've any thoughts on making that screen clearer then let me know :). Next version the auto pulling of those settings is fixed so anyone with PSU control already setup when OF scans won't hit this.

itsyoshio, 2021-12-09:
### Fetch Requests
Oh, i would've caught that if i had actually tested POST requests with curl, so sorry about that!\
I i didn't even think about the possibility that i'm using incorrect JSON in my POST.\
Not to blame you at all, but the help text below is a bit misleading, maybe change the help text to a link for the [Docs](https://docs.octofarm.net/guides/octoprint-supported-plugins.html#octoprint-s-power-control-plugins)?

### Docs
The Docs are really helpfull. Sorry i didn't think of searching the Website, my Google Skills didn't yield any results.

### Ticket
I'll go ahead and close this, i'll only be able to implement this on Wednesday since this is running in my University MakerSpace and that is my next Shift. I'll reopen in the most unlikely event that didn't resolve my issue.

### Usage
Regarding the Handling : i have not looked at the actual cpu utilization or mem yet, but i will do so and report back.\
Printing with a Job from OctoFarm did result in regular, stable printing.

Actually we have 4 Pi's in a Eletric Box with Dual SSR Relais for each Channel / Printer.\
In the End one OctoFarm will oversee 4 OctoPrints (where OctoFarm is on the same Pi as one OctoPrint).

Once that's done (probably next Wednesday) i will report back with more Stats.

---

Thank you so much for your help.\
Have a great Day.

closed 2021-12-09 by itsyoshio

NotExpectedYet, 2021-12-11:
> ### Fetch Requests
>
> Oh, i would've caught that if i had actually tested POST requests with curl, so sorry about that! I i didn't even think about the possibility that i'm using incorrect JSON in my [...]

Tbh, A little button with those images that pop up might be nice, one for each supported plugin. This is exactly why I asked, those interface notes haven't been touched since I first wrote them so I thought they might be a bit misleading. If the automation grabbing of settings was working we wouldn't be here regardless! aha. 

No trouble on the rest either, do let me know if it resolves with those objects. 

That's good to know regardless, I've had some issues with increased CPU over a Pi which is resolved in V1.2 but if you've not hit it so far doubt you will.