OctoFarm/OctoFarm #791: 4 instances of Octoprint 'Awaiting Websockets' custom setup.

wanderling, 2022-01-18:
### OctoFarm Version

1.1.13-hotfix

### OctoPrint Version(s)

1.7.2

### pm2 or docker?

pm2

### Operating System

Raspberry Pi OS (64 bit)

### OctoPrint Plugins

BetterHeaterTimeout (1.3.0), Consolidate Temp Control (0.1.8), Custom Background (1.0.0), Dashboard (1.19.5), DisplayLayerProgress Plugin (1.27.1), Filament Sensor Simplified (0.2.3), Floating Navbar (0.3.7), ipOnConnect (0.2.4), Preheat Button (0.8.0), PrettyGCode (1.2.4), PrintJobHistory (1.14.0), PrintTimeGenius Plugin (2.2.8), Themeify (1.2.2)

### Location of the problem

Not sure

### Description of the problem

```bash
Hand built MongoDB for Raspberry Pi OS (64bit). Configured Raspberry Pi to use built in eth0 for LAN facing connection. Configured USB->ethernet eth1 for PVLAN through dumb switch for the four printers.
Printers are isolated from the parent LAN and thus the internet. The Raspberry Pi can access the internet through the LAN facing side no problem. OctoFarm can see and talk to all of the printers but evidently not make a websocket connection. CORS is enabled on all printers. API key generated through a user account for each printer separately. And finally, yes, I did press save.

I think this may be a firewall issue, but I don't know enough about websockets to continue.
```

### Steps taken to reproduce

```bash
1. Get Raspberry Pi 4B 8Gb
2. Install Raspberry pi OS (64 bit)
3. Compile MongoDB from source (so fun...)
4. Configure Pi to isolate LAN and PVLAN eth0 and eth1 respectively
5. Install Octofarm on Pi
6. Attach Pi to LAN and PVLAN switch, test.
7. Connect and configure all instances of Octoprint.
8. Profit...? Damnit.
...
```

### What did you expect to happen

```bash
Profit...?
```

### To avoid the problem I had to

```bash
...
```

wanderling, 2022-01-19:
found the issue though I don't know how to communicate a fix. I have three 'admin' accounts on each OctoPrint instance. OctoFarm is ignoring the user the token is made with and picking one (obviously not at random but I don't know). In this case it is picking the wrong one. I can mitigate the issue by creating an API key with the user OctoFarm is choosing.

wanderling, 2022-01-19:
Okay, second problem. Files will not upload to octofarm. Asks me if CORS is turned on (yes)... All of the above info still stands.

NotExpectedYet, 2022-01-19:
I'll have a think about this and see if I get time to reproduce and see what's going on. 

The user was an initial limitation, that's fixed for 1.2. That version will check for an OctoFarm user and use that, or allow you to change. This current version looks for the first admin user octoprint gives it, not the greatest I know. Until 1.2 yeah your method is best way just wanted to give a little explanation aha, was trying to save time so people didn't have to add a user. 

I can only gather the files problem is related to the network configuration somehow. 

At a coffee lacking guess I would say it is the usage of two Ethernet ports and some internal routing, OctoFarm will bind to 0.0.0.0 so your local host but I'll admit I am not sure which port that will result in using so I'll have to check up more as it's somewhat a new setup to encounter for me. 

Could I just ask the reason for two ports so I can understand that better?

NotExpectedYet, 2022-01-19:
@wanderling thanks for that, that message isn't very clear at all I'll update that. The files will always first attempt to action your command on OP (where yours is failing) and then once succeeded will be replicated on octofarms records. 

Did you manage to try all this local?

NotExpectedYet, 2022-01-19:
Infact thinking about it, as long as your using the PiVPN IP address it shouldn't matter the port. 
I'll have to see if I can find a secondary ethernet to try don't think I've got any laying around. WiFi might work.

wanderling, 2022-01-19:
So, to clarify, the PVLAN I was speaking about was my internal naming of a PriVate LAN (not to be confused with a Pirate LAN). I am pretty sure if you are bonding to localhost you are bonding to the Lo interface. I will be re-baking this whole setup again today and getting rid of all routing, iptables, and and ufw. The only thing I will be keeping is wireless hostapd and dhcpcd for IP address allocation on eth1 and wlan0. We'll see if that fixes anything. I'll let you know. Thanks for the pointers!

wanderling, 2022-01-19:
Just saw your first question. Sorry for that. I have the two ports due to me wanting to restrict access to Octoprint & SSH on the printer side. Basically I am using your program as a GUI for the entire farm. eth0 is user facing. eth1 is printer facing. wlan0 is for admin use and was suppose to let me connect to both sides (eth0 and eth1) of the octofarm pi directly (not working as of yet).

NotExpectedYet, 2022-01-22:
> Just saw your first question. Sorry for that. I have the two ports due to me wanting to restrict access to Octoprint & SSH on the printer side. Basically I am using your program as a GUI for the entire farm tracking prints, filament, and user access. eth0 is user facing. eth1 is printer facing. wlan0 is for admin use and was suppose to let me connect to both sides (eth0 and eth1) of the octofarm pi directly (not working as of yet).

Ah makes sense! Well honestly the connection that comes from OctoFarm should work as long as you can see something on the network. I do apologise as well, I had another use on Discord mentioning something similar with multiple ports and a VPN setup and I did completely forget to mention one thing to you, as I'd said it to him already aha. 

So for OctoFarm there's a little limitation you might not be aware of. OctoFarms connections to OctoPrint come from two places, the first been OctoFarm (mostly websockets and general information collection). The second connection comes from the browser, so to create a working connection for OctoFarm the server + whatever PC your accessing the client from needs to be able to see the instances. It's a bit of legacy code in the client and I haven't had the time to rip it out and proxy those requests through the server yet. 

I think that might be the key to why your files arn't working.

wanderling, 2022-01-23:
Understood. If you let me know where to start poking, I can fork your code and mess around on my own time. Might be easier for you... Otherwise I will sit in the corner and wait patiently. :)

NotExpectedYet, 2022-01-23:
> Understood. If you let me know where to start poking, I can fork your code and mess around on my own time. Might be easier for you... Otherwise I will sit in the corner and wait patiently. :)

It's already on the development branch all done. There's just some bugs I'm working through here: https://github.com/OctoFarm/OctoFarm/pull/790 to get RC8 ready for testing. You could pull and try either. If your wanting to look at code then it's better looking at that pull requests branch as I've cleaned up and moved a lot of files. 

Just to clarify further the V1.2 version has two options for users on OctoFarm. Pro-Active and Retro-Active methods. 

Pro-Active: Create a user on your OctoPrint instance called "OctoFarm" or "octofarm" and on the initial scan I will pick that user up. 
Retro-Active: Let OctoFarm connect and then use the Settings modal to change the current user. (won't work on PR branch due to me currently refactoring that specific code aha) https://github.com/OctoFarm/OctoFarm/blob/db111b20f6b9ac1009019ee6d9453cdc0b13347d/server/services/printers/create-octoprint.service.js#L877-L951

That's specifically where I grab the user list. (not great, but it works and I'm out of a lot of refactoring steam aha)

wanderling, 2022-02-07:
So, just finished spinning up a new copy using same mongodb version and newest OctoFarm 1.2.0-rc7. I think my school intranet is blocking websockets. I decided to use this version the way you intended as a unit plugged into the same network as the printers (same subnet). I still cant 'connect' to any of the added printers but I can see that OctoFarm can see (and poll) them when it automatically scans for them on the network. 

`07/02/2022, 20:01:36 | INFO | OctoFarm-Server | Running automatic scan...
07/02/2022, 20:01:36 | INFO | OctoFarm-Server | Device found! data: http://10.60.4.135:80/plugin/discovery/discovery.xml | 
07/02/2022, 20:01:36 | INFO | OctoFarm-Server | Device found! data: http://10.60.4.162:80/plugin/discovery/discovery.xml | 
07/02/2022, 20:01:36 | INFO | OctoFarm-Server | Attempting to grab device information! data: http://10.60.4.162:80/plugin/discovery/discovery.xml | 
07/02/2022, 20:01:36 | INFO | OctoFarm-Server | Captured device data: http://10.60.4.162:80/plugin/discovery/discovery.xml | 
07/02/2022, 20:01:36 | INFO | OctoFarm-Server | Attempting to grab device information! data: http://10.60.4.135:80/plugin/discovery/discovery.xml | 
07/02/2022, 20:01:36 | INFO | OctoFarm-Server | Captured device data: http://10.60.4.135:80/plugin/discovery/discovery.xml | `

How would I determine whether my school is blocking websockets? I know I can go to our IT department, I just want to know what to say.

Also, is there a work around for this that I could implement?

Thanks again.

NotExpectedYet, 2022-02-08:
Hmm It's pretty reliant on websocket connections for operations of OctoFarm so there isn't really a work around unless you can create your own segregated network. 

I am not sure how to detect if they're actively blocking websockets, are they using proxy servers for all their connections? If so it's more than likely it's stripping some headers. I think anyway, I'm no networking expert at that level. 

Have you any knowledge of another app that utilises websockets, that you might also have access too?, I'm wondering if we could rule it out that way. Would you mind at all forwarding me a log dump from the system page? 

There will be API keys and such so your welcome to forward to info@notexpectedyet.com. I doubt there's anything revealing to the specific issue but worth a look.

NotExpectedYet, 2022-02-19:
Morning, 

I've checked through your logs and none of the initial connections are succeding. Which would make me think that OctoFarm can't actually see the Pi devices at all. It never seems to grab a websocket connection, or even attempt to make one because of this. 

My loggin on that is unbeivably poor though, I'm failing to see the specific error status code that is returned. 

Dunno if you wanna risk an alpha version, but I've improved the logging there a lot! If an error code fails to produce there then it's something with the connection between OctoFarm -> OctoPrint devices, not the actual services running on it. 

That however confuses me with your responses because none of these logs have established a connection.

wanderling, 2022-02-21:
Understood, I will reroll this with the alpha version in the next week or so. Sorry for the slow input. Thanks for all of your help!

NotExpectedYet, 2022-02-22:
No rush at all, its the nature of this thing. Especially with this been a hobby for me. Take your time :). Your also most welcome. 

I was just a little confused as I couldn't match your log dump to your comments above. Didn't know whether that was my functions goof or something else.

https://docs.octofarm.net/installation/trying-alpha-versions.html#setting-up-development-version-for-docker

There's some instructions there just encase there needed.

wanderling, 2022-02-22:
Ah, so to clarify a little, One of the printers was added with autodiscovery, the other was manually added. Both have the OctoFarm user and API key. I wanted to try to eliminate as many possibilities as possible with each iteration. I find it as perplexing as you as to the fact that they can't connect on the LAN here. I will set up the Alpha version and try on the Main LAN and then just a simple done isolated LAN

NotExpectedYet, 2022-02-24:
> Ah, so to clarify a little, One of the printers was added with autodiscovery, the other was manually added. Both have the OctoFarm user and API key. I wanted to try to eliminate as many possibilities as possible with each iteration. I find it as perplexing as you as to the fact that they can't connect on the LAN here. I will set up the Alpha version and try on the Main LAN and then just a simple dumb isolated LAN.

No trouble at all thanks for the clarification. Yeah the autodiscovery is just going to pass back whatever IP:Port combo OctoPrint is aware of. 

There are a couple of things that could have potentially happened. 

OctoFarm on the older version had some bugs in the edit feature, depending which one you used. It didn't correctly re-setup the websocket connection. You could confirm by deleting the printer and re-adding it that way, not sure if you've tried. It's something I wondered into recently whilst re-writing the thing. 

Multiple users has always been a pain point for me currently. The system was setup for a single user and it's only recently I've really been trying to do something about it. That wasn't in 1.1.13 and only in the later V1.2 development branches I've been working on.

wanderling, 2022-03-28:
So, just to update you. I've been playing with various complicated setups
to try to work around the blocked socket connections without too much
success. Meaning, I can get it to work but if anything big changes I will
have to re-roll most of the setup.

Blarg.

I was searching the nets for some simpler solutions when I stumbled on
FarmPi. It looks to be based on your project but I wanted to ask you if
this has your endorsement and is updated through your repos. I really want
to stay as close to your builds as possible, if possible.

As always, thank you for your time.

NotExpectedYet, 2022-04-02:
> So, just to update you. I've been playing with various complicated setups to try to work around the blocked socket connections without too much success. Meaning, I can get it to work but if anything [...]

What big changes do you mean? Like network changes? 

FarmPi is "official" it's not created by me but another developer. It will only pull down the master version of OctoFarm though. I also don't see how the installation host would make any difference here to your setup / issues.

NotExpectedYet, 2022-05-19:
Hello, 

The master version has now been updated with all the latest development work. V1.2 basically. 

I believe this should be resolved, so I'm going to close it. Please feel free to re-open after updating if that's not the case.

closed 2022-05-19 by NotExpectedYet