OctoFarm/OctoFarm #837: File manager ERROR

maxxeweb, 2022-02-10:
### Description of the problem

```bash
...
When I open File Manager i get 
UNKNOWN ERROR (999):
Please report this error to OctoFarm Issues!
Cannot read properties of undefined (reading 'length')
```

### Reproduction of the problem

```bash
..FILE INFO
LINE: undefined
COL: undefined
STACK:
UNKNOWN: Cannot read properties of undefined (reading 'length') at http://localhost:4000/assets/dist/file-manager.runner.min.js:2:191465 at Array.forEach () at Function.drawPrinterList (http://localhost:4000/assets/dist/file-manager.runner.min.js:2:189592) at async Function.init (http://localhost:4000/assets/dist/file-manager.runner.min.js:2:193883).
```

maxxeweb, 2022-02-10:
This seems to stop Multi Load 

cannot set multi printers with same job

NotExpectedYet, 2022-02-10:
Issues for development issues are not reported here, these are for master branch issues. The development version currently has those two features broken because I'm still working through the websocket changes. 

Please either come to the developer discussions on discord, or add the issue to the lastest RC discussion list but I'm already aware of these two.

NotExpectedYet, 2022-02-11:
https://github.com/OctoFarm/OctoFarm/discussions/779

Added here

NotExpectedYet, 2022-02-11:
Changing how I'm doing this now

NotExpectedYet, 2022-04-11:
Hi this should be fixed now. Multi-Upload is disabled until I get around to cleaning it up though :) will close this. Multi-Upload is on my list of to-do once I've sorted filament manager.

closed 2022-04-11 by NotExpectedYet