OctoFarm/OctoFarm #869: printers not connecting

sidneyns, 2022-02-15:
### OctoFarm Version

1.0.2 rc7

### OctoPrint Version(s)

1..3

### pm2 or docker?

pm2

### Operating System

windows 

### Location of the problem

Web Interface

### Description of the problem

```bash
...
octofarm is not connecting the printers ( API and everything set)

this is the error 

UNKNOWN ERROR (999):
Please report this error to OctoFarm Issues!
Cannot read properties of null (reading 'powerConsumption')
FILE INFO
LINE: undefined
COL: undefined
STACK:
UNKNOWN: Cannot read properties of null (reading 'powerConsumption') at Function.setupCostTab (http://192.168.0.102:4000/assets/dist/printer-management.runner.min.js:20:7448) at ye (http://192.168.0.102:4000/assets/dist/printer-management.runner.min.js:20:2376) at async HTMLButtonElement. (http://192.168.0.102:4000/assets/dist/printer-management.runner.min.js:20:182477)
```

### Steps taken to reproduce

```bash
1.
2.
3.
...
```

### What did you expect to happen

```bash
...
```

### To avoid the problem I had to

```bash
...
```

NotExpectedYet, 2022-02-15:
Hi there isn't enough information here at all. 

The error is related to the settings page or printer management, what were you doing to get it? 

Can you please provide a log dump from the system page and some information about your setup. 

Octoprint 1.3 isn't supported either. I support the previous 4 versions of OP there or abouts. So 1.4+

NotExpectedYet, 2022-02-15:
Also I don't have an OctoFarm 1.0.2 version, and rc7 is quite old. If your testing the development versions then please update to the latest development branch and reconfirm the issue.

NotExpectedYet, 2022-02-15:
https://docs.octofarm.net/installation/trying-alpha-versions.html

There is documentation here. 

For now I recommend a fresh database because there are some issues with the upgrade path I need to resolve.

NotExpectedYet, 2022-02-15:
Ignore the docs part about reporting issues I'm going to collate them here and tag them with the branch: development tag :)

sidneyns, 2022-02-15:
sorry my bad..the version I am using is 1.2.0 RC7 

The error is related to the printer setup. 

Octofarm loads correctly, but when I add the printer ,but after that it keeps searching for the printer ( API was connected ok)

[image]

I am using octoprint 1.7.3 
[OctoFarm-InformationCleaning.log](https://github.com/OctoFarm/OctoFarm/files/8074278/OctoFarm-InformationCleaning.log)
[OctoFarm-Server.log](https://github.com/OctoFarm/OctoFarm/files/8074279/OctoFarm-Server.log)
[OctoFarm-SoftwareUpdateChecker.log](https://github.com/OctoFarm/OctoFarm/files/8074280/OctoFarm-SoftwareUpdateChecker.log)
[OctoFarm-State.log](https://github.com/OctoFarm/OctoFarm/files/8074281/OctoFarm-State.log)
[OctoFarm-TaskManager.log](https://github.com/OctoFarm/OctoFarm/files/8074282/OctoFarm-TaskManager.log)
[pm2.error.log](https://github.com/OctoFarm/OctoFarm/files/8074283/pm2.error.log)
[pm2.out.log](https://github.com/OctoFarm/OctoFarm/files/8074284/pm2.out.log)
[OctoFarm-API.log](https://github.com/OctoFarm/OctoFarm/files/8074285/OctoFarm-API.log)

sidneyns, 2022-02-15:
ok ..thank you..

I hope I help..kinda new at this..thank you

NotExpectedYet, 2022-02-16:
> 

No worries!, sorry if I was a little brash I was rather tired when I saw it. 

I think to establish a base line we should upgrade to the latest alpha version as I mentioned previously. 

Did you download the zip and run it that way? or did you use git clone?

If so I can give you a little walk-through to move over to the git version and get that updated. The docs I shared don't explain that at all and they are more geared towards a linux install.

sidneyns, 2022-02-16:
I think I am going to install in a raspberry pi 4 I have around here..i
think it should fix all the issues quick.

sidneyns, 2022-02-16:
Answering your question

I have downloaded the zip file on 1.2.0 RC7 folder

NotExpectedYet, 2022-02-17:
> I think I am going to install in a raspberry pi 4 I have around here..i think it should fix all the issues quick.
> […](#)

That would be great, all the documentation stands for that for checking out the development version. 

If you'd like instructions to move your current windows install over let me know, but I think it might be easier on the linux version personally.

sidneyns, 2022-02-17:
I am new at linux...but I know you can run linux on the background of
windows...
I will try it
If you have some documentations to help, it would be appreciated

NotExpectedYet, 2022-02-17:
> I am new at linux...but I know you can run linux on the background of windows... I will try it If you have some documentations to help, it would be appreciated On Thu, Feb 17, 2022 at 12:26 AM James Mackay ***@***.***> wrote:
> […](#)

I'll sort something out hopefully tomorrow if not Saturday for you. 

I'll work from your current install so you don't have to mess with wsl.

NotExpectedYet, 2022-02-19:
@sidneyns 

I have created some windows documentation: 
https://docs.octofarm.net/installation/install-windows.html

And I have also updated the alpha version checkout. 
https://docs.octofarm.net/installation/trying-alpha-versions.html

Hope that helps! 

If anythings confusing let me know and I'll update the wording.

sidneyns, 2022-02-19:
wow..thank you for all your help and effort...

sidneyns, 2022-02-23:
Hello my friend.

Question.

As we know there is a slowness on the last version, should I use the
version on your guide ( git clone --depth 1
https://github.com/OctoFarm/OctoFarm.git) or should I use 1.2.0-rc7?

NotExpectedYet, 2022-02-23:
> Hello my friend. Question. As we know there is a slowness on the last version, should I use the version on your guide ( git clone --depth 1 https://github.com/OctoFarm/OctoFarm.git) or should I use 1.2.0-rc7?
> […](#)

Rc7 will be less buggy honestly than the latest development build.

[](https://stackoverflow.com/posts/21699307/timeline)618

git clone --depth 1 --branch 1.2.0-rc7 https://github.com/OctoFarm/OctoFarm 

For that one :)

sidneyns, 2022-02-23:
> You won't have to do any of the checkout commands with that btw. You can
> skip those and and run the npm run start as it's already checked out on
> that branch.

I got this error message

One of the printers ( I have some), connected ok, some dont ...

[image: image.png]

NotExpectedYet, 2022-02-24:
> 

Sorry I can't see that image?

sidneyns, 2022-02-24:
UNKNOWN ERROR (999):
Please report this error to OctoFarm Issues
<https://github.com/octofarm/octofarm/issues>!
Cannot read properties of null (reading 'powerConsumption')
*FILE INFO*
LINE: undefined
COL: undefined
STACK:
UNKNOWN: Cannot read properties of null (reading 'powerConsumption') at
Function.setupCostTab (
http://192.168.0.102:4000/assets/dist/printer-management.runner.min.js:20:7448)
at ye (
http://192.168.0.102:4000/assets/dist/printer-management.runner.min.js:20:2376)
at async HTMLButtonElement. (
http://192.168.0.102:4000/assets/dist/printer-management.runner.min.js:20:182477
)

sidneyns, 2022-02-24:
I kinda got it to work on another computer installing the git command you
share on you documentation.

I was able to add the printers, but I got this message when I tried to go
to list or panel

TypeError: C:\OctoFarm\views\panelView.ejs:3
    1| <% if(typeof printerCount != 'undefined' && printerCount > 0 ) {%>
    2|
 >> 3|     <% if(clientSettings.panelView.currentOp){%> <%-
    4|     include('./partials/currentOperations.ejs') %> <% }%>
    5|     <div class="col-12 col-sm-12 col-md-12 col-lg-12 col-xl-12">
    6|       <div id="printerArea" class="row">

Cannot read properties of undefined (reading 'currentOp')
    at eval ("C:\\OctoFarm\\views\\panelView.ejs":13:36)
    at panelView (C:\OctoFarm\node_modules\ejs\lib\ejs.js:692:17)
    at tryHandleCache (C:\OctoFarm\node_modules\ejs\lib\ejs.js:272:36)
    at View.exports.renderFile [as engine]
(C:\OctoFarm\node_modules\ejs\lib\ejs.js:489:10)
    at View.render (C:\OctoFarm\node_modules\express\lib\view.js:135:8)
    at tryRender (C:\OctoFarm\node_modules\express\lib\application.js:640:10)
    at Function.render
(C:\OctoFarm\node_modules\express\lib\application.js:592:3)
    at ServerResponse.render
(C:\OctoFarm\node_modules\express\lib\response.js:1012:7)
    at ServerResponse.res.render
(C:\OctoFarm\node_modules\express-ejs-layouts\lib\express-layouts.js:77:18)
    at C:\OctoFarm\server_src\routes\index.js:206:9
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)

NotExpectedYet, 2022-02-24:
> I kinda got it to work on another computer installing the git command you share on you documentation. I was able to add the printers, but I got this message when I tried to go to list or panel [...]

Can you reach the System page? 

If you can, go to the "System" -> "Client" -> and then press the save button. 

Hopefully that will sort

sidneyns, 2022-03-05:
> at C:\OctoFarm\server_src\routes\index.js:206:9 at runMicrotasks () at

Hello

I am trying to run it on raspberry pi 4 , do you have the step by step?

sidneyns, 2022-03-05:
I was able to download farmpi and load into raspberry pi, but did not
understand why I have to download ubuntu again and save into another sd
card

following this page
https://docs.octofarm.net/installation/install-raspberry-pi.html#farmpi-distribution

NotExpectedYet, 2022-03-09:
Hello sorry I've taken a break for my birthday. 

Those are two seperate installation methods. FarmPi is the recommended.

closed 2022-05-19 by NotExpectedYet