OctoFarm/OctoFarm #884: How are Update to 1.2.0 alpha

tomtim00700, 2022-04-10:
Octofarm is running on my raspberry pi very well, but the cameras are not every time activ
I will test the alpha, but i can not Update.

How can i update in ssh the 1.1.3. hotfix to 1.2.0 ?

NotExpectedYet, 2022-04-10:
There's instructions here

https://docs.octofarm.net/installation/trying-alpha-versions.html

NotExpectedYet, 2022-04-10:
Bugs I have listed and are known are here: 

https://github.com/OctoFarm/OctoFarm/projects/30

In the V1.2-ToDo and In Progress

NotExpectedYet, 2022-04-10:
I will warn if you use the Filament manager then I'd wait because that is borked and currently been worked on for the next alpha :)

tomtim00700, 2022-04-10:
Thank you... but after the install is the system not starting :( But i will test it in the next days agein. Thank you.

NotExpectedYet, 2022-04-11:
Would need more information to help out further.

NotExpectedYet, 2022-05-19:
Hello, 

The master version has now been updated with all the latest development work. V1.2 basically. 

I believe this should be resolved, so I'm going to close it. Please feel free to re-open after updating if that's not the case.

closed 2022-05-19 by NotExpectedYet