OctoFarm/OctoFarm #886: Rolling closed websockets 1.2.0 alpha.4

ndarrow, 2022-04-11:
### OctoFarm Version

1.2.0.alpha.4

### OctoPrint Version(s)

1.7.3

### pm2 or docker?

pm2

### Operating System

Linux 5.13.0-1024-raspi (linux)

### OctoPrint Plugins

none

### Location of the problem

Not sure

### Description of the problem

```bash
OF is running on an 8GB Pi4. I have 32 printers connected via WiFi router + 1 additional access point. Initially, all the basics worked great (some things didn't work- multi upload... but I suspect that was just the result of using an Alpha), but in general no major issues. 

I had to shut the farm down for a couple days (printers, OctoPrint, OctoFarm) when I turned everything back on, all hell broke loose. 
The printers' index numbers were shuffled out of the order I put them in- not a huge deal, but really annoying. 

Once I can establish connection to the Octoprint instances and connection to the printer, they lose connections randomly and says the websocket is closed. The connections from OF to OP will come back, but the connections to the printers don't always come back. This goes on and on. Web socket closed, searching, disconnected, online, offline... it's all over the place. 
Some printers forever show Disconnected between OcotPrint and the printer, even though the OctoPrint UI shows that they are connected. 

I've tried chasing the problems around with API rescans, Disabling, Enabling printers, deleting and re-adding printers, generating new API keys and updating OF with the new keys... Sometimes this reestablishes connections, but they fail a few minutes later. 

I've rebooted the OP pi's and OF pi multiple times, but this doesn't fix the problem. 
I don't know what to do next. 

As is stands, it is currently unusable.
```

### Steps taken to reproduce

```bash
It seems the trigger was shutting the farm down completely. I can't fix it, so I don't know how to reproduce.
```

### What did you expect to happen

```bash
The farm to reestablish reliable connections once restarted.
```

### To avoid the problem I had to

```bash
Problem ongoing.
```

ndarrow, 2022-04-11:
[octofarm_dump-4-10-2022-8-22-32 PM.zip](https://github.com/OctoFarm/OctoFarm/files/8460412/octofarm_dump-4-10-2022-8-22-32.PM.zip)

ndarrow, 2022-04-11:
I just thought to increase all of the timeouts to the max and see if that helps once I leave it alone for awhile.

ndarrow, 2022-04-11:
And I'm also an idiot. When I was fighting the max router clients (before adding the additional access point) I set the 2.4GHz band to the min data rate (54Mbps). I just changed it back to 450Mbps. I may have caused this issue. Either way, I'll let this go for a while and update.

NotExpectedYet, 2022-04-11:
> And I'm also an idiot. When I was fighting the max router clients (before adding the additional access point) I set the 2.4GHz band to the min data rate (54Mbps). I just changed it back to 450Mbps. I may have caused this issue. Either way, I'll let this go for a while and update.

I would try updating to the latest alpha certainly, it's on like alpha.20 at the moment :) 

`git pull`
`npm run install-server`
`pm2 restart OctoFarm-Testing`

NotExpectedYet, 2022-04-11:
Thanks for the log dump though, will make sure nothing is me :)

NotExpectedYet, 2022-04-11:
> I just thought to increase all of the timeouts to the max and see if that helps once I leave it alone for awhile.

Just to note here, the timeouts wouldn't have caused the issue. The websocket would go stale if connection speeds drop to an extreme amount which might happen then they "should" be trying to reconnect. This is handled much better in later alphas so hopefully resolves your issue with the speed increase on the AP.

ndarrow, 2022-04-11:
Yeah, upping the speed fixed it... Sorry about that!

NotExpectedYet, 2022-04-11:
> Yeah, upping the speed fixed it... Sorry about that!

No worries! Would rather the report and be nothing than no report and something :D I'll close. 

You may start to notice a lot of websocket throttles in the logs, it's a new feature I implemented to automate the speed socket messages are received and relieve a bit of unneeded stress on OP clients. I'm still working on whether/how I should show that information to the user. Basically OP will initially send messages at 500ms, OctoFarm now checks the response time on these, and with a lot of network congestion or some wifi setups if it's slow it will adjust the rate. This will result in temps and stuff not coming back as quick, but it's only on information not commands, just to mention anyway if you notice messages coming back slow. I've done a lot of work on trying to make sure people are aware of network issues on these next few versions.

closed 2022-04-11 by NotExpectedYet