OctoFarm/OctoFarm #901: docker-entrypoint.sh permission denied

watzon, 2022-04-25:
### OctoFarm Version

1.2.0-alpha.4

### OctoPrint Version(s)

layest

### pm2 or docker?

docker

### Operating System

Raspberry Pi

### Location of the problem

Server

### Description of the problem

```bash
When trying to pull and run the latest alpha (which is required since the alphas are the only docker containers that support running on ARM) OctoFarm fails to start due to a permission error with `docker-entrypoint.sh`. When checking the Dockerfile [here](https://hub.docker.com/layers/octofarm/octofarm/1.2.0-alpha.4/images/sha256-b19b1ea2e68c46386d53645c9966a2fe82f2c1b920cf8a5f6bdf7609758f04d0?context=explore) I can see that there is no 'chmod` being ran on the entrypoint script.
```

### Steps taken to reproduce

```bash
Use the following docker-compose.yml on a Raspberry Pi

version: '2.4'

services:
  octofarm:
    container_name: octofarm
    image: octofarm/octofarm:1.2.0-alpha.4
    restart: always
    mem_limit: 650m # Feel free to adjust! 400 MB is quite high and a safety limit.
    ports:
      - 8082:4000 # port of SYSTEM : port of CONTAINER
    volumes:
    # Volumes as local relative folders (validate with 'docker-compose config')
      - octofarm:/app
    environment:
      - MONGO=mongodb://mongo:password@mongodb:27017/octofarm?authSource=admin
    depends_on:
      - mongodb
  mongodb:
    image: arm64v8/mongo:4.2
    restart: always
    environment:
      MONGO_INITDB_ROOT_USERNAME: mongo
      MONGO_INITDB_ROOT_PASSWORD: password

Attempting to start it with `docker-compose up` should fail.
```

### What did you expect to happen

```bash
It should have worked
```

NotExpectedYet, 2022-04-26:
For some reason it seems my alpha tags are not updating on docker. 

You can use the :development / :dev tags for the latest alpha work. It's currently on version alpha.24. I "think" that issue should be resolves. If it isn't let me know and I'll double check.

watzon, 2022-04-26:
I actually tried dev first, and then switched to the alpha tags when that one didn’t work

NotExpectedYet, 2022-04-26:
Must have been broken for a fair while then, will take a look. Thanks for the report! 

If you know of a fix and fancy doing a PR feel free as I'll be at work majority of the week.

watzon, 2022-04-26:
I’ll see if I can figure anything out for sure, but I just started a new job myself so I can’t guarantee anything :D

Thanks!

NotExpectedYet, 2022-04-26:
> I’ll see if I can figure anything out for sure, but I just started a new job myself so I can’t guarantee anything :D Thanks!
> […](#)

Absolutely no worries xD enjoy the new venture

NotExpectedYet, 2022-04-26:
I'm hoping I've fixed it with this: #902

NotExpectedYet, 2022-04-26:
https://github.com/OctoFarm/OctoFarm/pull/902/commits/aaf4ec2e6992b55b38b01a388c7b1139ea24c9b7

This specifically

closed 2022-05-19 by NotExpectedYet