OctoFarm/OctoFarm #905: Web Interface slowing down and losing connection

L0rd0fCha0s, 2022-04-27:
### OctoFarm Version

1.1.13-hotfix

### OctoPrint Version(s)

1.7.3

### pm2 or docker?

pm2

### Operating System

FarmPi, Ubuntu

### Location of the problem

Web Interface

### Description of the problem

```bash
Octofarm web interface gets progressively slower after adding printers, after the 4th printer being added the web interface will get to the point where it will take many minutes for the web interface to display and progress further to where it will just crash and lose connection and fail to reconnect. We suspect it may be a network related issue as on occasion we have noticed the ssh will temporarily freeze along side with the web interface losing connection.

We are running a Pi3b with 8 instances of Octoprint on a single Pi, this has been running for 3 years without any issues.
We have tried running Octofarm on a Pi3B, Pi3B+ and right now a Pi4 8gb.
```

### Steps taken to reproduce

```bash
We have tried the Farmpi Image as well as manually installing Ubuntu and Octofarm on the Pi and achieved the same results, works fine and gets progressively slower as we add printers, usually crashing at about 4-6 printers
```

### What did you expect to happen

```bash
I expected Octofarm to work seamlessly with 8 instances as I have read on the forums of other members being able to achieve much higher numbers of instances without having this issue.
```

### To avoid the problem I had to

```bash
...
```

NotExpectedYet, 2022-04-27:
Do you have any logs?

L0rd0fCha0s, 2022-04-28:
Which log files do you require? Ill do my best to export them

NotExpectedYet, 2022-04-28:
> Which log files do you require? Ill do my best to export them

There's a log dump on the system page if you can get to it. Packages it into a nice zip file. 

It might also be worth trying the latest development version as there's been a lot of CPU improvements. The logs will help though as a Pi4 on 1.1.13 should be able to run 8 printers, it's windows where I had slow down initially.

NotExpectedYet, 2022-05-19:
Hello, 

The master version has now been updated with all the latest development work. V1.2 basically. 

I believe this should be resolved, so I'm going to close it. Please feel free to re-open after updating if that's not the case.

closed 2022-05-19 by NotExpectedYet