OctoFarm/OctoFarm #912: V1.2 - RC8 Issues - Please report here!

NotExpectedYet, 2022-05-09:
### Description of the problem

```bash
Below are the issues found in V1.2-RC8 that are required to fix before V1.2 final release. 

- [ ] Group view is still disabled due to needing a check through. 
- [ ] Filament Manager plugin needs checking through fully
```

HappyRave, 2022-05-11:
Hi! I just discovered Octofarm and did a new install of V1.2-RC8 (it says 1.2.0-alpha.36 on the webpage). There seems to be an issue with the user registration step. Clicking on "Register" after filing all the fields disables the button, but nothing happens afterward. No console error, nothing relevant in the logs files, no user created.

NotExpectedYet, 2022-05-12:
> Hi! I just discovered Octofarm and did a new install of V1.2-RC8 (it says 1.2.0-alpha.36 on the webpage). There seems to be an issue with the user registration step. Clicking on "Register" after [...]

Thanks for the report! Will take a look.

NotExpectedYet, 2022-05-12:
> Hi! I just discovered Octofarm and did a new install of V1.2-RC8 (it says 1.2.0-alpha.36 on the webpage). There seems to be an issue with the user registration step. Clicking on "Register" after [...]

I've fixed and pushed this to development. Goofed the onSubmit, for an onClick doh!

HappyRave, 2022-05-12:
Registration working now, thank you!
[image]
It looks like I can't add any printer, the logs tell that there was an issue retrieving the user list. I tried Application API, User API and Global API (CORS enabled)

NotExpectedYet, 2022-05-13:
> Registration working now, thank you! [image] It looks like I can't add any printer, the logs tell that there was an issue retrieving the user list. I tried Application API, User API and Global API (CORS enabled)

Needs to be User or Application, global won't work for websockets. If you've one user it's probably a timeout issue, are they on WiFi? You can increase it in System -> Server settings. 
[image]

HappyRave, 2022-05-13:
Clicking the "system" tab seems to just load indefinitely. I changed the settings directly in MongoDB and now acquiring the user list no longer timeouts, but return a "FetchError: request to http://octoprusa1.local/api/users failed, reason: getaddrinfo EAI_AGAIN octoprusa1.local".
I did try the url in the browser (same Windows 10 PC runing Octofarm) and get a succesful response from Octoprint.

NotExpectedYet, 2022-05-14:
> Clicking the "system" tab seems to just load indefinitely. I changed the settings directly in MongoDB and now acquiring the user list no longer timeouts, but return a "FetchError: request to http://octoprusa1.local/api/users failed, reason: getaddrinfo EAI_AGAIN octoprusa1.local". I did try the url in the browser (same Windows 10 PC runing Octofarm) and get a succesful response from Octoprint.
> 
> Edit: using the IP address instead of the network name fixed this issue, so I guess there's an issue with DNS resolution

No trouble, yeah was about to suggest. mDNS doesn't work everywhere and I need to put a note about not using it as I've seen all sorts of issues caused by it. Would you mind getting me the logs? System tab shouldn't be failing and I'm wondering what's going on there as I couldn't reproduce with a fresh install.

NotExpectedYet, 2022-05-14:
I am going to swap the timeout settings though. Those are literally for offline printers, and the old scan time issues that would hold up the queue with offline printers. Think I'll default that setting to max rather than a close min as that setting was hiding the error.

closed 2022-05-14 by NotExpectedYet