OctoFarm/OctoFarm #939: Base URL support

oscarsiles, 2022-05-14:
### OctoFarm Version

1.1.13

### OctoPrint Version(s)

1.7.3

### pm2 or docker?

pm2

### Operating System

Raspberry Pi

### Location of the problem

Server

### Description of the problem

```bash
I currently host my various OctoPrint instances and OctoFarm behind different base urls as I am unable to assign different subdomains. This works fine for OctoPrint with the X-Script-Name header, but for OctoFarm a number of URLs are hardcoded to begin with / and therefore always attempt to find assets from the root folder of the subdomain.
```

### Steps taken to reproduce

```bash
For subdomain.example.com:
1. Set up a reverse proxy with a location of "/octofarm"
2. Go to subdomain.example.com/octofarm
3. The webpage will not render properly as links point to the root subdomain.example.com (e.g. subdomain.example.com/assets/, instead of subdomain.example.com/octofarm/assets/)
```

### What did you expect to happen

```bash
Assets and links take into account the base url correctly, i.e. subdomain.example.com/octofarm/assets/ instead of subdomain.example.com/assets/
```

### To avoid the problem I had to

```bash
Modify the templates to remove the leading / from href/src/action attributes.
```

NotExpectedYet, 2022-05-17:
Maybe a while before I personally get to this because I don't use this setup myself. My reverse proxies are all setup under subdomains. 

Happy for a PR.

NotExpectedYet, 2022-06-22:
Going to close this as not planned, because it requires some heafty server and client work adding like defining a base url and updating the client support. I may re-look at this sometime in the future but until the client re-working is done (in progress) then it's not really feasable

closed 2022-06-22 by NotExpectedYet