OctoFarm/OctoFarm #948: TypeError when opening the dashboard in version 1.2.0

EricTSten, 2022-05-18:
### OctoFarm Version

1.2.0

### OctoPrint Version(s)

1.8.0

### pm2 or docker?

pm2

### Operating System

Ubuntu

### Location of the problem

Web Interface

### Description of the problem

```bash
When opening up the dashboard view this error occurs.

TypeError ERROR (undefined):
Please report this error to OctoFarm Issues!
t[0] is null
FILE INFO
LINE: 2
COL: 33821
FILE:
http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js
STACK:
value@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:33821 value@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:482260 value/<@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:5949 value/<@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:488159 value@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:488087 value/<@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:385883 value@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:385542 value@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:487274 ct@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:635695 updateCurrentUtilisation@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:637115 currentStatusAndUtilisation@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:639297 @http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:665550 ft/ut.onmessage@http://octofarm.deamon.local/assets/dist/dashboard.runner.min.js:2:637419
```

### Steps taken to reproduce

```bash
1.Open a browser window to the OctoFarm dashboard
2.
3.
...
```

### What did you expect to happen

```bash
No error to occur
```

### To avoid the problem I had to

```bash
Have the printers configured in OctoFarm online and reachable.  The error only occurs when the printers are not reachable.
```

EricTSten, 2022-05-18:
[octofarm_dump-5-18-2022-1-54-04 PM.zip](https://github.com/OctoFarm/OctoFarm/files/8718933/octofarm_dump-5-18-2022-1-54-04.PM.zip)

NotExpectedYet, 2022-05-18:
Thanks, will take a look in the morning for you.

NotExpectedYet, 2022-05-19:
If you `git pull` and then `pm2 restart OctoFarm` should be fixed

closed 2022-05-19 by NotExpectedYet

NotExpectedYet, 2022-05-19:
Hello, 
the master branch is updated with the latest development changes now. 

I will formalize the release over the weekend. 

If you'd like to get back on the master branch to stay with stable releases then do the below:
`git checkout master`
`git pull`
`pm2 restart OctoFarm`

EricTSten, 2022-05-19:
It is all good now both on the dev and the master branches.  Thanks!

NotExpectedYet, 2022-05-19:
Absolutely epic! Thanks for replying.