OctoFarm/OctoFarm #959: Updating to V1.2

NotExpectedYet, 2022-05-19:
### OctoFarm Version

1.2

### OctoPrint Version(s)

1.8

### pm2 or docker?

pm2

### Operating System

Ubuntu

### Location of the problem

Server

### Description of the problem

```markdown
To
```

### Steps taken to reproduce

Do

### What did you expect to happen

To

NotExpectedYet, 2022-05-20:
If your here because you used the auto updater inside the application then do the following to restore your service to working order: 
Change into the OctoFarm directory
`pm2 delete OctoFarm`
`npm start` 

You will again have to re-generate the pm2 start up script mentioned above.

IHVD, 2022-06-09:
Let's say I did use the in-app update button and am now facing 
[image]
and 
[image]
what can I do to get the farm up and running again?
can I overwrite this installation via the cli commands?

NotExpectedYet, 2022-06-09:
> edit: i'm blind, octofarm is running again, thanks for the great tool :D

aha no worries and most welcome :D

closed 2022-08-06 by NotExpectedYet