OctoFarm/OctoFarm #962: System page will not load v1.2

e1miran, 2022-05-20:
### OctoFarm Version

1.2

### OctoPrint Version(s)

1.80

### pm2 or docker?

docker

### Operating System

Debian

### Location of the problem

Web Interface

### Description of the problem

```markdown
I'm having the issue with failing to acquire user list as described here: https://github.com/OctoFarm/OctoFarm/issues/912. However, I cannot access the System tab in order to increase the timeout. Clicking on the System link just tries to load the page indefinitely.
```

### Steps taken to reproduce

Clicked on System link

### What did you expect to happen

Open System page

### To avoid the problem I had to

No workaround

NotExpectedYet, 2022-05-20:
> ### OctoFarm Version
> 
> 1.2
> ### OctoPrint Version(s)
> 
> 1.80
> ### pm2 or docker?
> 
> docker
> ### Operating System
> 
> Debian
> ### OctoPrint Plugins
> 
> _No response_
> ### Location of the problem
> 
> Web Interface
> ### Description of the problem
> 
> ```gfm
> I'm having the issue with failing to acquire user list as described here: https://github.com/OctoFarm/OctoFarm/issues/912. However, I cannot access the System tab in order to increase the timeout. Clicking on the System link just tries to load the page indefinitely.
> ```
> 
> ### Steps taken to reproduce
> 
> Clicked on System link
> ### What did you expect to happen
> 
> Open System page
> ### To avoid the problem I had to
> 
> No workaround
Cheers for the report will sort it. It's because there's 0 printers for some reason.

NotExpectedYet, 2022-05-20:
Alternatively you could try access your database with mongo compass and edit it there. I should have it sorted for tomorrow though.

e1miran, 2022-05-20:
Thank you. Just an update. I managed to increase the timeout using mongo shell. This allowed the printers to connect successfully and I was able to configure them. They appear as online and operational. However, System Page still does not load.

[image]

NotExpectedYet, 2022-05-20:
Do you have access to the logs folder at all? It's not what I originally thought. 

I'll setup a new install in the morning to double check what's going on as well like.

e1miran, 2022-05-20:
I'll check. Do you know where it's located on a Docker install?

NotExpectedYet, 2022-05-20:
Should be /data/logs if monolithic or app/logs if normal I think.

e1miran, 2022-05-20:
Which of the logs do you need to see?

NotExpectedYet, 2022-05-20:
pm2.error.log would be best start as that should show anything I didn't catch.

e1miran, 2022-05-20:
This appears to be from the last restart. I did attempt the the System page after restart.

`2022-05-20T13:48:42: GitError: fatal: Not a git repository (or any of the parent directories): .git
2022-05-20T13:48:42: 
2022-05-20T13:48:42:     at Object.action (/app/server/node_modules/simple-git/dist/cjs/index.js:1194:25)
2022-05-20T13:48:42:     at PluginStore.exec (/app/server/node_modules/simple-git/dist/cjs/index.js:1229:29)
2022-05-20T13:48:42:     at /app/server/node_modules/simple-git/dist/cjs/index.js:1590:43
2022-05-20T13:48:42:     at new Promise (<anonymous>)
2022-05-20T13:48:42:     at GitExecutorChain.handleTaskData (/app/server/node_modules/simple-git/dist/cjs/index.js:1588:16)
2022-05-20T13:48:42:     at GitExecutorChain.<anonymous> (/app/server/node_modules/simple-git/dist/cjs/index.js:1572:44)
2022-05-20T13:48:42:     at Generator.next (<anonymous>)
2022-05-20T13:48:42:     at fulfilled (/app/server/node_modules/simple-git/dist/cjs/index.js:55:24) {
2022-05-20T13:48:42:   task: {
2022-05-20T13:48:42:     format: 'utf-8',
2022-05-20T13:48:42:     commands: [ 'branch', '-v', '-a' ],
2022-05-20T13:48:42:     parser: [Function: parser]
2022-05-20T13:48:42:   }
2022-05-20T13:48:42: }
2022-05-20T13:52:20: GitError: fatal: Not a git repository (or any of the parent directories): .git
2022-05-20T13:52:20: 
2022-05-20T13:52:20:     at Object.action (/app/server/node_modules/simple-git/dist/cjs/index.js:1194:25)
2022-05-20T13:52:20:     at PluginStore.exec (/app/server/node_modules/simple-git/dist/cjs/index.js:1229:29)
2022-05-20T13:52:20:     at /app/server/node_modules/simple-git/dist/cjs/index.js:1590:43
2022-05-20T13:52:20:     at new Promise (<anonymous>)
2022-05-20T13:52:20:     at GitExecutorChain.handleTaskData (/app/server/node_modules/simple-git/dist/cjs/index.js:1588:16)
2022-05-20T13:52:20:     at GitExecutorChain.<anonymous> (/app/server/node_modules/simple-git/dist/cjs/index.js:1572:44)
2022-05-20T13:52:20:     at Generator.next (<anonymous>)
2022-05-20T13:52:20:     at fulfilled (/app/server/node_modules/simple-git/dist/cjs/index.js:55:24) {
2022-05-20T13:52:20:   task: {
2022-05-20T13:52:20:     format: 'utf-8',
2022-05-20T13:52:20:     commands: [ 'branch', '-v', '-a' ],
2022-05-20T13:52:20:     parser: [Function: parser]
2022-05-20T13:52:20:   }
2022-05-20T13:52:20: }
2022-05-20T13:56:07: GitError: fatal: Not a git repository (or any of the parent directories): .git
2022-05-20T13:56:07: 
2022-05-20T13:56:07:     at Object.action (/app/server/node_modules/simple-git/dist/cjs/index.js:1194:25)
2022-05-20T13:56:07:     at PluginStore.exec (/app/server/node_modules/simple-git/dist/cjs/index.js:1229:29)
2022-05-20T13:56:07:     at /app/server/node_modules/simple-git/dist/cjs/index.js:1590:43
2022-05-20T13:56:07:     at new Promise (<anonymous>)
2022-05-20T13:56:07:     at GitExecutorChain.handleTaskData (/app/server/node_modules/simple-git/dist/cjs/index.js:1588:16)
2022-05-20T13:56:07:     at GitExecutorChain.<anonymous> (/app/server/node_modules/simple-git/dist/cjs/index.js:1572:44)
2022-05-20T13:56:07:     at Generator.next (<anonymous>)
2022-05-20T13:56:07:     at fulfilled (/app/server/node_modules/simple-git/dist/cjs/index.js:55:24) {
2022-05-20T13:56:07:   task: {
2022-05-20T13:56:07:     format: 'utf-8',
2022-05-20T13:56:07:     commands: [ 'branch', '-v', '-a' ],
2022-05-20T13:56:07:     parser: [Function: parser]
2022-05-20T13:56:07:   }
2022-05-20T13:56:07: }
2022-05-20T13:56:45: GitError: fatal: Not a git repository (or any of the parent directories): .git
2022-05-20T13:56:45: 
2022-05-20T13:56:45:     at Object.action (/app/server/node_modules/simple-git/dist/cjs/index.js:1194:25)
2022-05-20T13:56:45:     at PluginStore.exec (/app/server/node_modules/simple-git/dist/cjs/index.js:1229:29)
2022-05-20T13:56:45:     at /app/server/node_modules/simple-git/dist/cjs/index.js:1590:43
2022-05-20T13:56:45:     at new Promise (<anonymous>)
2022-05-20T13:56:45:     at GitExecutorChain.handleTaskData (/app/server/node_modules/simple-git/dist/cjs/index.js:1588:16)
2022-05-20T13:56:45:     at GitExecutorChain.<anonymous> (/app/server/node_modules/simple-git/dist/cjs/index.js:1572:44)
2022-05-20T13:56:45:     at Generator.next (<anonymous>)
2022-05-20T13:56:45:     at fulfilled (/app/server/node_modules/simple-git/dist/cjs/index.js:55:24) {
2022-05-20T13:56:45:   task: {
2022-05-20T13:56:45:     format: 'utf-8',
2022-05-20T13:56:45:     commands: [ 'branch', '-v', '-a' ],
2022-05-20T13:56:45:     parser: [Function: parser]
2022-05-20T13:56:45:   }
2022-05-20T13:56:45: }
2022-05-20T13:59:54: GitError: fatal: Not a git repository (or any of the parent directories): .git
2022-05-20T13:59:54: 
2022-05-20T13:59:54:     at Object.action (/app/server/node_modules/simple-git/dist/cjs/index.js:1194:25)
2022-05-20T13:59:54:     at PluginStore.exec (/app/server/node_modules/simple-git/dist/cjs/index.js:1229:29)
[... 151 lines cut ...]
2022-05-20T14:50:20:     at GitExecutorChain.<anonymous> (/app/server/node_modules/simple-git/dist/cjs/index.js:1572:44)
2022-05-20T14:50:20:     at Generator.next (<anonymous>)
2022-05-20T14:50:20:     at fulfilled (/app/server/node_modules/simple-git/dist/cjs/index.js:55:24) {
2022-05-20T14:50:20:   task: {
2022-05-20T14:50:20:     format: 'utf-8',
2022-05-20T14:50:20:     commands: [ 'branch', '-v', '-a' ],
2022-05-20T14:50:20:     parser: [Function: parser]
2022-05-20T14:50:20:   }
2022-05-20T14:50:20: }
[... cut ...]
2022-05-20T14:50:30:     at GitExecutorChain.<anonymous> (/app/server/node_modules/simple-git/dist/cjs/index.js:1572:44)
2022-05-20T14:50:30:     at Generator.next (<anonymous>)
2022-05-20T14:50:30:     at fulfilled (/app/server/node_modules/simple-git/dist/cjs/index.js:55:24) {
2022-05-20T14:50:30:   task: {
2022-05-20T14:50:30:     format: 'utf-8',
2022-05-20T14:50:30:     commands: [ 'branch', '-v', '-a' ],
2022-05-20T14:50:30:     parser: [Function: parser]
2022-05-20T14:50:30:   }
2022-05-20T14:50:30: }`

NotExpectedYet, 2022-05-20:
Hmm one of the tasks cheers for that. Gives me something to go on. The system page will be failing due to that I reckon.

e1miran, 2022-05-20:
I managed to get a full log dump from Portainer and attach it here.

[_octofarm_logs.txt](https://github.com/OctoFarm/OctoFarm/files/8743478/_octofarm_logs.txt)

NotExpectedYet, 2022-05-20:
> I managed to get a full log dump from Portainer and attach it here.
> 
> [_octofarm_logs.txt](https://github.com/OctoFarm/OctoFarm/files/8743478/_octofarm_logs.txt)

Brilliant thank you! 

Could you do me a favour and check the connection log on printer manage please? I'd like to know the response time your /api/user endpoint is returning in. I'm going to increase the default so other users don't run into the same user issue.

NotExpectedYet, 2022-05-20:
Settings as well actually as that was failing to grab initially even at 5 seconds by the looks. Seems I underestimated those response times a lot!

e1miran, 2022-05-20:
These are just the UI logs from the menu on the Manage button?

e1miran, 2022-05-20:
Not sure which logs you mean. Maybe this will help??

[image]

NotExpectedYet, 2022-05-20:
It's on the top bar called "Connection Overview" sorry. If you expand a printer it will show you some endpoint URLs like /API/users and a milliseconds time next to it. I believe 10 seconds should be safe bet but just want to confirm.

NotExpectedYet, 2022-05-20:
Ledge! Thanks, perfect.

e1miran, 2022-05-20:
I'm using Rpi Zero 2's. Maybe they respond a bit slower.

NotExpectedYet, 2022-05-20:
Dead on 5s huh, I'll go safer for the defaults to cover different setups. Thanks again, should have the system page fixed tomorrow morning GMT.

e1miran, 2022-05-20:
You are the man! A steak is in order.

NotExpectedYet, 2022-05-20:
Could be the case yeah, the settings endpoint is rather intensive at times.

e1miran, 2022-05-20:
I had increased it the timeout to 5s, and it was doing well. But then I had problems again and raised it to 10s.

NotExpectedYet, 2022-05-20:
I'm going to add a specific timeout for those two endpoints as it seems they are the problematic ones. Then the other settings don't need to be touched. 

It should only affect adding the printers though.

NotExpectedYet, 2022-05-20:
I'll dig through your logs properly in the morning anyway and see if there's anything else I've missed :) helps loads so thanks again.

e1miran, 2022-05-20:
Sure thing.

NotExpectedYet, 2022-05-21:
Both the issues you reported should be now fixed with the latest update

rogodra, 2022-05-21:
In my case, the problem persists with version 1.21, there is no way to enter System.

NotExpectedYet, 2022-05-21:
Please post more logs because I confirmed myself this morning the other issue had gone so there is likely something else. 

If it doesn't docker as well it needs a new ticket opening please.

e1miran, 2022-05-21:
The issue still persists with me (OP), as well. I've attached my latest logs below.

On a related note, I appear to have successfully updated and  run the latest version because [my other issue regarding iframes](https://github.com/OctoFarm/OctoFarm/issues/965) has been resolved. However, maybe a version indicator would be beneficial in the header or footer of the dashboard page since it's only in the System page.

[_octofarm_logs (1).txt](https://github.com/OctoFarm/OctoFarm/files/8747941/_octofarm_logs.1.txt)

NotExpectedYet, 2022-05-21:
> The issue still persists with me (OP), as well. I've attached my latest logs below.
>
> On a related note, I appear to have successfully updated and run the latest version because [my other issue [...]

Yeah there is one but it only displays on development versions currently. I'll enable it for any version as it is handy as you say. 

Looking at the logs it's the same error, let me check what's going on and get back to you both

NotExpectedYet, 2022-05-21:
Wow... sorry coding whilst tired this morning I seem to have never pushed the actual fix I made -_- will do a 1.2.2 now shortly and move the version to the top bar as well.

NotExpectedYet, 2022-05-21:
Done! Will build the images now and do a release when they're done so when this closes your safe to pull

[image]

e1miran, 2022-05-21:
> Wow... sorry coding whilst tired this morning I seem to have never pushed the actual fix I made -_- will do a 1.2.2 now shortly and move the version to the top bar as well.

I had figured as much. I do some coding and have had this happen as well.

NotExpectedYet, 2022-05-21:
> > Wow... sorry coding whilst tired this morning I seem to have never pushed the actual fix I made -_- will do a 1.2.2 now shortly and move the version to the top bar as well.
> 
> I had figured as much. I do some coding and have had this happen as well.

Easily done sometimes. Didn't pay attention to which branch I was on. 

Should be all live now :) I'll leave this open until confirmed.

e1miran, 2022-05-21:
SUCCESS! System page is accessible and version numbering indicator appears on the header bar. Thanks.

closed 2022-05-21 by e1miran

NotExpectedYet, 2022-05-21:
Welcome! 

Thanks for updating

rogodra, 2022-05-21:
This latest version fixes the problem. Thanks!

NotExpectedYet, 2022-05-22:
Most welcome :)

commit 65e7ae07 (2022-05-21): fix(client): fixed #962
commit aaffb6cd (2022-05-21): release(main): release 1.2.1 (#967)

* chore(main): initial v1.2.1 prep

* chore(main): initial v1.2.1 prep

* fix(server): added an amended timeout time to certain octoprint endpoints that are reportedly slow (#968)

* chore(main): clean up downloads on releases in readme

* chore(main): add an increased timeout to slow timeout calls fixes #963

* chore(docker): update monolithic mongo version to 4.4.14 (#969)

* fix(docker): fix docker containers not starting (#971)

* fix(docker): fix docker containers not starting

* chore(main): update changelog

* chore(docker): fix build scripts

* chor [... cut]