OctoFarm/OctoFarm #979: Monolithic Docker fails at start

taker218, 2022-05-22:
### OctoFarm Version

monolithic-latest

### OctoPrint Version(s)

1.8

### pm2 or docker?

docker

### Operating System

Debian

### Location of the problem

Server

### Description of the problem

```markdown
I just updated to the latest docker image since I saw that there is a new one, now the container always gets an error.

`14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:     at NativeConnection.Connection.openUri (/app/server/node_modules/mongoose/lib/connection.js:807:32)
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51: 22/05/2022, 12:05:51 | ERROR | OctoFarm-Server  
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:  ERROR MESSAGE: MongooseServerSelectionError: connect ECONNREFUSED 127.0.0.1:27017
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:     at /app/server/node_modules/mongoose/lib/index.js:342:10
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:     at /app/server/node_modules/mongoose/lib/helpers/promiseOrCallback.js:32:5
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:     at new Promise (<anonymous>)
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:     at Mongoose._promiseOrCallback (/app/server/node_modules/mongoose/lib/index.js:1181:10)
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:     at promiseOrCallback (/app/server/node_modules/mongoose/lib/helpers/promiseOrCallback.js:31:10)
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:     at Mongoose.connect (/app/server/node_modules/mongoose/lib/index.js:341:20)
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:     at Object.<anonymous> (/app/server/app.js:42:6)
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:     at Module._compile (internal/modules/cjs/loader.js:1063:30)
14:05:51 0|OctoFarm  | 2022-05-22T14:05:51:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10) 
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19: MongoServerSelectionError: connect ECONNREFUSED 127.0.0.1:27017
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:     at Timeout._onTimeout (/app/server/node_modules/mongodb/lib/sdam/topology.js:318:38)
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:     at listOnTimeout (internal/timers.js:554:17)
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:     at processTimers (internal/timers.js:497:7) {
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:   reason: TopologyDescription {
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:     type: 'Unknown',
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:     servers: Map(1) { '127.0.0.1:27017' => [ServerDescription] },
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:     stale: false,
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:     compatible: true,
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:     heartbeatFrequencyMS: 10000,
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:     localThresholdMS: 15,
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:     logicalSessionTimeoutMinutes: undefined
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:   },
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:   code: undefined,
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19:   [Symbol(errorLabels)]: Set(0) {}
14:06:19 0|OctoFarm  | 2022-05-22T14:06:19: }`

I also had a look at issue 964, but I can't downgrade the docker container since I can't seem to find a version from the monolithic one which is old enough to perform the steps.
And I don't really know where to perfom those steps. Inside the container?
```

### Steps taken to reproduce

Updated the docker image to the latest monolithic version

### What did you expect to happen

OctoFarm starting without a problem

NotExpectedYet, 2022-05-22:
You don't need to downgrade the container, the steps spin up a secondary mongodb container that backs up your old database, and imports it ready for v4.4. You may then start the original octofarm-monolithic container

NotExpectedYet, 2022-05-22:
You perform the steps on your hosts cli

NotExpectedYet, 2022-05-22:
Let me know if you need further help with it, it's not an ideal situation but was a bit stuck with that container and how it's configured.

taker218, 2022-05-22:
I'll try later and let you know how it went.

NotExpectedYet, 2022-05-22:
> I'll try later and let you know how it went.

No worries, I've had someone already rescue his borked container on UnRaid so should be fine. Just a bit verbose. Won't happen again either as I'm aware of the issue at least now and can take mitigation steps when I move over to mongodb v5.

taker218, 2022-05-22:
It's working again, thanks for the help!

Didn't see the link in the first comment on the issue and was a bit lost.

I'll close this issue here.

closed 2022-05-22 by taker218

NotExpectedYet, 2022-05-22:
> It's working again, thanks for the help!
> 
> Didn't see the link in the first comment on the issue and was a bit lost.
> 
> I'll close this issue here.

No trouble glad it's working at least. I'll get to the other issue with spools through the week.