OctoFarm/OctoFarm #980: Filament Page Does not seem to be working

Ronin0000, 2022-05-22:
### OctoFarm Version

1.2.2

### OctoPrint Version(s)

1.8.0

### pm2 or docker?

pm2

### Operating System

Raspberry Pi OS

### OctoPrint Plugins

none

### Location of the problem

Web Interface

### Description of the problem

```markdown
When I enter the filament page on my webpage, It shows this error:
TypeError ERROR (undefined):
Please report this error to OctoFarm Issues!
Cannot read properties of undefined (reading 'data')
FILE INFO
LINE: undefined
COL: undefined
STACK:
TypeError: Cannot read properties of undefined (reading 'data') at http://farmpi.local/assets/dist/filament-manager.runner.min.js:2:578306 at async http://farmpi.local/assets/dist/filament-manager.runner.min.js:2:578202
It also shows this error when I try to add filament or do anything related to the filament.
```

### Steps taken to reproduce

1.GO to the filament page on 1.2.2 and it will show this:
[image]
2.Then close the error and try to add filament. Then it will show this error again.
See the video below:
https://user-images.githubusercontent.com/89286474/169701224-9a0c8b2a-2817-4a3d-ab2b-e89ff07033f2.mov
Log Dump:
[octofarm_dump-5-22-2022-2-48-58 PM.zip](https://github.com/OctoFarm/OctoFarm/files/8749651/octofarm_dump-5-22-2022-2-48-58.PM.zip)

### What did you expect to happen

I would expect for no error to pop up, and to be able to add spool profiles, and spools, and for the Filament Statistics to not be NaN%:
[image]

### To avoid the problem I had to

There was nothing I could do to get it to work and whenever I tried to add filament spool or go to the page it popped up the error.

taker218, 2022-05-22:
I get the same error and it seems like I can't assign a spool to my printers.
The dropdown is just empty.

NotExpectedYet, 2022-05-22:
Are either of you using filament manager plugin? 

Or was it just spools loaded from before the upgrade?

NotExpectedYet, 2022-05-22:
Or 0 spools at all aha. I'll check the logs through the week anyway at some point going to be at work all week so not sure when. Will push a bug fix soon as I've sorted it though.

Ronin0000, 2022-05-22:
I am not using the filament manager plugin, thanks.

taker218, 2022-05-22:
Yes, I'm using the filament manager plugin and I have spools listed on the page, and as of now I'm not getting the error message when going to the filament page. But I got the error message before.
It's also showing me that some spools are assigned to a printer, but I can't seem to change anything on the spools itself. or is that because I have the plugin installed?

I attached my logs here as well.

[octofarm_dump-5-22-2022-8-18-58 PM.zip](https://github.com/OctoFarm/OctoFarm/files/8749996/octofarm_dump-5-22-2022-8-18-58.PM.zip)

NotExpectedYet, 2022-05-22:
@Ronin0000 is this a fresh install? 

@taker218 Can you do a resync of the spools? Might help for now. Do you mean when opening the spool manager? Just to confirm further, you have to press edit and then select the printer, then save to change there. The views should just to select. 

I'll check the logs anyway cheers for the reports.

Ronin0000, 2022-05-22:
@NotExpectedYet, yes I had some issues updating so I decided to do a fresh install of the server on my Raspberry Pi.

taker218, 2022-05-22:
The resync spool button doesn't seem to do anything for me.

I could also do a new issue for my problem since it seems to be different from the one Ronin has.

NotExpectedYet, 2022-05-22:
@Ronin0000 cheers for that think I've missed something. 

@taker218 yeah please it's specific to the filament manager plugin by the sounds.

NotExpectedYet, 2022-05-22:
@taker218 hopefully this doesn't make it worse but if it's what I think it, it should fix the issue for you, is you could try disabling the filament manager sync in settings. Give the page a refresh, and then re-enable it.  

Still open another issue like as it shouldn't be happening with old spools in the database and that needs fixing.

NotExpectedYet, 2022-05-24:
This will be on master shortly. 

Please be aware the version for the server will still state v1.2.2, and the client will now be on v1.2.2 to match. 

I'm half way through sorting out an automation for my releases and pushed all this manually to master. The CHANGELOG.md has been updated though. 

Docker builds will be shortly.

closed 2022-05-24 by NotExpectedYet

NotExpectedYet, 2022-05-24:
I'll resolve the NaN% in 1.2.3 #990

NotExpectedYet, 2022-05-24:
@Ronin0000 

`git pull`
`npm install-server`
`pm2 restart OctoFarm`

should be all you need to manually update.

commit 55c0963d (2022-05-24): fix(client): fixed #980 - filamant manager failed to load on fresh installs due to graphs
commit 6d67b386 (2022-05-24): fix(client): fixed #980 - filamant manager failed to load on fresh installs due to graphs