OctoFarm/OctoFarm #981: cant connect to printers

br3jlm, 2022-05-22:
### OctoFarm Version

1.2.2

### OctoPrint Version(s)

1.8.0

### pm2 or docker?

pm2

### Operating System

windows

### Location of the problem

Web Interface

### Description of the problem

```markdown
cant connect to printers
```

### Steps taken to reproduce

cant connect to printer

### What did you expect to happen

i be able to connect to printers and control them

### To avoid the problem I had to

[octofarm_dump-5-22-2022-11-29-36 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8749727/octofarm_dump-5-22-2022-11-29-36.AM.zip)

NotExpectedYet, 2022-05-23:
At the time of the logs though your timeout settings we're 1000ms, I think this is too short. Can you try and put this up to 10000 please?

br3jlm, 2022-05-27:
Ok it was the operator.  All I needed to do was delete all instances and add them again 🤡

closed 2022-05-27 by br3jlm

NotExpectedYet, 2022-05-27:
> Ok it was the operator. All I needed to do was delete all instances and add them again clown_face

Glad it's working!