OctoFarm/OctoFarm #982: Filament Manager connection not working after upgrade

taker218, 2022-05-22:
### OctoFarm Version

1.2.2 (monolithic docker)

### OctoPrint Version(s)

1.8

### pm2 or docker?

docker

### Operating System

debian

### OctoPrint Plugins

Filament Manager

### Location of the problem

Server

### Description of the problem

```markdown
After the upgrade the filament manager plugin connection is not working anymore.

I couldn't select a spool for my printers and the resync button didn't do anything.

As suggested in another issue I disabled the filament manager sync and now all spools are gone (which is fine for me). But as I wanted to reenable the sync nothing happens.

No big deal since I have the spools in the filament manager and just use OctoFarm to monitor all printers, but would be nice to have the spools in it as well.
```

### Steps taken to reproduce

1. Upgrade from Version below 1.2 to the latest one
2. Go to filament manager tab and resync

### What did you expect to happen

Filament Manager to work as before.

I attached the latest log dump here, maybe you can see what happens (or doesn't happen).

[octofarm_dump-5-22-2022-9-58-16 PM.zip](https://github.com/OctoFarm/OctoFarm/files/8750177/octofarm_dump-5-22-2022-9-58-16.PM.zip)
e

NotExpectedYet, 2022-05-23:
Strange the sync didn't re-enable. Cheers for the report and extra log dump though will take a look.

taker218, 2022-05-23:
alright, I just tried it from a different PC and now I could reenable the sync.

That's more then strange, but it seems to be working now for me. I attached the new log dump now, maybe it can help :)

[octofarm_dump-5-23-2022-11-57-08 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8753569/octofarm_dump-5-23-2022-11-57-08.AM.zip)

NotExpectedYet, 2022-05-23:
Thanks for the additional log dump! 

Yeah that is strange, sounds like something is been incorrectly cached. The dump wll help find it, at least it's working again and there's a work around until it's sorted. 

Did that resolve the selections on the views too?

taker218, 2022-05-23:
Yes. I can select the spools now in the views

NotExpectedYet, 2022-05-28:
Due to some limitations with supporting this plugin and some early choices I'd made with how I do the sync (grabbed from first online printer) I am going to close this as "fixed". I've added in a note to the upgrade to v1.2 issue for people to do the Disable and Enable of the Filament Manager sync to get the plugin working again. 

It would just be too much work and may cause confusion for users doing a database migration and then forcing the resync in the backend as I can't rely on a farm starting up with at least 1 online printer. No ideal but at least there is a work around for it. 

Will close, V1.2.3 should be getting pushed shortly where the rest of the issues are fixed. :)

closed 2022-05-28 by NotExpectedYet