OctoFarm/OctoFarm #983: Error When Installing OctoFarm

RonenGru, 2022-05-22:
### OctoFarm Version

1.2

### OctoPrint Version(s)

1.8

### pm2 or docker?

pm2

### Operating System

Linux Mint (Ubuntu)

### Location of the problem

Server

### Description of the problem

```markdown
I tried to install OctoFarm several times with the same problem.  I have had OctoFarm running without issue previously but with the older version of OctoFarm.

I installed a fresh installation of Linux Mint on the computer and followed all the steps at https://docs.octofarm.net/installation/install-linux-ubuntu.html.  Everything installed as expected (using terminal & SSH).  When running npm start I get the following error(s):

npm ERR! fsevents not accessible from chokidar

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/rg/.npm/_logs/2022-05-22T22_07_37_462Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! octofarm@1.2.2 prestart: `npm ci`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the octofarm@1.2.2 prestart script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/rg/.npm/_logs/2022-05-22T22_07_37_547Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! octofarm-entry@1.0.1 start: `cd server && npm run start`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the octofarm-entry@1.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/rg/.npm/_logs/2022-05-22T22_07_37_615Z-debug.log

I tried reinstalling a number of times but with the same result.  Not sure what is going on exactly or how to fix.

Any help would be appreciated.  Ronen
```

### Steps taken to reproduce

Problem always occurs with Linux Mint 20.3 using the online installation procedure found at https://docs.octofarm.net/installation/install-linux-ubuntu.html

### What did you expect to happen

I expect OctoFarm to run and to be accessible via the browser.

### To avoid the problem I had to

I was unable to avoid the problem after multiple attempts to install.

NotExpectedYet, 2022-05-23:
Update your npm
`sudo npm install -g npm@latests` is all that was needed

NotExpectedYet, 2022-05-23:
I'll add it to the instruction set

NotExpectedYet, 2022-05-23:
I don't test or support nodejs 17 just so your aware. Only LTS versions.

NotExpectedYet, 2022-05-24:
Fixed the docs, closing.

closed 2022-05-24 by NotExpectedYet

RonenGru, 2022-05-24:
Thanks!  Reinstalled with the fix.  All is working.  Excellent software.  Ronen

 
 

Closed #983 as completed.

NotExpectedYet, 2022-05-24:
Thanks for the praise!