OctoFarm/OctoFarm #984: Error on Filament page when Filament manager is not configured

nftrix, 2022-05-23:
### OctoFarm Version

latest

### OctoPrint Version(s)

1.8

### pm2 or docker?

docker

### Operating System

Windows

### Location of the problem

Web Interface

### Description of the problem

```markdown
When browsing to the Filament page I receive the following error. I do not have Filament Manager currently configured within OF or OP.

TypeError ERROR (undefined):
Please report this error to OctoFarm Issues!
Cannot read properties of undefined (reading 'data')
FILE INFO
LINE: undefined
COL: undefined
STACK:
TypeError: Cannot read properties of undefined (reading 'data') at http://192.168.50.11:4000/assets/dist/filament-manager.runner.min.js:2:578306 at async http://192.168.50.11:4000/assets/dist/filament-manager.runner.min.js:2:578202
```

### Steps taken to reproduce

Login to OF, click on Filament Manager in the toolbar

### What did you expect to happen

Nothing, an error should not be thrown but perhaps a message informing me that I have not setup Filament Manager and a link directing me to the page for the setup

nokemono42, 2022-05-24:
I just installed OctoFarm for the first time in a Docker container and I have a similar issue.

```
undefined is not an object (evaluating 'e[0].data')
FILE INFO
LINE: undefined
COL: undefined
STACK:
@http://192.168.42.10:4000/assets/dist/filament-manager.runner.min.js:2:578305 asyncFunctionResume@[native code] @[native code] promiseReactionJobWithoutPromise@[native code] promiseReactionJob@[native code]
```

NotExpectedYet, 2022-05-24:
Duplicate #980

closed 2022-05-24 by NotExpectedYet

NotExpectedYet, 2022-05-24:
This will be on master shortly. 

Please be aware the version for the server will still state v1.2.2, and the client will now be on v1.2.2 to match. 

I'm half way through sorting out an automation for my releases and pushed all this manually to master. The CHANGELOG.md has been updated though. 

Docker builds will be shortly.