OctoFarm/OctoFarm #985: Inaccurate Tooltip for "Group" link on toolbar

nftrix, 2022-05-23:
### OctoFarm Version

1.2

### OctoPrint Version(s)

1.8

### pm2 or docker?

docker

### Operating System

Windows

### Location of the problem

Web Interface

### Description of the problem

```markdown
After enabling "Group" to show up in the nav toolbar, I see the following tooltip message when I mouse over it:

"davids super power tool"

This tooltip seems inaccurate, unless you are david lol
```

### Steps taken to reproduce

enable Group in nav toolbar
mouse over Group and read tooltip

### What did you expect to happen

tooltip should tell me what group view does

NotExpectedYet, 2022-05-24:
Thanks for that, missed it!

NotExpectedYet, 2022-05-24:
This is on master

Please be aware the version for the server will still state v1.2.2, and the client will now be on v1.2.2 to match.

I'm half way through sorting out an automation for my releases and pushed all this manually to master. The CHANGELOG.md has been updated though.

commit b69e022e (2022-05-24): fix(client): fixed #985 - incorrect title tag on group view
closed 2022-05-24 by NotExpectedYet