OctoFarm/OctoFarm #995: Saving printer settings causes client error to throw due to missing "status" key

NotExpectedYet, 2022-05-25:
### OctoFarm Version

1.2.2

### OctoPrint Version(s)

Any

### pm2 or docker?

pm2

### Operating System

Ubuntu

### Location of the problem

Web Interface

### Description of the problem

```markdown
TypeError ERROR (undefined):
 Please report this error to [OctoFarm Issues](https://github.com/octofarm/octofarm/issues)!
 t.status is undefined
 FILE INFO
 LINE: 20
 COL: 39734
 FILE:
 http://192.168.178.7:4000/assets/dist/printer-management.runner.min.js
 STACK:
createServerResponseMessageForSaveAction@http://192.168.178.7:4000/assets/dist/printer-management.runner.min.js:20:39734 setupSaveButton/<@http://192.168.178.7:4000/assets/dist/printer-management.runner.min.js:20:38911
```
```

### Steps taken to reproduce

Click save on Printer Manager -> Manage -> Settings

### What did you expect to happen

Save without error

taker218, 2022-05-25:
Here's the dump from my attempt to save the power settings.

[octofarm_dump-5-25-2022-10-57-42 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8769540/octofarm_dump-5-25-2022-10-57-42.AM.zip)

NotExpectedYet, 2022-05-26:
@taker218 Looking at the logs your client seems to be using old code... Could you try clearing your browser cache and trying the saving again for me please? I just tried it here on my local version that is set to not cache on my dev browser and it didn't produce the same error anyway so worth a try. 

A nuclear option might be needed on the cache. I'm going to try and work in some cache busting with webpack because this issue has cropped up before.

NotExpectedYet, 2022-05-26:
Did you mean to send blank instructions too? 

https://github.com/jneilliii/OctoPrint-Tasmota/wiki/API
https://docs.octofarm.net/getting-started/octoprint-supported-plugins.html#tasmota

My docs and the Plugin docs both state that an object is required with the power on command and ip of the device to be added. 

Your call sent the following. It could just be the cache problem as I think I changed a few id's there. Just wanted to make sure.

```
    "powerCommands": {
        "powerOffCommand": "",
        "powerOffURL": "[PrinterURL]/api/plugin/tasmota",
        "powerOnCommand": "",
        "powerOnURL": "[PrinterURL]/api/plugin/tasmota",
        "powerStatusCommand": "",
        "powerStatusURL": "[PrinterURL]/api/plugin/tasmota",
        "powerToggleCommand": "",
        "powerToggleURL": "",
    },
    
 ```

NotExpectedYet, 2022-05-26:
Found the bug, I don't think it's cache. Sorry just noticed there was no toggle command sent. Will sort a fix for this in 1.2.3.

taker218, 2022-05-26:
I just tried it with a different browser and here it's working.

will clear the cache on my main browser and try it there again.

NotExpectedYet, 2022-05-26:
There's not much to go on with that error. I'll add something to improve that the next version. 

Is that from the console? F12?

There's a network tab near the console if you press F12. Could you open that and press the power button please? 

Screenshot the main tab on network and the "Request", "Response tabs for me? 

I have it working here with PSU control but I've never personally tried with tasmota plugin. Will help chase down what it is.

taker218, 2022-05-26:
I see where the problem is:
I connect to octofarm via reverse proxy because of https, so ressources from http-connections are blocked.

I don't know why the webcam image is shown, but once I try to power on a printer I get the error in the network tab.
But it seems like the tasmota plugin doesn't support getting the status via api, so I can only power on the printer, but not shut down.

I'll open an issue on their github for that, but I now know why I can't use the power buttons via https.

NotExpectedYet, 2022-05-26:
Ah I see, I'm surprised any commands go through at all because they should all suffer the same https -> http issue. 

Currently my commands to OctoPrint are mostly client side from legacy code when OctoFarm was a single index.html. I am hoping to finally get started on moving that to the backend then this wouldn't be a problem. 

Yeah I've just noticed on Jim's API there's no status command. I will have to update my documentation as I don't currently support any tracking of state internally to resolve this. Sure Jim will add status to the API endpoint. 

Yeah the cameras never use to work with that setup, I dunno what has changed recently but I think it might be something in the browser as I haven't changed anything and other users reported cameras working but commands not.

NotExpectedYet, 2022-05-26:
Thanks for the return. 

I'll have to make amendments to OctoFarm to support that but it's an easy fix so I'll add it to my list for 1.2.3 :)

commit bc50b12e (2022-05-26): fix(main): fixed #995 Saving printer settings without toggle command caused json parse error (#997)
commit 694e2b65 (2022-05-28): fix(main): fixed #995 Saving printer settings without toggle command caused json parse error (#997)
closed 2022-05-28 by NotExpectedYet