OctoFarm/OctoFarm #1006: Add support for "on" and "off" return from power plugins

NotExpectedYet, 2022-05-26:
### OctoFarm Version

1.2.2

### OctoPrint Version(s)

Any

### pm2 or docker?

pm2

### Operating System

Ubuntu

### OctoPrint Plugins

Lots

### Location of the problem

Web Interface

### Description of the problem

```markdown
Returns from power plugins can contain "on"/"off" not just true/false. 

State check commands need to support this return value.
```

### Steps taken to reproduce

Setup tasmota plugin

### What did you expect to happen

Check the status of printer power correctly.

taker218, 2022-05-28:
I don't know if this is related to this problem, but it seems like the checkstatus for tasmota isn't working in the newest version.

I just installed a fresh version on a Pi I had lying around and added one of my printers, set everything up.
I can turn the printer on but after some seconds I get the following error in the OctoFarm-Client.log:

`28/05/2022, 20:40:58 | ERROR | OctoFarm-Client
 ERROR MESSAGE: UNKNOWN_ERROR: SyntaxError: JSON.parse: unexpected end of data at line 1 column 1 of the JSON data : {"name":"UNKNOWN","statusCode":999,"type":"UNKNOWN"}
28/05/2022, 20:40:58 | INFO | OctoFarm-Client
 INFO MESSAGE: Developer Message:  : {"lineNumber":0,"colu
`

[octofarm_dump-5-28-2022-10-42-46 PM.zip](https://github.com/OctoFarm/OctoFarm/files/8791832/octofarm_dump-5-28-2022-10-42-46.PM.zip)
mnNumber":0,"fileName":"","stack":""}

Don't mind the error message at the start regarding the database, I had to spawn a MongoDB container because Ubuntu 22.04 doesn't support Mongodb 4.4

If I can check anything else, please let me know :)

NotExpectedYet, 2022-05-29:
Is there anything else in the console log on the network tab? 

OctoFarm actually supports all the way up to 5.0 from my testing. I just haven't gotten around to updating the documentation for that yet. 

Upgrading the mongodb database is a bit of a pain as you have to do each version incrementally. You can however use mongodump and mongorestore to dump your data out and update.

NotExpectedYet, 2022-06-05:
I believe this error is because I wasn't awaiting the .json() call. 

Should be fixed on 1.2.5 which will be releasing shortly. Please let me know if it isn't and I'll need the outputs mentioned above.

closed 2022-06-05 by NotExpectedYet

taker218, 2022-06-07:
seems to be working now, thank you for your work!

NotExpectedYet, 2022-06-07:
Epic thanks for the confirmation! Most welcome too :D

commit a8ee6cbc (2022-05-27): fix(client): fixed #1006 - support on/off text based responses for power plugins (#1007)

* fix(server): fix json parsing old commands

* fix(client): fixed #1006 - support on/off text based responses from power plugins
commit 549eea97 (2022-05-28): fix(client): fixed #1006 - support on/off text based responses for power plugins (#1007)

* fix(server): fix json parsing old commands

* fix(client): fixed #1006 - support on/off text based responses from power plugins
commit 02ad21c1 (2022-06-05): fix(client): fixes #1006 fix grabbing response from power plugins inc. on/off and true/false returns
commit 6935125c (2022-06-05): fix(client): fixes #1006 fix grabbing response from power plugins inc. on/off and true/false returns