OctoFarm/OctoFarm #1012: 401 error when adding printer

jacobguldbergdk, 2022-05-27:
### OctoFarm Version

1.2

### OctoPrint Version(s)

1.8

### pm2 or docker?

pm2

### Operating System

Windows

### Location of the problem

Web Interface

### Description of the problem

```markdown
Getting a 401 error when trying to add any printer

OctoFarm Error ERROR (401):
Please report this error to OctoFarm Issues!
Unauthorized
FILE INFO
LINE: undefined
COL: undefined
STACK:
OctoFarm Error: Unauthorized at http://192.168.1.66:4000/assets/dist/printer-management.runner.min.js:2:663488 at async sn.savePrinter (http://192.168.1.66:4000/assets/dist/printer-management.runner.min.js:20:147300)
```

### Steps taken to reproduce

1. Add printer, only enter IP + API key

Tried with multiple different printers. I assume its related to octofarm, rather than the octoprint instances

### What did you expect to happen

Printer to be added.

NotExpectedYet, 2022-05-27:
401 unauthorized is a pretty standard response. Doubt it OctoFarm specifically. 

What API key are you using?

NotExpectedYet, 2022-05-27:
Oh wait ignore that, that's my client to server... 

I've never seen that before. Can you get on the system page and do a log dump? 

How did you install OctoFarm? With the documentation steps?

jacobguldbergdk, 2022-05-27:
Used the guide yes, API is the user one. Have Admin / operatior rights

Here is a dump of all log files
https://drive.google.com/file/d/1C9pUdhAdnvQlZn9CYKWZSrQUyxalfTby/view?usp=sharing

I ordered an extra pi to install octofarm on, so dont waste to much time looking into it. Its probably realted to my install and permissions somehow

NotExpectedYet, 2022-05-27:
I'd possibly say database permissions at a guess, 401 is a weird one to throw back. 

The log dump will help though thanks! 

I'll leave this open till I can investigate further at least.

NotExpectedYet, 2022-05-28:
It's certainly something with permissions. There are a load of 404's (missing files) from the client like not been able to find style sheets or javascript files.

It's because it was installed here: "C:\Program Files\" 

On Windows if I remember rightly the nodejs process runs as the logged in user so "Program Files" would be inaccessible to it. Never seen that before, so I'll update my windows documentation to ward against that.

NotExpectedYet, 2022-05-28:
I've updated the documentations to reflect this now. Will close. Please re-open if it persists.

closed 2022-05-28 by NotExpectedYet