OctoFarm/OctoFarm #1016: Power On Command is sent when selecting file

taker218, 2022-05-28:
### OctoFarm Version

1.2.3

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu 22.04

### Location of the problem

Web Interface

### Description of the problem

```markdown
I recently saw that when I want to select a file to print, it seems like it's sending the power on command to the printer.
It's also showing the notification in the lower right corner "CR10 successfully powered on"
```

### Steps taken to reproduce

1. Open file manager on a printer which has a power on command set up
2. Select a file

### What did you expect to happen

That the file get's selected, not the power on command to fire up

### To avoid the problem I had to

Unfortunately I did not find a workaround, but here is a fresh log dump. Already tried it with different browsers, clearing cache and everything.

[octofarm_dump-5-28-2022-10-56-35 PM.zip](https://github.com/OctoFarm/OctoFarm/files/8791855/octofarm_dump-5-28-2022-10-56-35.PM.zip)

NotExpectedYet, 2022-05-29:
Hmm that should be only on pressing print, will look into that cheers.

NotExpectedYet, 2022-05-29:
For print it should only send the command if powered off as well. Gather that's related to the state not correctly been grabbed from your other issue.

commit 97d27c30 (2022-06-05): fix(client): fixes #1016 don't send connect/power command when selecting a file
commit 96cf5bf0 (2022-06-05): fix(client): fixes #1016 don't send connect/power command when selecting a file
closed 2022-06-05 by NotExpectedYet