OctoFarm/OctoFarm #1033: Docker-Compose  error on startup permission denied, open '../images/bg.jpg'

kelso6969, 2022-06-01:
### OctoFarm Version

latest

### OctoPrint Version(s)

1.8

### pm2 or docker?

docker

### Operating System

Debian

### Location of the problem

Server

### Description of the problem

```markdown
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:  INFO MESSAGE: everything good | /app/server/utils/bg_default.jpg
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25: Error: EACCES: permission denied, open '../images/bg.jpg'
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:     at Object.openSync (fs.js:497:3)
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:     at Object.writeFileSync (fs.js:1528:35)
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:     at Object.ensureBackgroundImageExists (/app/server/utils/env.utils.js:112:8)
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:     at setupEnvConfig (/app/server/app-env.js:279:12)
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:     at Object.<anonymous> (/app/server/app.js:27:3)
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:     at Module._compile (internal/modules/cjs/loader.js:1085:14)
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:     at Module.load (internal/modules/cjs/loader.js:950:32)
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:     at Function.Module._load (internal/modules/cjs/loader.js:790:12)
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:     at Object.<anonymous> (/usr/local/lib/node_modules/pm2/lib/ProcessContainerFork.js:33:23) {
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:   errno: -13,
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:   syscall: 'open',
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:   code: 'EACCES',
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25:   path: '../images/bg.jpg'
octofarm    | 14:35:25 0|OctoFarm  | 2022-06-01T14:35:25: }
octofarm    | 2022-06-01T14:35:25: PM2 log: App [OctoFarm:0] exited with code [1] via signal [SIGINT]
octofarm    | 2022-06-01T14:35:25: PM2 log: App [OctoFarm:0] will restart in 1500ms
```

### Steps taken to reproduce

Use example docker-compose file
add port under environment
docker-compose up

docker-compose file
  octofarm:
    container_name: octofarm
    # choose octofarm/octofarm:latest
    image: octofarm/octofarm:latest
    restart: unless-stopped
    links:
      - mongodb
    ports:
      - 4000:4000 # port of SYSTEM : port of CONTAINER
    environment:
      - MONGO=mongodb://root:M0nk3yp0x@mongodb:27017/octofarm?authSource=admin
      - OCTOFARM_PORT=4000
    volumes:
    # Volumes as local relative folders (validate with 'docker-compose config')
      - ./OctoFarm/logs:/app/logs
      - ./OctoFarm/scripts:/app/scripts
      - ./OctoFarm/images:/app/images
      - ./OctoFarm/server/images:/app/server/images

Note, I had to add that bottom volume as I was getting this error previously.
ctofarm    | 14:27:57 0|OctoFarm  | 2022-06-01T14:27:57: 6/1/22, 2:27:57 PM | ERROR | OF-Utils-Env
octofarm    | 14:27:57 0|OctoFarm  | 2022-06-01T14:27:57:  ERROR MESSAGE: cant find target folder... | /app/server/images

### What did you expect to happen

Server to start

NotExpectedYet, 2022-06-01:
That's due to your system permissions regarding the EACCES error. OctoFarm needs read write permissions to the directory.  

The second is a bug by the looks, that's an old path I must have missed will check that out cheers.

closed 2022-06-05 by NotExpectedYet

blathers123, 2022-07-21:
I'm running the latest docker image (1.4.3) and am running into this issue. Chmod 777-ing the OctoFarm folder did fix it for me, but for obvious reasons I'd rather not have to do this.

blathers123, 2022-07-21:
Looks like placing any jpeg named "bg.jpg" in ./OctoFarm/images will fix the issue, no CHMOD 777 needed

NotExpectedYet, 2022-07-21:
> Looks like placing any jpeg named "bg.jpg" in ./OctoFarm/images will fix the issue, no CHMOD 777 needed

If you use any of the history features for capturing images they might have issue with this. I'm going to turn the docker back to root user for next image so shouldn't be a problem going forward.

blathers123, 2022-07-21:
Sounds good, thanks for the quick reply!

commit 68a49cb5 (2022-06-05): fix(server): fixes #1033 - octofarm environment utilities still looking for old images directory