OctoFarm/OctoFarm #1064: printer list does not show all printers, but other screens do

theamzngq, 2022-06-06:
### OctoFarm Version

1.2.5

### OctoPrint Version(s)

1.7.3

### pm2 or docker?

pm2

### Operating System

ubuntu

### Location of the problem

Web Interface

### Description of the problem

```markdown
panel, list, camera, most places show all 10 of my printers except when I click printers in the upper right corner, then I only see one.  I'm not seeing any kind of display filter or anything on that page.  Yet, when I click 'printer overview' or 'connections overview', I see all the printers.
```

### Steps taken to reproduce

click 'printers' in upper right corner
only 1 of 10 printers appear on that page

### What did you expect to happen

I thought I'd see all printers

theamzngq, 2022-06-06:
[image]
[image]
[image]

I'm not able to see all printers even to remove them and try to re-add them.  Re-connect sockets doesn't seem to change anything either.

As a side note, not sure why some printers are showing offline, they are actually online and OctoPrint shows them as online...could this be an API issue?

theamzngq, 2022-06-06:
[octofarm_dump-6-6-2022-12-18-37 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8842539/octofarm_dump-6-6-2022-12-18-37.AM.zip)

NotExpectedYet, 2022-06-06:
It's failing to read some information somewhere in the chain but I'm not entirely sure why, it looks to have scanned it all properly. 

Can you go into system and increase your Api timeout setting to 5000+ and see if it helps? You'll have to restart the OctoFarm service afterwards just to get it to do a fresh information grab. I'm a bit confused why that's happening as it shouldn't be progressing without that data but somethings not right. Sadly you won't be able to trigger a connection reset without them showing in the Printer Manager. 

I won't get much chance to have a proper look until late in the week, but I'll sort a bug fix out once I do. 

Is this one of your offline printer urls?
 ERROR MESSAGE: Failed to fetch! : {"message":"request to http://192.168.0.64/api/login?passive=true failed, reason: socket hang up","type":"system","errno":"ECONNRESET","code":"ECONNRESET"} - if so this is down to an error on OctoPrints side, basically ECONNRESET means the other end of the connection from OctoFarm abruptly closed the connection which would cause OctoFarm to read it offline.

theamzngq, 2022-06-06:
Yes, the .64 is the printer called Mini Plus and it's RPi is online and running print jobs.  I can reboot the Pi if you think that'll help.

The max allowed in the API initial timeout is 10 seconds, so that's what I set it at.

[image]

theamzngq, 2022-06-06:
the strange thing is that the .64 printer seems to be connecting fine according to the 'connections overview' page
[image]

theamzngq, 2022-06-06:
I did the sequence you mentioned, restarting the OctoFarm service, printer manager > recollect info > restart octofarm again and initially when I went in to "printers", they were all there.
[image]

But then I clicked Printers again (top right) and it went back to the single printer showing:
[image]

No worries on the time frame, you don't owe me a thing :-)  The reverse is true...

NotExpectedYet, 2022-06-07:
I'm now wondering if the errors are registering correctly on that connection tab page, I've never actually been able to force a ECONRESET on OP's side so this is the first i'm encountering it. A restart is always a good thing I suppose, I would maybe try it in safe mode and see if it's a plugin causing it. 

That's extremely weird that it worked then vanished again... Thank you for trying that and the additional information, will hopefully help track down what is actually going on. Do you have access to the environment file? It's a `.env` file inside OctoFarm folder. 

If you could add this to it, `LOG_LEVEL=debug` and then restart OctoFarm, it will look something like the below:
```
NODE_ENV=production
MONGO=mongodb://127.0.0.1:27017/octofarm
OCTOFARM_PORT=4000
SUPER_SECRET_KEY=dXFGgU1dBDCSgkZkEOPY
LOG_LEVEL=debug
```

If you need help with linux let me know, never sure of peoples knowledge levels with cli. 

Then give it a good bit of time, try the information retrieval and send me another log dump? It will just enable some extra logging. You can turn it back off after that by removing the line or setting it back to `LOG_LEVEL=info` after getting the later log dump. 

I'm just glad people get some use out of it more than anything :) steaks always nice though ;) haha

theamzngq, 2022-06-07:
Yes, let me get into that file and do as you suggest.  I'm pretty comfortable with Linux, though not a master...

theamzngq, 2022-06-08:
Interesting thing, I just had to format one of my OctoPrints, so I removed it using the Bulk Delete since it wasn't listed on the regular screen.  That caused an error.  I missed getting a screen shot of the pop up, but here's a new log dump that maybe will shed some more light. IIRC, the error had something in the text about a "remove" function maybe?

[octofarm_dump-6-8-2022-9-36-21 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8863494/octofarm_dump-6-8-2022-9-36-21.AM.zip)
?

NotExpectedYet, 2022-06-08:
The error should be in the log dump cheers. Hmm something certainly weird going on with yours. I've probably missed something stupid in the code but even putting my connection speeds down to dial up nothing really presents the same. The printer view shows way less volatile information too. I'll take a read through that log in the morning as .remove function would only really relate to the actual printer but that shouldn't really care whether information is there or not. 

The file is a .env file so it's probably hidden by default. You may have to turn on show hidden files in the settings of your file manager.

theamzngq, 2022-06-08:
Ah, didn't think about hidden files.  I enabled it and restarted OctoFarm, then did the info retrieval again, here's the dump file.  The full list didn't show up again briefly like it did last time though
[octofarm_dump-6-8-2022-10-32-18 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8863838/octofarm_dump-6-8-2022-10-32-18.AM.zip)
.

theamzngq, 2022-06-10:
I once more did the octoprint manager > update info process, restarted the OctoPrint service and the printers all showed up in the list again.  Then all but one went away once I clicked 'printers' in the upper right.  Log attached.  I'll turn the debug logging off now
[octofarm_dump-6-10-2022-10-52-41 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8881075/octofarm_dump-6-10-2022-10-52-41.AM.zip)
.

theamzngq, 2022-06-10:
ah, looks like I already had turned off the debug level logging...

I think at this point (unless you'd like me to wait) I'll just bulk delete the printers and add them again manually and see how that works

theamzngq, 2022-06-10:
tried to bulk delete and got this pop up:
[image]

after restarting the OctoFarm service, now all printers are actually gone from all screens (printers, connections overview, printer overview, bulk delete, etc), which at least is consistent!

theamzngq, 2022-06-10:
and the log dump from just after that error screenshot
[octofarm_dump-6-10-2022-11-02-31 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8881125/octofarm_dump-6-10-2022-11-02-31.AM.zip)
t

NotExpectedYet, 2022-06-11:
> ah, looks like I already had turned off the debug level logging...
> 
> I think at this point (unless you'd like me to wait) I'll just bulk delete the printers and add them again manually and see how that works

Feel free to try, if you want to give a good try then use the existing database printer delete in System. 

Just Export your printers first, and you can re-import them. The export just grabs the details so there completely fresh. 

Yeah the .remove is a function client side, it's trying to remove a printer ID that doesn't exist. I'll put some protection around that at least as there should be, just presumed it would never hit that eventuality.

NotExpectedYet, 2022-06-11:
Do you see files listed by anychance? 

From the logs it's seeming like something with that.

NotExpectedYet, 2022-06-11:
I've looked all over the server and can't really find anything weird or out of place. There's no errors logging there either. I think it's possibly something on the client that's been missed by the logs. 

Could you open up the "Printer Manager" for me please and just double check there's nothing additional in the console log? 

You should just be able to press "f12" and then press the console tab like here: 

If "F12" doesn't work then you can right click and press "Inspect" on majority of browsers. 

[image]

NotExpectedYet, 2022-06-11:
I'm about to cut another release for 1.2.6 to fix some filament manager bugs. 

If you provide me with the information I'll happily do a 1.2.7 if we can find the problem. 

I've improved some logging and made the printer connector fail harder so hopefully whatever problem there is shows itself. Just grab me another log dump after updating to V1.2.6 :)

nokemono42, 2022-06-15:
FYI 1.2.8 fixed a similar issue to this one  I was having on 1.2.7.

theamzngq, 2022-06-15:
I did just hit the "print harvested" button on several printers and a bunch of errors popped up in the console though.  Looks mostly like "file not found" type of errors.
[192.168.0.75-1655322302783.log](https://github.com/OctoFarm/OctoFarm/files/8912525/192.168.0.75-1655322302783.log)
[octofarm_dump-6-15-2022-12-45-55 PM.zip](https://github.com/OctoFarm/OctoFarm/files/8912530/octofarm_dump-6-15-2022-12-45-55.PM.zip)
.

NotExpectedYet, 2022-06-16:
> Nothing in the console on Chrome, though I probably "fixed" the issues by removing all the printers and re-adding them. I actually did it manually instead of importing because importing is how I got the printers in to begin with, so I thought I'd do them manually to avoid any import issues

I'll double check the importer, it uses the same onboarding mechanism as manually importing so I'm not entirely sure how that could be the issue but I haven't checked that yet. 

The log dumps great thanks. 

Hmmm the file not found errors might be because it's not encoding the URI correctly. It seems to be removing "/" which it shouldn't be, gather that's why CORS is triggering too. Cheers for the extra information, hoping to cut another bug fix release soon as I've a few to deal with.

theamzngq, 2022-06-16:
I did find that the printer at 192.168.0.25 did not have CORS enabled, which I remedied after I sent those logs.

NotExpectedYet, 2022-06-16:
Oh cool that's that solved then. 

Just double check the importer and I'll consider this closed unless you have anything else?

theamzngq, 2022-06-16:
sounds good.  Too bad you couldn't come to a clear conclusion as to why I was seeing those weird issues...

I don't think I've done an update before (this current instance was from scratch); is there an "update" button somewhere in the UI?  Or can you send me to a link that details the upgrade procedure for Ubuntu?

NotExpectedYet, 2022-06-17:
> sounds good. Too bad you couldn't come to a clear conclusion as to why I was seeing those weird issues...
>
> I don't think I've done an update before (this current instance was from scratch); is there [...]

I'm proper perplexed still about it still, I'm leaning currently to a race condition in the import on the api endpoint. I didn't think there could be as it passes to an array which is cycled through by another function but I need to double check that. I'll certainly leave this open until I at least find nothing to pin it down on. 

If your on 1.2.8 I fixed the update button in System -> Server. Previous versions to that it failed to activate for an update.

[image]

NotExpectedYet, 2022-06-17:
For command line update you can just do the below from the OctoFarm folder. 
`npm run update-octofarm`
[image]

NotExpectedYet, 2022-06-17:
Just to update there's certainly a race condition happening in the printer add endpoint. I'm re-working the UI/API endpoints completely in the next minor release for the client and also implementing and API key for administrator users so I'll patch it in the next version for now and make it far more robust in that version. 

I'm also going to add some safe guarding in against CORS been un-flagged in the next patch release as well, the little Alert icon I setup does not cut it.

closed 2022-06-17 by NotExpectedYet

theamzngq, 2022-06-17:
thanks again!  Updated to 1.2.9

commit ce1b08dc (2022-06-17): fix(client): add in delay for importing printers fixed #1064
commit 44655266 (2022-06-17): fix(client): add in delay for importing printers fixed #1064