OctoFarm/OctoFarm #1083: Preheat tool 1 for second extruder warms up first extruder instead

enishi1357, 2022-06-14:
### OctoFarm Version

Latest 1.2.7

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu

### Location of the problem

Not sure

### Description of the problem

```markdown
Preheat tool 1 for second extruder warms up first extruder instead for my idex printer. No matter what tool I select 0, 1, 2, 3, etc it warms first extruder only.
```

### Steps taken to reproduce

turn on printer

connect printer

select tool 1

enter temperature

### What did you expect to happen

Select tool 1 should warm up second extruder.

NotExpectedYet, 2022-06-14:
This is going to be related to your other issue I reckon. Just add the profile details there for me please.

NotExpectedYet, 2022-06-16:
Confirmed this, the button is using the dropdown selection for the tool. Will sort it out.

closed 2022-06-17 by NotExpectedYet