OctoFarm/OctoFarm #1084: No PSU control plugin to turn on printers

enishi1357, 2022-06-14:
### OctoFarm Version

Latest 1.2.7

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu

### OctoPrint Plugins

PSU Control, PSU Control - TP link

### Location of the problem

Not sure

### Description of the problem

```markdown
No way to turn on printers remotely
```

### Steps taken to reproduce

Search around to find a way to turn on printer

end up using alexa

### What did you expect to happen

a way to turn on printers

NotExpectedYet, 2022-06-14:
Could you tell me more about what you mean please as I'm not sure here? 

https://github.com/kantlivelong/OctoPrint-PSUControl

There's instructions here how to setup the power control plugins. It's in printer manager, the "Manage" drop-down and then "Settings"

enishi1357, 2022-06-15:
Is it like this [http://192.168.50.84:5005]/api/plugin/psucontrol
I can't make it work. I tried [PrinterURL]/api/plugin/psucontrol too

enishi1357, 2022-06-15:
I followed the guide but still no dice. I'm using PSU control which in turn is using PSU Control - TPLink. 
[image]

NotExpectedYet, 2022-06-15:
> I followed the guide but still no dice. I'm using PSU control which in turn is using PSU Control - TPLink. [image]

Hmm that should be automatically picked up when you add the printer. Was your installation pre-1.2? 

Would you mind doing me a log dump from System -> Logs and uploading it here please? Sounds like a bug as you should have a little power status icon on Priinter Manager and the views. Then two new dropdowns within the little blue burger menu saying "Power on" and "Power Off".

enishi1357, 2022-06-15:
My first installation was couple months ago but then I deleted the octofarm folder and reinstall everything. The top of the site shows 1.2.7

NotExpectedYet, 2022-06-16:
> So turns out after a laptop restart it can turn on. However the blue burger button still shows Turn on instead of Turn off after turning on so I guess the status didn't change. The connect button turns on the printer but didn't connect. Connect button does disconnect though.
> 
> [pm2.out.log](https://github.com/OctoFarm/OctoFarm/files/8912085/pm2.out.log) [OctoFarm.Access.log](https://github.com/OctoFarm/OctoFarm/files/8912086/OctoFarm.Access.log) [OctoFarm.Client.log](https://github.com/OctoFarm/OctoFarm/files/8912087/OctoFarm.Client.log) [OctoFarm.Printer.OctoPrint.log](https://github.com/OctoFarm/OctoFarm/files/8912088/OctoFarm.Printer.OctoPrint.log) [OctoFarm.System.log](https://github.com/OctoFarm/OctoFarm/files/8912089/OctoFarm.System.log) [pm2.error (1).log](https://github.com/OctoFarm/OctoFarm/files/8912090/pm2.error.1.log)

Cheers for the extra info, sounds like something broke when I added support for the "on" / "off" that some other power plugins report. Will take a look.

NotExpectedYet, 2022-06-16:
Found the issue, will sort for next bug fix release

commit f1d6d3e3 (2022-06-17): fix(client): couldn't connect printers or start prints from file manager power plugin fixed #1084
closed 2022-06-17 by NotExpectedYet