OctoFarm/OctoFarm #1089: Filament Statistics - Remaining Filament Calculation Issue

SuperTrollMan, 2022-06-16:
### OctoFarm Version

1.2.8

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu 20.04.4

### OctoPrint Plugins

Bed Visualizer (1.1.1), Cancel Objects (0.4.7), Consolidated Tabs (1.0.1), Cost Estimation (3.5.0), Dashboard (1.19.8), DisplayLayerProgress Plugin (1.28.0), Filament Manager (1.9.1), Octoslack (2.2.0), PrintJobHistory (1.17.0), PrintTimeGenius Plugin (2.2.8), Slicer Thumbnails (1.0.0), TP-Link Smartplug (1.0.2)

### Location of the problem

Web Interface

### Description of the problem

```markdown
The green box which lists the cumulative remaining filament is incorrect.
```

### Steps taken to reproduce

1. Add some spools
2. Review the numbers in Filament Tab > Filament Statistics > Material Overview > Remaining

### What did you expect to happen

Expected all columns to the right to be added together and that value shown for cumulative remaining amount
[image]
Less of an issue with only a few materials as you can add them up manually.

[image]
But more of a problem when you have lots.

NotExpectedYet, 2022-06-16:
Hu weird, must have goofed the calculation somewhere cheers for the report. I "think" it's taking into account something it shouldn't.

SuperTrollMan, 2022-06-17:
Upgraded to 1.2.9 but the issue remains.
However if I refresh the page, it momentarily shows the correct value (split second), then reverts to the prior incorrect value.
The overall filament statistics bar at the top also momentarily fills up, then returns to blank.

[image]

NotExpectedYet, 2022-06-17:
I'm pickled completely by this... 

I've fixed the bar at least

NotExpectedYet, 2022-06-17:
Could you try something for me please? 

run `npm run install-server` 

inside the OctoFarm directory 

and then `pm2 restart OctoFarm` 

If my theory is correct that should solve this specific issue.

NotExpectedYet, 2022-06-17:
Yeah confirmed the auto updater didn't install the new client grr. Have sorted for the next patch release. This should resolve the issue.

NotExpectedYet, 2022-06-19:
Going to close this as it should be resolved with the latest 1.2.10 release. 

The auto updater will fail to update the client, but the correct installed client will show now, so I recommend doing a manual update one last time. Both restart issue and that are resolved going forward.

closed 2022-06-19 by NotExpectedYet