OctoFarm/OctoFarm #1098: Resized "Weekly Utilisation" dashboard tile reverts to default

SuperTrollMan, 2022-06-17:
### OctoFarm Version

1.2.9

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu 20.04.4

### OctoPrint Plugins

Bed Visualizer (1.1.1), Cancel Objects (0.4.7), Consolidated Tabs (1.0.1), Cost Estimation (3.5.0), Dashboard (1.19.8), DisplayLayerProgress Plugin (1.28.0), Filament Manager (1.9.1), Octoslack (2.2.0), PrintJobHistory (1.17.0), PrintTimeGenius Plugin (2.2.8), Slicer Thumbnails (1.0.0), TP-Link Smartplug (1.0.2)

### Location of the problem

Web Interface

### Description of the problem

```markdown
The Weekly Utilisation dashboard widget/tile reverts to default size when navigating away from the dashboard or refreshing the page.
```

### Steps taken to reproduce

1. Set the dashboard up
2. Resize the Weekly Utilisation widget and move it to a nice spot
3. Admire your dashboard
4. Navigate to another tab (panel, camera, printers, file manager, etc)
5. Navigate back
6. Weekly Utilisation has been restored to original size and moved to a space it fits

### What did you expect to happen

The layout to stay the same
[image]
[image]
[octofarm_dump-6-17-2022-11-33-19 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8927051/octofarm_dump-6-17-2022-11-33-19.AM.zip)

NotExpectedYet, 2022-06-19:
Cannot for the life of me reproduce this. 

What browser are you using?

SuperTrollMan, 2022-06-19:
It wasn't happening to all the widgets, just that one.
The rest could be moved and resized without issue.

Generally I use Chrome,
Tried it in Edge just now and it did the same thing.
Also tried an incognito Chrome, did the same thing.

BUT, then I cleared my Chrome data for all of eternity, moved them all around some more and its taken it.
working in edge and incognito chrome aswell now?

How does that work, no idea - because Edge and Chrome incognito wouldn't have any cached data to cause the issue

Lets call it a client env bug.

closed 2022-06-19 by SuperTrollMan