OctoFarm/OctoFarm #1102: 2 of 4 printers not listed in File Manager (File Manager - Cannot read properties of undefined)

SuperTrollMan, 2022-06-19:
### OctoFarm Version

1.2.9

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu 20.04.4

### OctoPrint Plugins

Bed Visualizer (1.1.1), Cancel Objects (0.4.7), Consolidated Tabs (1.0.1), Cost Estimation (3.5.0), Dashboard (1.19.8), DisplayLayerProgress Plugin (1.28.0), Filament Manager (1.9.1), Octoslack (2.2.0), PrintJobHistory (1.17.0), PrintTimeGenius Plugin (2.2.8), Slicer Thumbnails (1.0.0), TP-Link Smartplug (1.0.2)

### Location of the problem

Server

### Description of the problem

```markdown
4 OP instances are connected
2 of them didn't appear in the File Manager tab and an error was thrown (attached to end)
```

### Steps taken to reproduce

1. All printers powered and connected
2. Navigated to File Manager, deleted some files
3. Navigated around OF
4. Came back to File Manager, error flagged, 2 printers not listed
5. Navigated to Printers tab, all printers listed, connected and health checks passed
6. Came back to File Manager, error flagged (again), 2 printers not listed

### What did you expect to happen

4x printers to be listed
[image]
[octofarm_dump-6-19-2022-6-36-32 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8934887/octofarm_dump-6-19-2022-6-36-32.AM.zip)

NotExpectedYet, 2022-06-19:
Nothing in the logs I could find as to why it wouldn't have this information. There we're some certificate errors but they failed at version grab, and there was a 403 a while back but that's about it. 

I've fixed the client for the next pull so it won't fail without this data. I'll leave this open for a bit just to see if it triggers again. The data was there, and I've nothing that would explicitly remove it so not sure why unless it was just random happenstance but that ain't a good resolution.

SuperTrollMan, 2022-06-23:
Kind of related:
Seeing it in 1.4.1

[image]

[octofarm_dump-6-23-2022-9-47-12 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8966121/octofarm_dump-6-23-2022-9-47-12.AM.zip)

NotExpectedYet, 2022-06-23:
Certainly related

SuperTrollMan, 2022-06-24:
I don't suppose we could have it skip the failing printer and continue to load the remaining working printers?

NotExpectedYet, 2022-06-24:
> I don't suppose we could have it skip the failing printer and continue to load the remaining working printers?

I'm going to put some better checks and fallbacks in, just didn't have time this morning. Will be working on it tomorrow :). If data not grabbed keep database values, if no database values fail for re-scan.

NotExpectedYet, 2022-06-25:
I'm "sure" I fixed this so it will close. Just let me know if it hasn't resolved.

SuperTrollMan, 2022-06-26:
I've tested on 1.5.0, but the issue looks to remain.

Working fine:
[image]

Restart server

No tool (due to profile issue)
[image]

Pressing manage > settings and waiting for it to refresh/check settings resolves the issue still.

Secondly, the health checks don't warn in the alerts column of the printers tab.

[image]

But health checks does report the issue.

[image]

Thanks!

SuperTrollMan, 2022-06-30:
Sorry mate, still doing it for me in 1.5.1
Is there something you need me to do to clear it, or do you want me to leave it so we can keep replicating?
[image]

NotExpectedYet, 2022-06-30:
> Sorry mate, still doing it for me in 1.5.1 Is there something you need me to do to clear it, or do you want me to leave it so we can keep replicating? [image]

I think you may have to delete and re-add your printer. 

I checked your export and it had all the required database stuff and there's nothing that would stop it loading from that side, and it now uses the database data before hand. If it doesn't resolve after deleting and re-adding let me know and I'll re-open.

SuperTrollMan, 2022-08-07:
This printer 3 is still having you on by the looks.

**1.6.0** Seems a little worse
Issue remains on file manager tab

New issue: Health Check popup fails to populate, looks like it hangs on it
[image]

Cant open settings to clear it like previous as it now errors too
[image]

[octofarm_dump-8-7-2022-12-59-38 AM.zip](https://github.com/OctoFarm/OctoFarm/files/9275979/octofarm_dump-8-7-2022-12-59-38.AM.zip)

SuperTrollMan, 2022-08-07:
I was able to temporarily clear it by connecting the printer in OP
Then going to settings in OF, setting again the port and re-saving.
[octofarm_dump-8-7-2022-1-05-05 AM.zip](https://github.com/OctoFarm/OctoFarm/files/9275982/octofarm_dump-8-7-2022-1-05-05.AM.zip)

Will likely return when I restart OF next.

However, its also not honoring the power on delay (which the other printers are), triggering OP to flag this error:
[image]

NotExpectedYet, 2022-09-04:
There any update with this for V1.6.6? 

I've honestly lost track of it.

SuperTrollMan, 2022-09-16:
Yep, I can still replicate it in 1.6.6

Time for a new OF instance you think? :P

NotExpectedYet, 2022-09-17:
I'd see if it fixes it personally. If it does fix it I'm even more confused 😂

NotExpectedYet, 2022-10-19:
We anywhere else with this? 

I've just not had any other users mention anything regarding the issue for a while. 

Thinking either

The issue is resolved and something is stuck in your instance. 

The issue is with your OctoPrint installation -> failing as card. My API calls are usually the first to go in the situation but not a 100% more a guess. 

The issue is still lurking in the darkness and somehow your triggering it with 1 printer.

commit a0637403 (2022-06-25): fix(server): not awaiting state call to finish before grabbing profile closes #1102
commit 13f2bf8f (2022-06-25): fix(server): not awaiting state call to finish before grabbing profile closes #1102
commit cb143953 (2022-06-29): fix(server): not correctly dealing with database and captured profile data, closes #1102
commit 9a0045cd (2022-06-29): fix(server): not correctly dealing with database and captured profile data, closes #1102