OctoFarm/OctoFarm #1104: History Tab - Cost Calculation Issue

SuperTrollMan, 2022-06-19:
### OctoFarm Version

1.2.11

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu 20.04.4

### OctoPrint Plugins

Bed Visualizer (1.1.1), Cancel Objects (0.4.7), Consolidated Tabs (1.0.1), Cost Estimation (3.5.0), Dashboard (1.19.8), DisplayLayerProgress Plugin (1.28.0), Filament Manager (1.9.1), Octoslack (2.2.0), PrintJobHistory (1.17.0), PrintTimeGenius Plugin (2.2.8), Slicer Thumbnails (1.0.0), TP-Link Smartplug (1.0.2)

### Location of the problem

Web Interface

### Description of the problem

```markdown
History Tab - Combined Total cost doesn't match Filament Cost + Printer Cost.
History Tab - Average cost per hour doesn't calculate
History Tab > Statistics > $Costs/Usage (State statistics) > Print totals Per Month - Doesn't calculate total count
```

### Steps taken to reproduce

1. Print some things
2. Navigate to the History tab

### What did you expect to happen

Calculate correctly

### To avoid the problem I had to

[image]
[image]

NotExpectedYet, 2022-06-19:
Huh, weird will take a look cheers for the report

What do you mean by the last one sorry? I can see the first too they're certainly incorrect. Think I know why. 

The last one seems to be calculating correctly, although you've not put a screenshot of the cost modal.

They only update once per day though which I did wonder if it was confusing or not but I wasn't sure on CPU usage for doing a full years worth of calculations like that on the database. I'll flood a database through and test see what it's like.

closed 2022-06-22 by NotExpectedYet

SuperTrollMan, 2022-10-11:
re: point 3

The print totals per month graph reads successful prints: 9
Total prints: 0

Shouldn't total prints read the sum of all (failed, cancelled and success)?

Kind Regards

Sent from mobile, please excuse my brevity.

commit 7cfbead7 (2022-06-22): fix(client): fix history state graph total count not generating closes #1104
commit 41ecfa4a (2022-06-22): fix(client): fix history state graph total count not generating closes #1104