OctoFarm/OctoFarm #1133: Lifespan in printer settings resets to default

taker218, 2022-06-24:
### OctoFarm Version

1.4.3

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

docker

### Operating System

Debian

### Location of the problem

Server

### Description of the problem

```markdown
I wanted to set the correct cost settings to the ones I have calculated but it seems like the value for the life span is always reset to the default (43800 hours).

The rest of the values is stored properly.
```

### Steps taken to reproduce

1. Set the lifespan to any value other then the default
2. Save the settings
3. Have a look at the value of the lifespan

### What did you expect to happen

That the correct value is stored and shown

Here is the log dump:
[octofarm_dump-6-24-2022-10-35-55 AM.zip](https://github.com/OctoFarm/OctoFarm/files/8974614/octofarm_dump-6-24-2022-10-35-55.AM.zip)

As far as I can see, the correct value is updated in the log, maybe it's just not correctly shown in the client.

NotExpectedYet, 2022-06-24:
I seem to remember fixing that recently, think it is just client side yeah. Cheers for the report, hoping to push another version over weekend so will add it in!

NotExpectedYet, 2022-06-25:
That was it not saving to the database... sorry about that had `estimateLifeSpan` instead of `estimateLifespan` on the key!

commit 7c65250f (2022-06-25): fix(server): not saving Estimated Life Span settings for printer properly closes #1133
commit 52fe1612 (2022-06-25): fix(server): not saving Estimated Life Span settings for printer properly closes #1133
closed 2022-06-25 by NotExpectedYet