OctoFarm/OctoFarm #1145: File Manager Statistics Incorrect

SuperTrollMan, 2022-06-26:
### OctoFarm Version

1.5.0

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu 20.04.4

### OctoPrint Plugins

Bed Visualizer (1.1.1), Cancel Objects (0.4.7), Consolidated Tabs (1.0.1), Cost Estimation (3.5.0), Dashboard (1.19.8), DisplayLayerProgress Plugin (1.28.0), Filament Manager (1.9.1), Octoslack (2.2.0), PrintJobHistory (1.17.0), PrintTimeGenius Plugin (2.2.8), Slicer Thumbnails (1.0.0), TP-Link Smartplug (1.0.2)

### Location of the problem

Server

### Description of the problem

```markdown
The File Manager Statistics reads incorrect data:
1 - The used/remaining/total stats list for only the first printer
(not sure if its meant to list them all, or just the selected one from the left pane)

2 - Largest, smallest and average reads 0 for size and length.
```

### Steps taken to reproduce

1. Load some files
2. Navigate to File Manager > Management > Manage > Statistics

[image]

[image]

### What did you expect to happen

.

NotExpectedYet, 2022-06-26:
[image]

Wow when did I break that -_-

closed 2022-06-29 by NotExpectedYet

SuperTrollMan, 2022-08-07:
Hey mate,
I'm still having part of this issue on 1.6.0.

The items below the line are working, but the statistics bar above the line just shows values for the first printer in the list only.

commit 136ac0a7 (2022-06-28): fix(server): not using correct information to generate file stats, closes #1145
commit 78fbd97c (2022-06-29): fix(server): not using correct information to generate file stats, closes #1145