OctoFarm/OctoFarm #1147: Receiving a TypeError on all of the printer views

brooksware2000, 2022-06-26:
### OctoFarm Version

1.5.0

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu Server on Raspberry Pi 4

### Location of the problem

Web Interface

### Description of the problem

```markdown
Whenever I go to any of the printer views (Panel, List, Camera, Group, Superlist, Current Operations), I receive the following pop-up error:

TypeError ERROR (undefined):
Please report this error to OctoFarm Issues!
Cannot read properties of undefined (reading 'operations')
FILE INFO
LINE: undefined
COL: undefined
STACK:
TypeError: Cannot read properties of undefined (reading 'operations') at http://192.168.1.118:4000/assets/dist/combined-view.runner.min.js?1.5.0:2:855567
```

### Steps taken to reproduce

1.  Click on any of the printer views (Panel, List, Camera, Group, Superlist, Current Operations).

### What did you expect to happen

N/a

### To avoid the problem I had to

N/a

brooksware2000, 2022-06-26:
Here is a screenshot and logfiles:
[OctoFarm.Printer.OctoPrint2.log](https://github.com/OctoFarm/OctoFarm/files/8986833/OctoFarm.Printer.OctoPrint2.log)
[image]

NotExpectedYet, 2022-06-26:
Is this a fresh install?

NotExpectedYet, 2022-06-26:
Try saving your client settings see if it resolves.

brooksware2000, 2022-06-26:
> 

No, I ran the internal upate.

brooksware2000, 2022-06-26:
> Try saving your client settings see if it resolves.

No dice here either.

NotExpectedYet, 2022-06-26:
Can you check what client version you have please?

brooksware2000, 2022-06-26:
Client version is 1.5.0

NotExpectedYet, 2022-06-27:
Weird, it's something server side by the looks as it's the current operations data. 

Would you mind generating a log dump for me please? It will have all the logs in it as there won't be anything in the printer data one. 

System -> Logs -> Log Dump

NotExpectedYet, 2022-06-27:
Would you also mind trying to clear your cache as well? It shouldn't be needed but that file referring to the "operations" key is really old which is why I'm confused as it's handled by the event.js file, not any of those view files, it's not even imported. So think it might be the browser not updating a file correctly. 

I'll work in some better cache busting system for the next version as it's currently only appending `?1.5.0` to the urls which isn't very robust.

NotExpectedYet, 2022-06-27:
[image]

Found it, I wasn't using the cache at all for this file. Should resolve when you clear cache, and next version will not require it.

commit a1d48448 (2022-06-29): refactor(client): improve cache busting without relying on ?version, closes #1147
closed 2022-06-29 by NotExpectedYet