OctoFarm/OctoFarm #1157: Failed due to Global API Key detection

robin7331, 2022-06-30:
### OctoFarm Version

1.5.1

### OctoPrint Version(s)

1.8,1

### pm2 or docker?

pm2

### Operating System

Windows

### Location of the problem

Not sure

### Description of the problem

```markdown
I am having a bunch of RPi4's with OctoPrint 1.8.1 installed.
I have created a new Application/User Key and copied that to OctoFarm which I have setup using the node/pm2 method on a Windows 10 machine.

Once I add the printer the OctoPrint API badge turns green and says 'Online'. However the Printer State shows a red "Global API Fail!" badge. If I force reconnect I get a `Failed due to Global API Key detection triggering` popup. 

I am def. not using the Global API key from OctoPrint.
Also I tried downgrading to OctoFarm 1.5.0 and even 1.4.3 but that did not change anything.
```

### Steps taken to reproduce

Not sure what to write here. :/

### What did you expect to happen

I would have expected OctoFarm to be able to connect to the printer :/

robin7331, 2022-06-30:
Dang the problem was that I was trying to add machine with the hostname and not the IP.
Had no clue that this was unsupported :)

Anyways. Its working! 🥲

swindmiller, 2022-07-03:
Don't know if this is related but I had a similar issue where my pi would show online, Printer State: Disconnected and Websocket was not connected. My issue turned out to be multiple users setup on my PI. I noticed the username showing under Octoprint User in OF did not match my logged in user in OP (the one I generated the API key from). As soon as I deleted all but one account in OF and re-added the printer to OF all was good. 
This was only an issue on 1 of 7 printers and didn't even know I had multiple accounts setup in OF. May be worth a shot to check.

closed 2022-08-06 by NotExpectedYet