OctoFarm/OctoFarm #1158: Freezing when printer disconnects

swindmiller, 2022-07-05:
### OctoFarm Version

1.5.1

### OctoPrint Version(s)

1.5.3 1.6.1 1.7.2

### pm2 or docker?

docker

### Operating System

Unraid Docker

### OctoPrint Plugins

Heater Timeout, PrintTimeGenius, PSU Control

### Location of the problem

Web Interface

### Description of the problem

```markdown
Just updated to 1.5.1 and noticed twice now where it looks semi-frozen. Everything appears to be working but what I am noticing is if I have a printer go offline for some reason the status on my other printers (like current print status) does not reflect correctly or is not updating. I will try to give an example. I look at the list view and all printers show online, connected and not printing but I have 2 currently printing. I tried refreshing and everything like that with no change. I restart OctoFarm and I now see the status correctly on all printers (the 2 printers printing show printing) and one printer is now offline (API issue, that turns out my Octoprint stopped), that did not show offline before the reboot but was most likely offline earlier.
Not knowing how everything works, to me, it appears that because one printer had an issue it kind of threw something off and I have to restart OF to get everything to show correctly. This does not appear to happen every time a printer goes offline but has happened twice in the past few days since I updated from 1.1.13. Its just concerning that an issue with one printer causes all the rest to show incorrectly.
I have the logs I took while it was (semi-frozen) and then again right after I rebooted if it would help, I can send them somewhere if that will help troubleshoot.
```

### Steps taken to reproduce

Notice a printer that is physically printing does not show as printing in OF, all printers show online though 

Restart OF and printer that is printing now shows printing but one other printer is now offline due to a Pi issue (OctoPrint).

Happened 2 different times with different printers being offline. The printers being offline had nothing to do with OF.

### What did you expect to happen

Printer not working would show disconnected (without needing to reboot) and other printers would show the correct status.

### To avoid the problem I had to

Check OF and if it appeared "frozen", restart OF

NotExpectedYet, 2022-08-06:
I think this should be fixed in the next release. I can't reproduce anyway by having a printer go offline.

NotExpectedYet, 2022-08-06:
Please let me know if this isn't fixed, will close for now and keep an eye out myself

closed 2022-08-06 by NotExpectedYet