OctoFarm/OctoFarm #1159: Camera view collapsed when accessing OctoFarm from local DNS name

HeyRay2, 2022-07-06:
### OctoFarm Version

Client: 1.4.1 | Server 1.4.3

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

docker

### Operating System

Ubuntu Server

### Location of the problem

Web Interface

### Description of the problem

```markdown
When accessing OctoFarm from a local DNS name (example: octofarm.mydomain.com) the camera views are collapsed.

Accessing the same OctoFarm instance from the IP address results in the camera views being properly rendered.
```

### Steps taken to reproduce

1. Create a local DNS entry for OctoFarm (example: octofarm.mydomain.com)
2. Access OctoFarm from local DNS name
3. Go to "Camera" view
4. Confirm that camera views are collapsed
5. Access OctoFarm from IP address
6. Go to "Camera" view
7. Confirm that camera vies are properly rendered

[image]

### What did you expect to happen

Camera views should be properly rendered even when accessing OctoFarm from a local DNS entry

### To avoid the problem I had to

Access OctoFarm by IP address instead of local DNS entry

NotExpectedYet, 2022-07-06:
[image]

It's HTTPS, if your camera's are http accessible then it won't route due to a browser limitation. 

I recommend updating to the latest 1.5.1 version and then turning on the camera proxy in the settings. It will route through your DNS name properly then. 

v1.4.3 of the server disabled this functionality been automatic as there was a limitation with 4 streams for some reason. I've now fixed this and improved the functionality but left it up to the user to enable. 

You can find the setting here: 

[image]

HeyRay2, 2022-07-06:
Thanks @NotExpectedYet.

The OctoFarm Docker images doesn't appear to be updated to 1.5.1 yet (pulling "Latest" keeps pulling 1.4.3), but I will give your suggestion a shot after I'm able pull down this latest version.

NotExpectedYet, 2022-07-06:
> Thanks @NotExpectedYet.
> 
> The OctoFarm Docker image doesn't appear to be updated to 1.5.1 yet (pulling "Latest" keeps pulling 1.4.3), but I will give your suggestion a shot after I'm able pull down this latest version.

Hmm weird, let me check if my automation ran.

NotExpectedYet, 2022-07-06:
My builds broke sometime recently. Will take a look and let you know, cheers for pointing that out!

NotExpectedYet, 2022-08-06:
Docker is now built so closing this.

closed 2022-08-06 by NotExpectedYet

HeyRay2, 2022-08-06:
Excellent! Thanks!

NotExpectedYet, 2022-08-06:
Do let me know if there any other issues :)