OctoFarm/OctoFarm #1162: Error on dependencies (dashboard)

gonchogg, 2022-07-11:
### OctoFarm Version

latest 1.5.1

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Debian

### Location of the problem

Web Interface

### Description of the problem

```markdown
There is a pop up that advice me everytime i enter the dashboard menu, i suppose there is an old variable calling. screenshoot at the end
It only pop up in dashboard menu, but all works fine, its just the annoying pop up display.
Can someone help me to delete that error?

Sorry for my english and thx for all the work!
```

### Steps taken to reproduce

1.Enter dashboard in web interface
2.close the error pop up
3.all works fine

### What did you expect to happen

Dont show me the pop up

### To avoid the problem I had to

close the pop up
[image]

NotExpectedYet, 2022-07-11:
Hello thanks for the praise. Do you have the environmental data dashboard enabled in System -> Client -> Dashboard? 

Weird it's loading that data if you haven't like. Will take a look but I think if that's enabled, disabling it will work as it's expecting data to the chart and there isn't any.

gonchogg, 2022-07-12:
Thanks for the support, i disabled the environmental data and now the pop up doesnt appear, but when i try to enable current operations it doesnt change to enabled status.
What i have to do to activate the 2 options?
I suppose its a version bug

Thanks for the fast help!

gonchogg, 2022-07-12:
[image]

gonchogg, 2022-07-12:
[image]

NotExpectedYet, 2022-07-12:
> Thanks for the support, i disabled the environmental data and now the pop up doesnt appear, but when i try to enable current operations it doesnt change to enabled status. What i have to do to activate the 2 options? I suppose its a version bug
> 
> Thanks for the fast help!

Thanks for the confirmation. I'll put some checks in so that doesn't try and load without the data for the environmental one. 

I've just confirmed the second issue regarding current operations. I'll put that on my list to sort for the next release :) this ticket will automatically close once that's done. 

I'll update the title to reflect that.

commit cff13f74 (2022-08-06): fix(client): not been able to enable current operations on dashboard, closes #1162
closed 2022-08-06 by NotExpectedYet