OctoFarm/OctoFarm #1165: File Manager - TypeError: t.currentProfile is undefined

AronsPC, 2022-07-15:
### OctoFarm Version

1.5.1

### OctoPrint Version(s)

1.8.1

### pm2 or docker?

pm2

### Operating System

Ubuntu

### Location of the problem

Server

### Description of the problem

```markdown
File Manager not displaying Printers.

Getting ...
OctoFarm Error ERROR (999):
FILE INFO
LINE: 2
COL: 609428
FILE:
http://10.10.2.170:4000/assets/dist/file-manager.runner.1.5.1.min.js
STACK:
n@http://10.10.2.170:4000/assets/dist/file-manager.runner.1.5.1.min.js:2:609428 @http://10.10.2.170:4000/assets/dist/file-manager.runner.1.5.1.min.js:2:711761 

I can however access the files from the Panel view individually with each printer. Currently running three.
```

### Steps taken to reproduce

Clicking on the File Manager Tab

### What did you expect to happen

Display each Printer with re-sync controls

AronsPC, 2022-07-15:
Which log file specifically would I need to upload?

NotExpectedYet, 2022-07-16:
> Which log file specifically would I need to upload?

You can do a log dump on System -> Logs -> Log Dump. It will generate a .zip file with everything I should need. 

Think it's related to this #1102

AronsPC, 2022-07-16:
[octofarm_dump-7-16-2022-8-03-16 AM.zip](https://github.com/OctoFarm/OctoFarm/files/9125689/octofarm_dump-7-16-2022-8-03-16.AM.zip)
Here it is. Thanks. I generated the error message again just before downloading it.

AronsPC, 2022-07-16:
Awhile back I was getting a similar error with the Spool Manager but it went away. The last effort attempt I was going to do was a fresh install in a new VM and try and transfer over all my information.

NotExpectedYet, 2022-07-17:
That's perfect thanks. 

Will take a look through. 

It's to do with the profile not scanning. Another user reported that after boot, opening the Printers -> Manage -> Settings use to trigger the data back.

AronsPC, 2022-07-17:
That work around did the trick just now. I remember finding something similar with the Spool Manager before as well. If I went to the printer directly it would let it re-appear.

NotExpectedYet, 2022-07-31:
I'm hoping this will be solved in the next version as I've re-written the onboarding logic for a printer.

commit e3bccdf9 (2022-08-06): refactor(server): initial onboarding for printers from configuration file, closes #1165 #1102 #1157
commit 108ad66d (2022-08-06): refactor(server): initial onboarding for printers from configuration file, closes #1165 #1102 #1157
closed 2022-08-06 by NotExpectedYet