OctoFarm/OctoFarm #1167: Correct filament spool not selected in Octoprint

KKriz, 2022-07-20:
### OctoFarm Version

1.5.1

### OctoPrint Version(s)

latest

### pm2 or docker?

docker

### Operating System

Unraid

### Location of the problem

Not sure

### Description of the problem

```markdown
I have the correct filaments loaded according to the file manager. I have PETG on printer 1 and PLA on printer 2. However, if I open Octoprint they both have PETG as the selected spool?
```

### Steps taken to reproduce

 I set both spools to no spool in Octoprint  to rule out an issue there. Selected the correct spools in Octofarm, and then it loaded both printers with PETG. Octofarm itself still indicates the selected PETG and PLA in use.

### What did you expect to happen

The selected spools to be used by Octoprint

NotExpectedYet, 2022-07-31:
I couldn't reproduce with your steps starting from nothing assigned but I did manage to get it confused. I think the issue is a mis-match between setting on OP / setting on OF as OF doesn't pay much attention to what's currently on OP. So if something is selected on OP then OF won't know about it. 

I'm not sure if this will be fixed the next version but I'm looking into making the plugin support more robust as a whole starting with collecting the currently selected spools. 

There's also another bug I've noticed, you can't select "No Spool" on OctoFarms side without it causing a crash.

NotExpectedYet, 2022-09-04:
Has this resolved with v1.6.6? I tried it recently but I couldn't reproduce the exact issue. The "No Spool" was resolved. 

Just to check. 

I don't think it should have as nothing has changed although I have made some tweaks to the integration. 

Next version I will be adding a scan on boot for the currently selected spool and a listener on the websocket message the plugin sends so these should stay in sync a lot better.

KKriz, 2022-09-07:
No, Still having issues. No spool selected at all in Octoprint or the same one used on both of my printers.

NotExpectedYet, 2022-09-20:
Closing because next release removed the plugin support check #1202 for more information

closed 2022-09-20 by NotExpectedYet